espressif/esp-box #96: chatgpt demo 获取chatgpt结果后，一直没有tts语音合成播放，tts返回超时

MichaelDu9226, 2023-09-04:
在espressif__esp-box中的BSP_LCD_SUB_NAME选择为esp32-s3-box-3后，编译使用已经没有问题。

但是在获取chatgpt结果后，一直没有tts语音合成播放，导致需要等很久，40s左右，tts超时报错后，才能进行下一次语音交互。
[image]

另外请问：
1. 如何选择配置成中文交互，目前提中文问题，会自动转为英文问题，获得英文结果。
2. setting的Region Select，OpenAI及Espressif是用来配置什么的？

https://www.bilibili.com/video/BV1La4y1578f

MichaelDu9226, 2023-09-04:
[image]
如图所示，已经可以获得结果，但是并没有语音合成及播放，40s后会自动超时退出到待机界面

MichaelDu9226, 2023-09-04:
我看bilibili里的视频是中文问题，中文结果。但是我这边中文也会识别为英文问题，然后出英文结果。
我测试过很多遍，每次结果的TTS合成都失败了，所以会不会是dui.ai的接口或者网络问题？

alibukharai, 2023-09-04:
@MichaelDu9226  Default language is set to English. For Chinese, please attempt to modify this [code line](https://github.com/espressif/esp-box/blob/e2bb387d67145d3935842e362f1f44d88692acba/examples/chatgpt_demo/main/main.c#L42C5-L42C66) and test it. The result is unsatisfactory because we lack control over the CHATGPT response.

Here is the modifiedline 
audioTranscription->setLanguage(audioTranscription,"zh");

MichaelDu9226, 2023-09-04:
@alibukharai 谢谢，修改为audioTranscription->setLanguage(audioTranscription,"zh")可以成功显示中文了
@espressif2022 但是tts 失败的问题还是没解决，我将https://github.com/espressif/esp-box/blob/e2bb387d67145d3935842e362f1f44d88692acba/examples/chatgpt_demo/main/tts_api.c#L136C15-L136C15的URL打印出来后，直接在浏览器调用这个接口的时候，是可以获取mp3的
所以很奇怪

MichaelDu9226, 2023-09-08:
@espressif2022 你好 请问tts的问题，有什么解决办法吗
我测试了一下URL放到浏览器里是可以获得音频的

espressif2022, 2023-09-08:
> @espressif2022 你好 请问tts的问题，有什么解决办法吗 我测试了一下URL放到浏览器里是可以获得音频的

我刚下午有测试过，是可以的。text 比较长的情况下会失败，这需要优化。
短的语句测试还是 OK 的

alibukharai, 2023-09-08:
@MichaelDu9226 I recommend reducing the MAX_TOKEN value to 100 in the code found [here](https://github.com/espressif/esp-box/blob/e2bb387d67145d3935842e362f1f44d88692acba/examples/chatgpt_demo/main/main.c#L47). Longer texts require more processing time and can lead to HTTPS timeouts. A smaller MAX_TOKEN value allows CHATGPT to generate shorter responses, improving Text-to-Speech (TTS) performance. You can find more information about TTS in the [documentation](https://github.com/espressif/esp-box/tree/master/examples/chatgpt_demo#text-to-speech).

MichaelDu9226, 2023-09-11:
@espressif2022 @alibukharai 
感谢两位的回复，我找到tts合成失败的原因了，与所使用的的网络环境有关。demo里使用的dui.ai好像对国外网络支持不是很好，导致语音合成失败。
因为openai的API需要外网IP，而且我们的产品主要也是国外使用。请问是否有适合国外的语音合成API推荐，谢谢。

espressif2022, 2023-09-20:
> @espressif2022 @alibukharai 感谢两位的回复，我找到tts合成失败的原因了，与所使用的的网络环境有关。demo里使用的dui.ai好像对国外网络支持不是很好，导致语音合成失败。 因为openai的API需要外网IP，而且我们的产品主要也是国外使用。请问是否有适合国外的语音合成API推荐，谢谢。

可以试下微软或者亚马逊的，不过这些都是需要收费的，具体可以自己适配下，我们短时间没有计划开展

ESP-Mars, 2023-11-09:
Temporarily closed due to the extended period of time with no response. Should you have additional questions or concerns, don't hesitate to reopen the issue.

closed 2023-11-09 by ESP-Mars