espressif/esp-box #97: GPT Demo on M5Stack CoreS3 ?

recrudesce, 2023-09-04:
Not an issue, more a question.

Would your GPT demo for your ESP32-S3-BOX potentially work on an M5Stack CoreS3 ?  Looks like essentially the same product...

espressif2022, 2023-09-04:
Hello, M5Stack Core3 is not maintained by us，but I think you can try to port it yourself.
I think It worked.

recrudesce, 2023-09-04:
I know it's not maintained or made by you, just wondering if the underlying hardware would be similar.  I'll try a port and see what happens :)

espressif2022, 2023-09-05:
> I know it's not maintained or made by you, just wondering if the underlying hardware would be similar. I'll try a port and see what happens :)

It's similar, they tried our `components_esp_sr` last week, we add a wakenet(`HI, I'm five`) for them.

ESP-Mars, 2023-11-09:
Temporarily closed due to the extended period of time with no response. Should you have additional questions or concerns, don't hesitate to reopen the issue.

closed 2023-11-09 by ESP-Mars