espressif/esp-box #98: API Key is not Valid

ld-t, 2023-09-05:
```
E (22400) app_audio: [Start] start_openai, timestamp: 22400
I (22400) OpenAI: OpenAI create, version: 0.1.2
I (22402) ui_ctrl: Swich to panel[2]
I (22421) I2S_IF: Pending out channel for in channel running
E (22425) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (22425) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (22432) I2S_IF: STD Mode 1 bits:16/16 channel:2 sample_rate:16000 mask:3
I (22458) Adev_Codec: Open codec device OK
E (22458) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (22461) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (22467) I2S_IF: STD Mode 0 bits:16/16 channel:2 sample_rate:16000 mask:3
I (22477) Adev_Codec: Open codec device OK
I (26445) app_main: replay audio end
E (40683) esp-tls: [sock=54] delayed connect error: Software caused connection abort
E (40683) esp-tls: Failed to open new connection
E (40686) transport_base: Failed to open a new connection
E (40692) HTTP_CLIENT: Connection failed, sock < 0
E (40697) OpenAI: ./managed_components/espressif__openai/OpenAI.c:2203 (OpenAI_Request):Failed to open client!
E (40708) OpenAI: ./managed_components/espressif__openai/OpenAI.c:1903 (OpenAI_AudioTranscriptionFile):Empty result!
I (40720) ui_ctrl: update listen speak
E (40724) app_audio: [End] start_openai, +offset:18324
```

My API Key is generated by my openAI account and is valid, but error

espressif2022, 2023-09-05:
did you printf the api_key for double check?

espressif2022, 2023-09-05:
> 我的IP在内地，电脑连接wifi再代理可以访问gpt，esp连接wifi不行，esp的代理怎么设置

I'm sorry, I think I can't support this.

closed 2023-09-06 by ld-t

kawashimaken, 2023-10-03:
I use ESP32-S3-Box-Lite.
I have newly created API key and the  date of the latest usage is shown "never used", after I start the chatgpt demo, I asked a question like "What is your name?" or something, and it responsed: "I am thinking about your question", after that it shows "API key is not valid", at this point I checked the Open AI API page, the last used date updated to <today's date> , so it means it somehow accessed the API endpoint. 
Any hint why is is shown "API key is not valid", I use idf.py menuconfig to set the API key, I just copied it, so it supposed to be correct.

kawashimaken, 2023-10-07:
> I use ESP32-S3-Box-Lite. I have newly created Open AI API key and the date of the latest usage is shown "never used", after I start the chatgpt demo, I asked a question like "What is your name?" or something, and it responsed: "I am thinking about your question", after that it shows "API key is not valid", at this point I checked the Open AI API page, the last used date updated to <today's date> , so it means it somehow accessed the API endpoint. Any hint why is is shown "API key is not valid", I use idf.py menuconfig to set the API key, I just copied it, so it supposed to be correct.

Well, it is working now!
The reason is that I didn't pay for enabling the key, after I added billing info and paid for 5 dollar to the credit balance, now it works without problem!

ESP-Mars, 2023-10-07:
> Well, it is working now! The reason is that I didn't pay for enabling the key, after I added billing info and paid for 5 dollar to the credit balance, now it works without problem!

Great, glad to hear it.

ld-t, 2023-10-08:
> Well, it is working now! The reason is that I didn't pay for enabling the key, after I added billing info and paid for 5 dollar to the credit balance, now it works without problem!

Thanks a lot