espressif/esp-box #101: how to modify the voice file under spifs/

hyoutom, 2023-09-11:
Factory under the esp-box project_ Demo v0.3.0 branch, how to modify the voice file under spifs/? I replaced it with another voice file with the same name and format, and after re burning, an error E (26068) sr was reported_ Handler: Header of wav format error

espressif2022, 2023-09-12:
> Factory under the esp-box project_ Demo v0.3.0 branch, how to modify the voice file under spifs/? I replaced it with another voice file with the same name and format, and after re burning, an error E (26068) sr was reported_ Handler: Header of wav format error

You can refer to this place [wav opeation.](https://github.com/espressif/esp-box/blob/a45ee1e985c95fb0108c65814b3c7344ad035a0f/examples/factory_demo/main/app/app_sr_handler.c#L101C1-L101C1)

hyoutom, 2023-09-12:
The code here only modifies the file name. I am using an audio file of the same type and name, but if an error is reported, is there a location or tool that compiled the audio file into a binary file bin? What should I do?

espressif2022, 2023-09-15:
> The code here only modifies the file name. When I use an audio file of the same type and name, an error occurs during runtime. Is there a location or tool that compiled the audio file into a binary file bin? What should I do?

The audio format is the same?  Is there any log info?
Speaking from experience, existing libraries will fail to decode some MP3 files.

ESP-Mars, 2023-11-09:
Temporarily closed due to the extended period of time with no response. Should you have additional questions or concerns, don't hesitate to reopen the issue.

closed 2023-11-09 by ESP-Mars