espressif/esp-box #102: Install firmware V0.5.0 on ESP32-S3-BOX3 (revision v0.2) not work

SepuWeb, 2023-09-14:
Hi,
today I received my brand new ESP32-S£-BOX-3, and works very well.
after reading the real guides, I read that it is better to update the firmware so I ended up in this guide:

https://github.com/espressif/esp-box/blob/v0.3.0/docs/firmware_update.md#firmware-update-for-macos

I downloaded the ESP-BOX_Demo_EN_V0.5.0.bin file on my Mac and launched the various commands.
[image]

It does everything as directed but then the device no longer works! Black screen! I tried resetting, unplugging and reconnecting, but nothing.
Then I tried from Launchpad, I did ERASE FLASH but it gets stuck like this:

[image]

so i tried to install this:

but when I do RESET DEVICE from Console appen this
[image]

in loop.

Please help!! I am a newbie!!

espressif2022, 2023-09-15:
Got it, I'll check this firmware today.

SepuWeb, 2023-09-15:
I just noticed this warning at the top of this page:

https://github.com/espressif/esp-box/blob/master/docs/firmware_update.md

Warning:

The firmware released in the latest Tag branch ([Version v0.5.0](https://github.com/espressif/esp-box/releases)) is only applicable to ESP32-S3-BOX and ESP32-S3-BOX-Lite. For users who have purchased ESP32-S3-BOX-3, we will release firmware updates during the project's progression. We suggest that users update the firmware via Espressif's online firmware flashing tool [ESP Launchpad](https://espressif.github.io/esp-launchpad/?flashConfigURL=https://raw.githubusercontent.com/espressif/esp-box/master/launch.toml).

so what can I do now? Isn't there a way to restore the original image?

alibukharai, 2023-09-16:
@SepuWeb please check the firmware is already updated onto the [LaunchPad](https://espressif.github.io/esp-launchpad/?flashConfigURL=https://raw.githubusercontent.com/espressif/esp-box/master/launch.toml). Meanwhile you can also try to build the factory demo app by yourself by following the [guidelines](https://github.com/espressif/esp-box/tree/master/examples/factory_demo).

ESP-Mars, 2023-11-09:
Temporarily closed due to the extended period of time with no response. Should you have additional questions or concerns, don't hesitate to reopen the issue.

closed 2023-11-09 by ESP-Mars