espressif/esp-box #103: USB Audio Playback Issue with ESP32-S3 DevKitC

kadharsh, 2023-09-22:
When running the example code for the USB headset on my ESP32-S3 DevKitC board connected to the ES7210 and ES8311, I encountered an issue where the board was resetting due to a watchdog reset. To fix this, I added the line `vTaskDelay((TickType_t)pdMS_TO_TICKS(1));` inside the infinite loop in the `usb_task` function in the file `usb_headset.c`, which resolved the watchdog reset issue.
```
static void usb_task(void *pvParam)
{
    (void) pvParam;

    do {
        // TinyUSB device task
        tud_task();
        vTaskDelay((TickType_t)pdMS_TO_TICKS(1));
    } while (true);

    vTaskDelete(NULL);
}
```
However, when I play audio from my PC through the speaker connected to the ESP32, after a few seconds, the audio becomes noisy, kind of like a poor FM reception. After a few more seconds, it returns to normal, and this cycle repeats randomly.
Also, this issue only occurs when running the USB headset example. When running the MP3 example from esp-adf, the audio works correctly.

It appears that there is a problem with the USB audio handling How can I fix this issue? I would really appreciate any help. 

Hardware and Configuration:
* ESP32-S3 DevKitC board with 8MB flash and 2MB PSRAM.
* ES7210 and ES8311 connected with the same pinouts.

LHYhorion, 2023-11-10:
I'm sorry for the late response. Based on your situation, I conducted some tests recently. I tested on the ESP-S3-BOX (which also uses ES8311 and ES7210), and I didn't encounter the watchdog reset issue you mentioned. However, I did observe some noise when playing music on a Linux system. May I ask if you are playing music on a Linux system? Music playback on Windows seems to be normal. If you are experiencing noise while playing music on a Linux system, we will work on optimizing and fixing this issue in the future.

kadharsh, 2023-11-17:
Hey @LHYhorion, 

Thank you for your response. Yes, I was indeed using this with Linux. My intention was to utilize it as a USB mic/speaker device for an embedded project with Raspberry Pi. I appreciate your testing efforts. I look forward to future optimizations and fixes.

LHYhorion, 2023-11-17:
> Hey @LHYhorion,
> 
> Thank you for your response. Yes, I was indeed using this with Linux. My intention was to utilize it as a USB mic/speaker device for an embedded project with Raspberry Pi. I appreciate your testing efforts. I look forward to future optimizations and fixes.

Thank you very much for your feedback. Currently, if you would like to experience the functionality of this demo, you can try it on a Windows system. We will address the noise issue when playing music on Linux systems in future updates.

closed 2024-02-03 by zhanzhaocheng

leeebo, 2024-02-04:
@kadharsh Hi the bug has been fixed in https://github.com/espressif/esp-box/commit/fde9930ede5201b27d2cc0a290281f10c5c9047c, it should work on both Linux, Windows and MacOS now.