espressif/esp-box #104: Reset to Factory Settings

Pajservices, 2023-09-27:
I've got a ESP32-S3-BOX-Lite - and have been doing some coding for a project. Unfortunately, I think I screwed it up with a piece of code - and it now isn't working properly.  I have tried to download the firmware using the flash_download tool but get an error as follows;

```
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 115200
No serial data received.
[2023-09-27 15:27:23,434][ESP8266Loader_spi[1]][espDownloader.py][line:750][ERROR]: ESP32S3 Chip stub error esp_stub_and_set_baud.
no log file output ...
```

Do I need to Factory RESET and if so How do I do this?

Thanks Phil

espressif2022, 2023-09-28:
keep press `boot btn`，and short `reset btn`, it'll enter bootloader mode.

ESP-Mars, 2023-09-28:
You can use [ESP-Launchpad](https://github.com/espressif/esp-box/blob/master/docs/firmware_update.md#updating-firmware-with-esp-launchpad-recommended) to flash the factory firmware, please choose the following file:

[image]

If download fails, please hold down the `Boot` button and then press the `Reset` button to initiate Firmware Download mode manually.

Pajservices, 2023-09-28:
Thanks Cao Yan - I appreciate your help.

However - when I try to Connect the Device it is saying it's paired, but it does Connect, then the main screen shows that it is being used. (see images in attachment)

There is nothing that it is paired with. I have rebooted the PC - have checked my phones and anything else with Bluetooth and nothing is paired to it.

How do I clear this?

Phil
 
Regards
 
Phil Jones
PAJ Services
Mob: 07768 746032

 

 <https://user-images.githubusercontent.com/85924095/271176438-ffa7d637-c114-417a-8899-ef56fcb23ad8.png> 

This email has been checked for viruses by AVG antivirus software.
www.avg.com

Pajservices, 2023-09-29:
Re the below, someone suggested I should manually enter Firmware Download mode by holding the box-lite's Boot button and then press the Reset button, then try to reconnect the device and flash firmware again.
 
I have tried this but doesn’t seem to make a difference.
 
Can you confirm for me which button is which – and how long do I hold them down? Do I do this while it is plugged in – and should they be pressed at same time. 
 
I really appreciate any help you can give.
 
Thanks
Phil
 
 
 
 
Thanks Cao Yan - I appreciate your help.

However - when I try to Connect the Device it is saying it's paired, but it does Connect, then the main screen shows that it is being used. (see images in attachment)

There is nothing that it is paired with. I have rebooted the PC - have checked my phones and anything else with Bluetooth and nothing is paired to it.

How do I clear this?

Phil
 
Regards
 
Phil Jones
PAJ Services
Mob: 07768 746032

 

 <https://user-images.githubusercontent.com/85924095/271176438-ffa7d637-c114-417a-8899-ef56fcb23ad8.png>

ESP-Mars, 2023-10-05:
Hi Phil, the boot button is located in the top left corner of the device, and the rst button is located in the bottom left corner. You can refer to this [hardawre overview](https://github.com/espressif/esp-box/blob/master/docs/hardware_overview/esp32_s3_box_lite/hardware_overview_for_lite.md)
Just hood on the boot button then one press the rst boot, please notice one press, which means just press one reset one time, then release the boot (DO NOT pressed at same time). Another same way we usually like used is hold on the boot button then plug and unplug one time the USB-C cable. Hope this can help you, more confusing or questions, you can show the serial log or record a video of the problem phenomenon and upload it to your personal YouTube or other platform.

Pajservices, 2023-10-05:
That’s great thanks – I’ve managed to do it now and reloaded the software.
 
I’ve also got some basic Bluetooth code running ok.
 
 
Could I just ask you something else – with the BOX-Lite which libraries should I use to display something on the screen? Can you point me to any examples of writing to the screen?
 
 
 
Regards
 
Phil Jones
PAJ Services
Mob: 07768 746032

espressif2022, 2023-10-07:
> […](#)
> -- This email has been checked for viruses by AVG antivirus software. [www.avg.com](http://www.avg.com)

You can refer t this bsp, https://github.com/espressif/esp-bsp/tree/master/esp-box-lite
Here is the example, https://github.com/espressif/esp-bsp/tree/master/examples/display

Pajservices, 2023-10-07:
Thank you so much

Pajservices, 2023-10-07:
I’m now trying to get the Display working with one of the examples, but have an issue where the screen is staying black unless I hold the reset button in.
 
I have started an Issue as follows;
 
https://github.com/espressif/esp-box/issues/106
 
 
Regards
 
Phil Jones

 
…

ESP-Mars, 2023-11-09:
Temporarily closed due to the extended period of time with no response. Should you have additional questions or concerns, don't hesitate to reopen the issue.

closed 2023-11-09 by ESP-Mars