espressif/esp-box #106: Problem with Example Code

Pajservices, 2023-10-07:
Trying to get some sample code working on the ESP-S3-Box-Lite using Arduino IDE but having problems.

I'm using the Example called 'Graphic Test' from  https://github.com/SuGlider/Adafruit_ESP32S3_BOX/tree/main/examples.

The code compiles - briefly displays something then goes black.
The serial Monitor echo's everything expected - but nothing on screen.

If I hold the Reset button in the screen slowly comes to life and displays the rounded rectangles in Green - but as soon as I let go of the button - it's black again.

Any help please?

Phil

Code for Reference:
```
`/*
   This sketch targets the ESP32-S3-BOX board ONLY
   It demonstrates a compatible Adafruit GFX Graphics Library benchmark sketch
*/

// Testing if the user has selected the right board in the menu
#ifndef  TT21100_ADDR
#error Please select the right board in the Arduino IDE: ESP32-S3-Box
#endif

#include "ESP32_S3_Box_TFT.h"

ESP32S3BOX_TFT tft = ESP32S3BOX_TFT();

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  delay(500);

  tft.init();
  tft.setRotation(0); // Landscape orientation
  tft.setTextSize(2);

  Serial.println("TFT Initialized.");

  // large block of text
  tft.fillScreen(ST7789_BLACK);
  tft.setTextWrap(true);
  tft.setTextColor(ST7789_WHITE);

  tft.setCursor(0, 0);
  tft.print("Hello ESP32S3-Box");

  tft.setCursor(0, 30);
  tft.print("Chip Model: " + String(ESP.getChipModel()));
  tft.setCursor(0, 50);
  tft.print("rotation: " + String(tft.getRotation()));
  tft.setCursor(0, 70);
  tft.print(String(tft.width()) + " x " + String(tft.height()));

  delay(4000);

  tft.setTextSize(4);
  tft.fillScreen(ST7789_RED);
  tft.setCursor(50, 50);
  tft.print("RED");
  delay(1000);
  tft.fillScreen(ST7789_GREEN);
  tft.setCursor(50, 50);
  tft.print("GREEN");
  delay(1000);
  tft.fillScreen(ST7789_BLUE);
  tft.setCursor(50, 50);
  tft.print("BLUE");
  delay(1000);
[... 297 lines cut ...]
  for (i = 0; i < w; i += 6) {
    i2 = i / 2;
    tft.drawRoundRect(cx - i2, cy - i2, i, i, i / 8, tft.color565(i, 0, 0));
  }

  return micros() - start;
}

unsigned long testFilledRoundRects() {
  unsigned long start;
  int           i, i2,
                cx = tft.width()  / 2 - 1,
                cy = tft.height() / 2 - 1;

  tft.fillScreen(ST7789_BLACK);
  start = micros();
  for (i = min(tft.width(), tft.height()); i > 20; i -= 6) {
    i2 = i / 2;
    tft.fillRoundRect(cx - i2, cy - i2, i, i, i / 8, tft.color565(0, i, 0));
    yield();
  }

  return micros() - start;
}`
```

espressif2022, 2023-10-08:
Hello，I notice this comment:
```
/*
   This sketch targets the ESP32-S3-BOX board ONLY
   It demonstrates a compatible Adafruit GFX Graphics Library benchmark sketch
*/
```
I think this example can't be used for `esp-box-lite`,maybe you can refer to this:https://github.com/esp-arduino-libs/ESP32_Display_Panel,

```
...
// #define ESP_PANEL_BOARD_ESP32_C3_LCDKIT
#define ESP_PANEL_BOARD_ESP32_S3_BOX
// #define ESP_PANEL_BOARD_ESP32_S3_BOX_LITE
...
```

Pajservices, 2023-10-08:
Thank you for that - I have it working and get some graphic colours showing. 

What Graphic commands do I use to draw things - can't find a description of the library it uses ....or can I use another library on top of this? eg LvGl etc.

espressif2022, 2023-10-09:
> Thank you for that - I have it working and get some graphic colours showing.
> 
> What Graphic commands do I use to draw things - can't find a description of the library it uses ....or can I use another library on top of this? eg LvGl etc.

https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/master/examples/LVGL/Porting/README.md
This example is use Graphic  of LVGL.

Pajservices, 2023-10-09:
I’m really struggling with this.
 
I’ve edited the Conf files as instructed and put in the correct places.
 
Try to compile Porting.ino and get the following;
 
xtensa-esp32s3-elf-g++: error: CreateProcess: No such file or directory
Error compiling for board ESP32-S3-Box.
 
All I am trying to do is draw some basic things on the S3-BOX-Lite display; some simple tools like DrawPixel, DrawLine etc...
 
Surely there must be an easy way to do this?
 
 
 
 
Regards
 
Phil Jones
PAJ Services
Mob: 07768 746032

 

https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/master/examples/LVGL/Porting/README.md

This email has been checked for viruses by AVG antivirus software.
www.avg.com

LHYhorion, 2023-10-10:
> […](#)
> -- This email has been checked for viruses by AVG antivirus software. [www.avg.com](http://www.avg.com)

The ESP32_Display_Panel can be used in conjunction with LVGL. For specifics on how to use LVGL for drawing, you can refer to the usage methods of the LVGL library. If you're working within Arduino, you can download the LVGL library from the IDE. After properly configuring the LVGL configuration files, you can use it directly. You can refer to the LVGL example within the ESP32_Display_Panel for guidance.

ESP-Mars, 2023-11-09:
Temporarily closed due to the extended period of time with no response. Should you have additional questions or concerns, don't hesitate to reopen the issue.

closed 2023-11-09 by ESP-Mars