espressif/esp-box #107: chatgpt demo connecting to wifi

kadisi, 2023-10-10:
硬件设备esp-box-32-s3-lite

测试1：[正常情况]
根据 chatgpt demo 烧录 https://github.com/espressif/esp-box/tree/master/examples/chatgpt_demo 完全按照说明文档构建，烧录完成后， CONFIG.INI  文件里输入wifi 名，密码，api key ，保存重启设备，设备正常运行，能够语音交互并能访问chatgpt。

测试2：【异常情况】

由于想用中文交互，所以使用 idf.py menuconfig  修改了一些参数：
ESP Speech Recognition > use wakenet > Select wake words  使用的是nihaoxiaozhi (wn9_nihaoxiaozhi)
ESP Speech Recognition > use multinet > Chinese Speech Commands Model 使用的是 (X) chinese recognition (mn5q8_cn)

因为参数有修改，按照之前的分区表构建，会出现溢出错误，所以重新修改了分区表：
```
nvs,      data,   nvs,    ,    0x4000,
otadata,  data,   ota,    ,    0x2000,
phy_init, data,   phy,    ,    0x1000,
ota_0,    app,    ota_0,  ,     3600K,
uf2,      app,   factory,  ,    1026K,
storage,  data,   spiffs,  ,    2048K,
model,    data,   spiffs,  ,    6020K
```

构建过程中输出信息为

```

Executing action: all (aliases: build)
Running cmake in directory /Users/zhangjie/Projects/src/github.com/espressif/esp-box/examples/chatgpt_demo/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/Users/zhangjie/.espressif/python_env/idf5.2_py3.11_env/bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /Users/zhangjie/Projects/src/github.com/espressif/esp-box/examples/chatgpt_demo"...
-- IDF_TARGET is not set, guessed 'esp32s3' from sdkconfig '/Users/zhangjie/Projects/src/github.com/espressif/esp-box/examples/chatgpt_demo/sdkconfig'
-- Found Git: /usr/bin/git (found version "2.37.1 (Apple Git-137.1)")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/zhangjie/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/zhangjie/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/zhangjie/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
Processing 18 dependencies:
[1/18] chmorgan/esp-file-iterator (1.0.0)
[2/18] chmorgan/esp-libhelix-mp3 (1.0.3)
[3/18] espressif/button (2.5.6)
[4/18] espressif/cmake_utilities (0.5.1)
[5/18] espressif/esp-box (3.0.2)
[6/18] espressif/esp-box-3 (1.1.0)
[7/18] espressif/esp-box-lite (2.0.1)
[8/18] espressif/esp-dsp (1.4.7)
[9/18] espressif/esp-sr (1.3.3)
[10/18] espressif/esp_codec_dev (1.0.3)
[11/18] espressif/esp_lcd_ili9341 (1.0.2)
[12/18] espressif/esp_lcd_touch (1.1.0)
[13/18] espressif/esp_lcd_touch_gt911 (1.1.0)
[14/18] espressif/esp_lcd_touch_tt21100 (1.1.0)
[15/18] espressif/esp_lvgl_port (1.3.0)
[16/18] espressif/openai (0.1.2)
[17/18] idf (5.2.0)
[18/18] lvgl/lvgl (8.3.10)
-- PLATFORM ESP32_S3_BOX_Lite.
-- Project sdkconfig file /Users/zhangjie/Projects/src/github.com/espressif/esp-box/examples/chatgpt_demo/sdkconfig
Loading defaults file /Users/zhangjie/Projects/src/github.com/espressif/esp-box/examples/chatgpt_demo/sdkconfig.defaults...
-- Compiler supported targets: xtensa-esp32s3-elf
-- Looking for sys/types.h
[... 145 lines cut ...]
Wrote 3072 bytes (158 compressed) at 0x00008000 in 0.1 seconds (effective 463.9 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 31...
Wrote 8192 bytes (31 compressed) at 0x0000d000 in 0.1 seconds (effective 649.9 kbit/s)...
Hash of data verified.
Compressed 3584944 bytes to 1910812...
Wrote 3584944 bytes (1910812 compressed) at 0x00010000 in 37.9 seconds (effective 755.9 kbit/s)...
Hash of data verified.
Compressed 2097152 bytes to 582403...
Wrote 2097152 bytes (582403 compressed) at 0x004a1000 in 17.0 seconds (effective 989.0 kbit/s)...
Hash of data verified.
Compressed 4671588 bytes to 3163068...
Wrote 4671588 bytes (3163068 compressed) at 0x006a1000 in 55.3 seconds (effective 676.1 kbit/s)...
Hash of data verified.
Compressed 1050624 bytes to 380119...
Wrote 1050624 bytes (380119 compressed) at 0x003a0000 in 11.0 seconds (effective 765.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

其中  CONFIG.INI 配置文件内容和测试1：里之前的配置完全一致（wifi ，密码，key 是正确的）

点击屏幕按钮restart 后，屏幕一直出现`Connecting to WI-FI` 黑屏界面，并一直屏闪。无法进入正常界面。

leeebo, 2023-10-11:
修改配置以后 app 变大 0x26abb0 bytes，OTA_0 分区需要加大，

```
ota_0,app,ota_0,0x10000,6400K,
```

kadisi, 2023-10-11:
> 修改配置以后 app 变大 0x26abb0 bytes，OTA_0 分区需要加大，
> 
> ```
> ota_0,app,ota_0,0x10000,6400K,
> ```

直接设置成6400K 还是同样的错误现象。

在上面 测试1：[正常情况] 的case 里，构建后各个文件大小如下
```
-rw-r--r--  1 xx  staff  3584944 10 11 23:57 build/chatgpt_demo.bin
-rw-r--r--  1 xx  staff     8192 10 11 23:55 build/ota_data_initial.bin
-rw-r--r--  1 xx  staff     3072 10 11 23:55 build/partition_table/partition-table.bin
-rw-r--r--  1 xx  staff  2468361 10 11 23:55 build/srmodels/srmodels.bin
-rw-r--r--  1 xx  staff  2097152 10 11 23:55 build/storage.bin
-rw-r--r--  1 xx  staff  1050624 10 11 23:54 factory_nvs/build/factory_nvs.bin
```
在测试2 里改了 
```
ESP Speech Recognition > use wakenet > Select wake words 使用的是nihaoxiaozhi (wn9_nihaoxiaozhi)
ESP Speech Recognition > use multinet > Chinese Speech Commands Model 使用的是 (X) chinese recognition (mn5q8_cn)
```
配置后， 重新构建后各个文件大小如下：

```
-rw-r--r--  1 xx  staff  3584944 10 12 00:15 build/chatgpt_demo.bin
-rw-r--r--  1 xx  staff     8192 10 11 23:55 build/ota_data_initial.bin
-rw-r--r--  1 xx  staff     3072 10 11 23:55 build/partition_table/partition-table.bin
-rw-r--r--  1 xx  staff  4671586 10 12 00:14 build/srmodels/srmodels.bin
-rw-r--r--  1 xx  staff  2097152 10 12 00:14 build/storage.bin
-rw-r--r--  1 xx  staff  1050624 10 11 23:54 factory_nvs/build/factory_nvs.bin
```

可见前后比较，变大的只是 `build/srmodels/srmodels.bin ` 变大了有2203225 

所以我又重新修改了分区表配置如下：（调大model 分区大小， 缩小ota_0 分区大小，因为ota_0 最终是烧录 build/chatgpt_demo.bin 的，4096K 大于实际二进制大小）
```
nvs,      data,   nvs,     0x9000,    0x4000,
otadata,  data,   ota,     0xd000,    0x2000,
phy_init, data,   phy,     0xf000,    0x1000,
ota_0,    app,    ota_0,   0x10000,    4096K,
uf2,      app,   factory,  ,    2M,
storage,  data,   spiffs,  ,    2M,
model,    data,   spiffs,  ,   5192K

```

重新构建日志

```

Executing action: all (aliases: build)
Running ninja in directory */github.com/espressif/esp-box/examples/chatgpt_demo/build
Executing "ninja all"...
[0/1] Re-running CMake...
-- Building ESP-IDF components for target esp32s3
Processing 18 dependencies:
[18/18] lvgl/lvgl (8.3.10)
-- PLATFORM ESP32_S3_BOX_Lite.
-- Project sdkconfig file */github.com/espressif/esp-box/examples/chatgpt_demo/sdkconfig
Loading defaults file */github.com/espressif/esp-box/examples/chatgpt_demo/sdkconfig.defaults...
-- Compiler supported targets: xtensa-esp32s3-elf
-- App "chatgpt_demo" version: v0.5.0-20-ge2d289f-dirty
-- Adding linker script */***
-- OPENAI: 0.1.2
-- BUTTON: 2.5.6
-- PLATFORM ESP32_S3_BOX_Lite.
-- Components: app_trace app_update bootloader bootloader_support bsp bt chmorgan__esp-audio-player chmorgan__esp-file-iterator chmorgan__esp-libhelix-mp3 cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_bootloader_format esp_coex esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump espressif__button espressif__cmake_utilities espressif__esp-box espressif__esp-box-3 espressif__esp-box-lite espressif__esp-dsp espressif__esp-sr espressif__esp_codec_dev espressif__esp_lcd_ili9341 espressif__esp_lcd_touch espressif__esp_lcd_touch_gt911 espressif__esp_lcd_touch_tt21100 espressif__esp_lvgl_port espressif__openai esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lvgl__lvgl lwip main mbedtls mqtt newlib nvs_flash nvs_sec_provider openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport touch_element ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: *****

[... 123 lines cut ...]
model,data,spiffs,0x9a0000,5192K,
*******************************************************************************
[1/1] cd /Users/zhangjie/Projects/src/github.com/espressif/esp-box/examples/ch...ub.com/espressif/esp-box/examples/chatgpt_demo/build/bootloader/bootloader.bin
Bootloader binary size 0x58f0 bytes. 0x2710 bytes (31%) free.
[47/48] Generating binary image from built executable
esptool.py v4.7.dev2
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
Generated /Users/zhangjie/Projects/src/github.com/espressif/esp-box/examples/chatgpt_demo/build/chatgpt_demo.bin
[48/48] cd /Users/zhangjie/Projects/src/github.com/espressif/esp-box/examples/.../src/github.com/espressif/esp-box/examples/chatgpt_demo/build/chatgpt_demo.bin
chatgpt_demo.bin binary size 0x36b3b0 bytes. Smallest app partition is 0x390000 bytes. 0x24c50 bytes (4%) free.
Warning: The smallest app partition is nearly full (4% free space left)!

Project build complete. To flash, run:
 idf.py flash
or
 idf.py -p PORT flash
or
 python -m esptool --chip esp32s3 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_size 16MB --flash_freq 80m 0x0 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0xd000 build/ota_data_initial.bin 0x410000 build/chatgpt_demo.bin 0x7a0000 build/storage.bin 0x9a0000 build/srmodels/srmodels.bin 

```

这么设置后，
warning 变成了， `Warning: The smallest app partition is nearly full (4% free space left)!`, 重新烧录， 重启， 还是同样的错误现象 ~~~~~~

LHYhorion, 2023-10-13:
> Chinese Speech Commands Model

如果想尝试中文唤醒，在 menuconfig 中改完唤醒词之后，在 app_sr 代码中也需要更改，更改后就不会出现崩溃问题了。

kadisi, 2023-10-13:
> > Chinese Speech Commands Model
> 
> 如果想尝试中文唤醒，在 menuconfig 中改完唤醒词之后，在 app_sr 代码中也需要更改，更改后就不会出现崩溃问题了。

请问有相关说明吗？

kadisi, 2023-10-15:
> > Chinese Speech Commands Model
> 
> 如果想尝试中文唤醒，在 menuconfig 中改完唤醒词之后，在 app_sr 代码中也需要更改，更改后就不会出现崩溃问题了。

请问具体怎么改呢？谢谢

kadisi, 2023-10-15:
> python -m esptool -p /dev/cu.usbmodem14201  --chip esp32s3 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_size 16MB --flash_freq 80m 0x0 [...]

使用以下分区
```
nvs,      data,   nvs,     0x9000,    0x4000,
otadata,  data,   ota,     0xd000,    0x2000,
phy_init, data,   phy,     0xf000,    0x1000,
ota_0,    app,    ota_0,   0x10000,    4096K,
uf2,      app,   factory,  ,    2M,
storage,  data,   spiffs,  ,    2M,
model,    data,   spiffs,  ,   5192K

```

app_sr.c 代码修改：

```
diff --git a/examples/chatgpt_demo/main/app/app_sr.c b/examples/chatgpt_demo/main/app/app_sr.c
index ba4cdbe..2e92b9e 100644
--- a/examples/chatgpt_demo/main/app/app_sr.c
+++ b/examples/chatgpt_demo/main/app/app_sr.c
@@ -166,7 +166,8 @@ esp_err_t app_sr_set_language(sr_language_t new_lang)
     } else {
         g_sr_data->lang = new_lang;
     }
-    ESP_LOGI(TAG, "Set language %s", SR_LANG_EN == g_sr_data->lang ? "EN" : "CN");
+    //ESP_LOGI(TAG, "Set language %s", SR_LANG_EN == g_sr_data->lang ? "EN" : "CN");
+    ESP_LOGI(TAG, "Set language %s", SR_LANG_CN == g_sr_data->lang ? "EN" : "CN");
     if (g_sr_data->model_data) {
         g_sr_data->multinet->destroy(g_sr_data->model_data);
     }
@@ -202,8 +203,10 @@ esp_err_t app_sr_start(bool record_en)
     g_sr_data->afe_handle = afe_handle;
     g_sr_data->afe_data = afe_data;

-    g_sr_data->lang = SR_LANG_MAX;
-    ret = app_sr_set_language(SR_LANG_EN);
+    //g_sr_data->lang = SR_LANG_MAX;
+    g_sr_data->lang = SR_LANG_CN;
+    //ret = app_sr_set_language(SR_LANG_EN);
+    ret = app_sr_set_language(SR_LANG_CN);
     ESP_GOTO_ON_FALSE(ESP_OK == ret, ESP_FAIL, err, TAG,  "Failed to set language");

     ret_val = xTaskCreatePinnedToCore(&audio_feed_task, "Feed Task", 8 * 1024, (void *)afe_data, 5, &g_sr_data->feed_task, 0);
diff --git a/examples/chatgpt_demo/partitions.csv b/examples/chatgpt_demo/partitions.csv
index 5273940..ef51795 100644
```

能够烧录成功，系统也可以正常进入开机画面，但是还是说的是英文。

LHYhorion, 2023-10-15:
> > python -m esptool -p /dev/cu.usbmodem14201  --chip esp32s3 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_size 16MB --flash_freq 80m 0x0 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0xd000 build/ota_data_initial.bin 0x10000 build/chatgpt_demo.bin 0x610000 build/storage.bin 0x810000 build/srmodels/srmodels.bin 0x410000 factory_nvs/build/factory_nvs.bin
> 
> 使用以下分区
> 
> ```
> nvs,      data,   nvs,     0x9000,    0x4000,
> otadata,  data,   ota,     0xd000,    0x2000,
> phy_init, data,   phy,     0xf000,    0x1000,
> ota_0,    app,    ota_0,   0x10000,    4096K,
> uf2,      app,   factory,  ,    2M,
> storage,  data,   spiffs,  ,    2M,
> model,    data,   spiffs,  ,   5192K
> ```
> 
> app_sr.c 代码修改：
> 
> ```
> diff --git a/examples/chatgpt_demo/main/app/app_sr.c b/examples/chatgpt_demo/main/app/app_sr.c
> index ba4cdbe..44d5e32 100644
> --- a/examples/chatgpt_demo/main/app/app_sr.c
> +++ b/examples/chatgpt_demo/main/app/app_sr.c
> @@ -166,7 +166,8 @@ esp_err_t app_sr_set_language(sr_language_t new_lang)
>      } else {
>          g_sr_data->lang = new_lang;
>      }
> -    ESP_LOGI(TAG, "Set language %s", SR_LANG_EN == g_sr_data->lang ? "EN" : "CN");
> +    //ESP_LOGI(TAG, "Set language %s", SR_LANG_EN == g_sr_data->lang ? "EN" : "CN");
> +    ESP_LOGI(TAG, "Set language %s", SR_LANG_CN == g_sr_data->lang ? "CN" : "EN");
>      if (g_sr_data->model_data) {
>          g_sr_data->multinet->destroy(g_sr_data->model_data);
>      }
> @@ -202,8 +203,10 @@ esp_err_t app_sr_start(bool record_en)
>      g_sr_data->afe_handle = afe_handle;
>      g_sr_data->afe_data = afe_data;
> 
> -    g_sr_data->lang = SR_LANG_MAX;
> -    ret = app_sr_set_language(SR_LANG_EN);
> +    //g_sr_data->lang = SR_LANG_MAX;
> +    g_sr_data->lang = SR_LANG_CN;
> +    //ret = app_sr_set_language(SR_LANG_EN);
> +    ret = app_sr_set_language(SR_LANG_CN);
>      ESP_GOTO_ON_FALSE(ESP_OK == ret, ESP_FAIL, err, TAG,  "Failed to set language");
> 
>      ret_val = xTaskCreatePinnedToCore(&audio_feed_task, "Feed Task", 8 * 1024, (void *)afe_data, 5, &g_sr_data->feed_task, 0);
> ```
> 
> 能够烧录成功，系统也可以正常进入开机画面，基本上成功了。
> 
> 不过还有瑕疵：比如： user：nihaoxiaozhi esp: Thinking ... user：太阳为什么从东边升起？ esp: iam think about your question , please wati for a monent: esp：why did the sun rise from east? esp: **** 英文回答，英文文字
> 
> 还是没达到中文回答，中文文字的目的。

在 app_sr 中更改代码 ret = app_sr_set_language(SR_LANG_CN); 并且 menuconfig 中设置 Chinese Speech Commands Model (chinese recognition (mn5q8_cn))，English Speech Commands Model (None)  即可达到中文回答，中文文字显示的目的。
user:今天星期几
esp:今天是星期日
user:你好
esp:有什么我可以帮助你的吗
P.S：有时会出现识别为英文或转换为英文的情况，可多测试几次进行验证

kadisi, 2023-10-15:
> 在 app_sr 中更改代码 ret = app_sr_set_language(SR_LANG_CN); 并且 menuconfig 中设置 Chinese Speech Commands Model (chinese recognition (mn5q8_cn))，English Speech Commands Model (None) 即可达到中文回答，中文文字显示的目的。 [...]

感谢，已经验证成功。 不过确实会出现您说的`有时会出现识别为英文或转换为英文的情况` 这种情况，而且出现的概率比较大，请问这是什么原因？

LHYhorion, 2023-10-16:
> 感谢，已经验证成功。 不过确实会出现您说的`有时会出现识别为英文或转换为英文的情况` 这种情况，而且出现的概率比较大，请问这是什么原因？

该 demo 默认语言设置为英文。若要使用中文，请尝试修改此[代码行](https://github.com/espressif/esp-box/blob/e2bb387d67145d3935842e362f1f44d88692acba/examples/chatgpt_demo/main/main.c#L42C5-L42C66) audioTranscription->setLanguage(audioTranscription,"zh"); 并进行测试。但是因为我们无法控制 chatgpt 的回应，所以也不能保证转换为中文的效果很好。

kadisi, 2023-10-16:
@LHYhorion 能否出一个针对于中文版的demo, 毕竟chatgpt 也是可以中文提问， 中文回答的。

LHYhorion, 2023-10-16:
> @LHYhorion 能否出一个针对于中文版的demo, 毕竟chatgpt 也是可以中文提问， 中文回答的。

很抱歉应该不会出中文版的 demo，因为这个 demo 主要还是面向全世界的开发者，需要中文问答还是要自行按指南修改语言配置，之后会完善文档指导。

kadisi, 2023-10-16:
相关文件修改如下：

```
diff --git a/examples/chatgpt_demo/main/app/app_sr.c b/examples/chatgpt_demo/main/app/app_sr.c
index ba4cdbe..6c90b27 100644
--- a/examples/chatgpt_demo/main/app/app_sr.c
+++ b/examples/chatgpt_demo/main/app/app_sr.c
@@ -202,8 +202,10 @@ esp_err_t app_sr_start(bool record_en)
     g_sr_data->afe_handle = afe_handle;
     g_sr_data->afe_data = afe_data;

-    g_sr_data->lang = SR_LANG_MAX;
-    ret = app_sr_set_language(SR_LANG_EN);
+    //g_sr_data->lang = SR_LANG_MAX;
+    g_sr_data->lang = SR_LANG_CN;
+    //ret = app_sr_set_language(SR_LANG_EN);
+    ret = app_sr_set_language(SR_LANG_CN);

diff --git a/examples/chatgpt_demo/main/main.c b/examples/chatgpt_demo/main/main.c
index 71ecfc4..b0e14c8 100644
--- a/examples/chatgpt_demo/main/main.c
+++ b/examples/chatgpt_demo/main/main.c
@@ -39,7 +39,7 @@ esp_err_t start_openai(uint8_t *audio, int audio_len)
         chatCompletion = openai->chatCreate(openai);

         audioTranscription->setResponseFormat(audioTranscription, OPENAI_AUDIO_RESPONSE_FORMAT_JSON);
-        audioTranscription->setLanguage(audioTranscription,"en");
+        audioTranscription->setLanguage(audioTranscription,"zh");
         audioTranscription->setTemperature(audioTranscription, 0.2);

diff --git a/examples/chatgpt_demo/partitions.csv b/examples/chatgpt_demo/partitions.csv
index 5273940..ef51795 100644
--- a/examples/chatgpt_demo/partitions.csv
+++ b/examples/chatgpt_demo/partitions.csv
@@ -3,7 +3,7 @@
 nvs,      data,   nvs,     0x9000,    0x4000,
 otadata,  data,   ota,     0xd000,    0x2000,
 phy_init, data,   phy,     0xf000,    0x1000,
-ota_0,    app,    ota_0,   0x10000,     6M,
-uf2,      app,   factory,  0x700000,    2M,
-storage,  data,   spiffs,  0x900000,    2M,
-model,    data,   spiffs,  0xb00000,   4000K
\ No newline at end of file
+ota_0,    app,    ota_0,   0x10000,    4096K,
+uf2,      app,   factory,  ,    2M,
+storage,  data,   spiffs,  ,    2M,
+model,    data,   spiffs,  ,   5192K

```

 idf.py menuconfig 修改了一些参数：
ESP Speech Recognition > use wakenet > Select wake words 使用的是nihaoxiaozhi (wn9_nihaoxiaozhi)
ESP Speech Recognition > use multinet > Chinese Speech Commands Model 使用的是 (X) chinese recognition (mn5q8_cn)
ESP Speech Recognition > use multinet >English Speech Commands Model (None)

烧录命令：
```
python -m esptool -p /dev/cu.usbmodem14201  --chip esp32s3 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_size 16MB --flash_freq 80m 0x0 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0xd000 build/ota_data_initial.bin 0x10000 build/chatgpt_demo.bin 0x610000 build/storage.bin 0x810000 build/srmodels/srmodels.bin 0x410000 factory_nvs/build/factory_nvs.bin

```

基本上中文请求，中文问答。问题已经解决。

非常感谢 @LHYhorion 的帮助。

closed 2023-10-16 by kadisi

lomon-emerson, 2023-12-09:
CONFIG.INI 路径在哪？谢谢

leeebo, 2023-12-11:
> CONFIG.INI 路径在哪？谢谢

将 ESP-BOX 通过 USB 接入电脑，会显示一个 U 盘，里面包含了 `CONFIG.INI` 文件