espressif/esp-box #108: build fails with Box-Lite

kingaza, 2023-10-11:
error msg from stderr_output

CMake Error at D:/04_Dev/Espressif/frameworks/esp-idf-v5.0.2/tools/cmake/build.cmake:519 (message):
  ERROR: The hash sum of the component "espressif/button(2.5.6)" does not
  match the one recorded in your dependencies.lock file.  This could be due
  to a potential spoofing of the download server, or your lock file may have
  become corrupted.  Please review the lock file and verify the download
  server's authenticity to ensure the component's security and integrity.

Call Stack (most recent call first):
  D:/04_Dev/Espressif/frameworks/esp-idf-v5.0.2/tools/cmake/project.cmake:447 (idf_build_process)
  CMakeLists.txt:11 (project)

ninja: error: rebuilding 'build.ninja': subcommand failed

LHYhorion, 2023-10-11:
> error msg from stderr_output
>
> CMake Error at D:/04_Dev/Espressif/frameworks/esp-idf-v5.0.2/tools/cmake/build.cmake:519 (message): ERROR: The hash sum of the component "espressif/button(2.5.6)" does [...]

You can try deleting the dependencies.lock and then recompile.

kingaza, 2023-10-12:
done
thx @LHYhorion

closed 2023-10-12 by kingaza