espressif/esp-box #109: usb_headset example on esp-s3-box-3

mjhewitt1, 2023-10-17:
I am wanting to use the usb_headset example to record audio on the esp-s3-box-3. I have flashed the example code and set the board to the esp-s3-box-3, however only some distorted noise is recorded and other sounds (e.g. speech) are not picked up. The microphones work in other scenarios (e.g. wakeword detection, speech recognition). Any ideas how to fix the issue?
Many thanks

LHYhorion, 2023-10-18:
> I am wanting to use the usb_headset example to record audio on the esp-s3-box-3. I have flashed the example code and set the board to the esp-s3-box-3, however only some distorted noise is recorded [...]

After testing, it has been proven that the demo can be used as a microphone for normal recording. If the burning is normal and recording is still not working, please check if another device has been selected as the recording device.

mjhewitt1, 2023-10-26:
Hi, I have tried a few things and it still does not work. The device is recognised as an input device, therefore I can select 'ESP-BOX Headset' as the microphone on my PC however when I test it there is no audio captured. 

I have tried recording on a Windows and Ubuntu PC. I have also uploading the example firmware via the ESP web flashing tool, as well as locally via ESP-IDF. The example works for the ESP-BOX's, but not the ESP-BOX-3. I have made sure to select the correct board in menuconfig when flashing the firmware. 

I can't think of why it is not working as expected. Could you provide any other advice, please? Thanks :)

espressif2022, 2023-10-27:
> Hi, I have tried a few things and it still does not work. The device is recognised as an input device, therefore I can select 'ESP-BOX Headset' as the microphone on my PC however when I test it [...]

did you try the fw from launchpad? 
https://espressif.github.io/esp-launchpad/?flashConfigURL=https://raw.githubusercontent.com/espressif/esp-box/master/launch.toml

mjhewitt1, 2023-10-27:
Yes, I have done. It's the same outcome where the mic does not capture any audio.

ESP-Mars, 2023-10-27:
I have tested the demo and also recorded distorted noice, please listen to my test audio and confirm if your recording output is similar to mine?
https://cowtransfer.com/s/2150ed72e03642 click link to check [ USB_headset_input_testvoice.m4a ] or access cowtransfer.com and input extract code: jahjhe to check;

We will work on fixing this bug.

espressif2022, 2023-10-27:
We tested several boards and they recorded effectively. We want to confirm that the firmware you set or downloaded from the web page is definitely esp-box-3.
In addition, there is no sound after you record it, or it is all noise. Can you share your audio recording?

what's your recording tool? I used window's audacity for testing.

ESP-Mars, 2023-12-22:
Temporarily closed due to the extended period of time with no response. Should you have additional questions or concerns, don't hesitate to reopen the issue.

closed 2023-12-22 by ESP-Mars