espressif/esp-box #111: esp-sr离线语音识别卡死问题

worth-moon, 2023-11-13:
Dear Espressif, I encountered a situation where the offline voice recognition stuck and did not trigger the watchdog restart when I modified the following routine

> https://github.com/espressif/esp-box/tree/master/examples/watering_demo

### The development environment is as follows
1. Hardware: ESP32-S3-BOX-Lite
2. ESP-IDF: esp-idf- v5.1.1
3. Programming Environment: vscode

### app_main
```
void app_main(void)
{
    bsp_board_init();
    bsp_display_start();
    app_sr_start();
}
```

### All logs are as follows
```
PS E:\ESP-IDF_Project\new_10.27\sr_deve\desktop_station\applications\independent_sr> set IDF_PATH=E:/Espressif/frameworks/esp-idf-v5.1.1/
PS E:\ESP-IDF_Project\new_10.27\sr_deve\desktop_station\applications\independent_sr> E:/Espressif/python_env/idf5.1_py3.11_env/Scripts/python.exe E:\Espressif\frameworks\esp-idf-v5.1.1\tools\idf_monitor.py -p COM11 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 e:\ESP-IDF_Project\new_10.27\sr_deve\desktop_station\applications\independent_sr\build\independent_sr.elf
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM11 instead...
--- idf_monitor on \\.\COM11 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x2 (DOWNLOAD(USB/UART0))
Saved PC:0x40041a79
0x40041a79: ets_delay_us in ROM

waiting for download
None
Waiting for the device to reconnect.
I (280) esp_image: segment 3: paddr=000ea0e0 vaddr=3fc9f6I (921) button: IoT Button Version: 2.5.6
I (921) adc button: ADC1 has been initialized
I (921) adc button: calibration scheme version is Curve Fitting
I (921) adc button: Calibration Success
I (921) button: IoT Button Version: 2.5.6
I (921) button: IoT Button Version: 2.5.6
I (921) MODEL_LOADER: The storage free size is 23744 KB
I (921) MODEL_LOADER: The partition size is 4300 KB
I (921) MODEL_LOADER: Successfully map model partition
I (921) app_sr: load wakenet:wn9_hilexin
I (921) AFE_SR: afe interface for speech recognition

I (921) AFE_SR: AFE version: SR_V220727

I (921) AFE_SR: Initial auido front-end, total channel: 3, mic num: 2, ref num: 1

I (921) AFE_SR: aec_init: 0, se_init: 1, vad_init: 1

I (921) AFE_SR: wakenet_init: 1

MC Quantized wakenet9: wakenet9_v1h24_hilexin_3_0.608_0.615, tigger:v3, mode:2, p:0, (Jul  7 2023 11:10:53)
I (1131) AFE_SR: wake num: 3, mode: 0, (Jul  7 2023 11:10:53)

Quantized8 Multinet5: MN5Q8_v2_chinese_8_0.9_0.90, beam search:v2, (Jul  7 2023 11:37:23)
I (1341) app_sr: load multinet:mn5q8_cn
I (1341) MN_COMMAND: ---------------------SPEECH COMMANDS---------------------
I (1341) MN_COMMAND: Command ID5, phrase ID0: ce shi zhi ling
I (1341) MN_COMMAND: Command ID0, phrase ID1: da kai kong tiao
I (1341) MN_COMMAND: Command ID1, phrase ID2: guan bi kong tiao
I (1341) MN_COMMAND: Command ID2, phrase ID3: zhi leng mo shi
I (1341) MN_COMMAND: Command ID3, phrase ID4: shang xia mo shi
I (1341) MN_COMMAND: ---------------------------------------------------------

phrase id:0, command id:5, str:c e sh i zh i l ing
[... 418 lines cut ...]
    afe_handle->destroy(afe_data);
    vTaskDelete(NULL);
}
```

经测试，程序卡死时间几乎没有上限（笔者测试，十分钟仍无新的日志输出，没有硬件卡死问题）

5. 添加指令时需要注意两点

- 调用以下函数添加语音指令到模型的时候，需要注意，两个参数必须存储在一个结构体中，并为结构体分配动态内存
esp_err_t esp_mn_commands_add(int command_id, char *string)
- 结构体需具备链表的指针域，即有一个成员变量用于连接链表中的其它元素
参考结构体
```

struct test_mn
{
    int id; //语音识别指令id号，不可重复    
    char* phoneme;  //语音指令的拼音字符串 
    struct sr_command_t* pnext; //链表指针
};
```

经测试，添加指令后，程序卡死仅是有几率出现，请多加尝试笔者自行添加五个语句，只有一个语句卡死，当出现该行为起，后续无论程序做出其它任何改变，卡死总会出现（当说出该语音指令后，修改语音指令无效）
希望得到回复，感谢！

espressif2022, 2023-11-13:
> 请多加尝试笔者自行添加五个语句，只有一个语句卡死，当出现该行为起，后续无论程序做出其它任何改变，卡死总会出现（当说出该语音指令后，修改语音指令无效）

是否是固定某个命令卡死，另外，使用的 SR 版本是？

worth-moon, 2023-11-13:
> > 请多加尝试笔者自行添加五个语句，只有一个语句卡死，当出现该行为起，后续无论程序做出其它任何改变，卡死总会出现（当说出该语音指令后，修改语音指令无效）
> 
> 是否是固定某个命令卡死，另外，使用的 SR 版本是？

不是固定命令，修改命令词，同样会卡死，sr的版本是1.3.3.

espressif2022, 2023-11-13:
> > > 请多加尝试笔者自行添加五个语句，只有一个语句卡死，当出现该行为起，后续无论程序做出其它任何改变，卡死总会出现（当说出该语音指令后，修改语音指令无效）
> > 
> > 
> > 是否是固定某个命令卡死，另外，使用的 SR 版本是？
> 
> 不是固定命令，修改命令词，同样会卡死，sr的版本是1.3.3.

更先更新到 1.4.2 版本看下，1.3.3 可能是有问题的，中文模型那

worth-moon, 2023-11-13:
> > > > 请多加尝试笔者自行添加五个语句，只有一个语句卡死，当出现该行为起，后续无论程序做出其它任何改变，卡死总会出现（当说出该语音指令后，修改语音指令无效）
> > > 
> > > 
> > > 是否是固定某个命令卡死，另外，使用的 SR 版本是？
> > 
> > 
> > 不是固定命令，修改命令词，同样会卡死，sr的版本是1.3.3.
> 
> 更先更新到 1.4.2 版本看下，1.3.3 可能是有问题的，中文模型那 另外，你是消息识别的消息那，有没有对端的处理？

为排查干扰，后续处理都注释了，我先更新到1.4.2试试吧

worth-moon, 2023-11-13:
> > > > 请多加尝试笔者自行添加五个语句，只有一个语句卡死，当出现该行为起，后续无论程序做出其它任何改变，卡死总会出现（当说出该语音指令后，修改语音指令无效）
> > > 
> > > 
> > > 是否是固定某个命令卡死，另外，使用的 SR 版本是？
> > 
> > 
> > 不是固定命令，修改命令词，同样会卡死，sr的版本是1.3.3.
> 
> 更先更新到 1.4.2 版本看下，1.3.3 可能是有问题的，中文模型那 另外，你是消息识别的消息那，有没有对端的处理？

[image]
如上图，esp-sr更新后，没有比较明显的改动，另外我使用的语音识别模型是mn5q8_cn

worth-moon, 2023-11-13:
> > > > > 请多加尝试笔者自行添加五个语句，只有一个语句卡死，当出现该行为起，后续无论程序做出其它任何改变，卡死总会出现（当说出该语音指令后，修改语音指令无效）
> > > > 
> > > > 
> > > > 是否是固定某个命令卡死，另外，使用的 SR 版本是？
> > > 
> > > 
> > > 不是固定命令，修改命令词，同样会卡死，sr的版本是1.3.3.
> > 
> > 
> > 更先更新到 1.4.2 版本看下，1.3.3 可能是有问题的，中文模型那 另外，你是消息识别的消息那，有没有对端的处理？
> 
> [image] 如上图，esp-sr更新后，没有比较明显的改动，另外我使用的语音识别模型是mn5q8_cn
发现esp-sr最新版本到了1.5.0，刚发现似乎这个问题已经解决了，待我多加测试，晚些再回复

espressif2022, 2023-11-13:
> > > > > 请多加尝试笔者自行添加五个语句，只有一个语句卡死，当出现该行为起，后续无论程序做出其它任何改变，卡死总会出现（当说出该语音指令后，修改语音指令无效）
> > > > 
> > > > 
> > > > 是否是固定某个命令卡死，另外，使用的 SR 版本是？
> > > 
> > > 
> > > 不是固定命令，修改命令词，同样会卡死，sr的版本是1.3.3.
> > 
> > 
> > 更先更新到 1.4.2 版本看下，1.3.3 可能是有问题的，中文模型那 另外，你是消息识别的消息那，有没有对端的处理？
> 
> [image] 如上图，esp-sr更新后，没有比较明显的改动，另外我使用的语音识别模型是mn5q8_cn

模型应该是 OK 的，之前中文 5.8 版本有问题，已经修复了。如果方便，可以提供下这个最小 demo，我们验证下。

ESP-Mars, 2023-12-22:
Temporarily closed due to the extended period of time with no response. Should you have additional questions or concerns, don't hesitate to reopen the issue.

closed 2023-12-22 by ESP-Mars