espressif/esp-box #116: Matter-switch example aborts on ESP32-S3-BOX-Lite

oidebrett, 2023-11-25:
After flashing the matter-switch example to the ESP32-S3-Box-Lite the code aborts and goes into a loop of re-initializing and aborting. The errors seem to be that the "Touch not found"

```
I (2476) ili9341: LCD panel create success, version: 1.2.0
W (2596) ili9341: The 36h command has been used and will be overwritten by external initialization sequence
W (2596) ili9341: The 3Ah command has been used and will be overwritten by external initialization sequence
E (2600) ESP-BOX-3: Touch not found
ESP_ERROR_CHECK failed: esp_err_t 0x105 at 0x42035e1e
0x42035e1e: bsp_display_indev_init at /home/ivob/Projects/esp-box/examples/matter_switch/managed_components/espressif__esp-box-3/esp-box-3.c:541
 (inlined by) bsp_display_start_with_config at /home/ivob/Projects/esp-box/examples/matter_switch/managed_components/espressif__esp-box-3/esp-box-3.c:570

file: "./managed_components/espressif__esp-box-3/esp-box-3.c" line 541
func: bsp_display_indev_init
expression: bsp_touch_new(((void *)0), &tp)

abort() was called at PC 0x40387be8 on core 0

```

oidebrett, 2023-11-25:
I hadnt set the correct board. This is how I fixed it

`
idf.py menuconfig
HMI Board Config  ---> BSP board ESP32-S3-BOX-LITE
`

closed 2023-11-25 by oidebrett

espressif2022, 2023-11-27:
> I hadnt set the correct board. This is how I fixed it
> 
> ` idf.py menuconfig
> 
> HMI Board Config ---> BSP board ESP32-S3-BOX-LITE `

yes, we should select the board by meuconfig.