espressif/esp-box #117: partitions.csv的改动

ZorkyZorky, 2023-11-28:
环境：离线包ESP-IDF5.1.1
板子：旧版ESP32-S3-BOX
步骤：按照README.md进行的 

[image]
默认配置会报错，overlap
[image]

[image]
修改后报错消失，但是会一致重启
[image]

partitions.csv该怎么修改？

LHYhorion, 2023-11-28:
> 环境：离线包ESP-IDF5.1.1、ESP-BOX-V0.5.0 板子：旧版ESP32-S3-BOX 步骤：按照README.md进行的
> 
> ![image](https://private-user-images.githubusercontent.com/64697283/286123422-58e8c58b-a177-4ef7-a92f-9d02377c9129.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDExNjE0OTgsIm5iZiI6MTcwMTE2MTE5OCwicGF0aCI6Ii82NDY5NzI4My8yODYxMjM0MjItNThlOGM1OGItYTE3Ny00ZWY3LWE5MmYtOWQwMjM3N2M5MTI5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzExMjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMTI4VDA4NDYzOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBkMjhjOGEzMTRhODcyMDM5ODg3ZTYyODllZjlkNWRmZDZmMzQyMjIyN2IwZjI4MzUxY2U0N2M2NWM4Yjk5ZmEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.03gQNYmzYlgg0bB5O33VdZfPo539BS6_75DI2KqkISY) 默认配置会报错，overlap ![image](https://private-user-images.githubusercontent.com/64697283/286123495-52cf8aee-d4fe-48ef-8c8d-d05ea02525c4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDExNjE0OTgsIm5iZiI6MTcwMTE2MTE5OCwicGF0aCI6Ii82NDY5NzI4My8yODYxMjM0OTUtNTJjZjhhZWUtZDRmZS00OGVmLThjOGQtZDA1ZWEwMjUyNWM0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzExMjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMTI4VDA4NDYzOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTVlYzdhYmEzMGJkYmY3YzBlY2QwZWJmNmY3ZGRhMzEwY2E3Zjc3Y2I3NmZhMTI1MTMxN2EwMzk0ZTM4MWE5MTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.s6FJkVCIIwl2Drv7dWIBhU2jchamqaNU3X-TZ7Ndt5E)
> 
> ![image](https://private-user-images.githubusercontent.com/64697283/286123732-4f80583c-f567-4124-9763-e943c457a56d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDExNjE0OTgsIm5iZiI6MTcwMTE2MTE5OCwicGF0aCI6Ii82NDY5NzI4My8yODYxMjM3MzItNGY4MDU4M2MtZjU2Ny00MTI0LTk3NjMtZTk0M2M0NTdhNTZkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzExMjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMTI4VDA4NDYzOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTYyODFhZjM2NGRmY2UwODczZmE3OWRlMWFjNzM3NGFhZWE5ODZhODlhNDJlYjEzYzhkMWIwOGQwNzdlODhkZWQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.jakUpRgkIEhHwt-rOZfmTP_HawaMgJKVb_H8VJocqnY) 修改后报错消失，但是会一致重启 ![image](https://private-user-images.githubusercontent.com/64697283/286123896-0d8a4985-54a3-4c60-8506-a0a0ee26525e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDExNjE0OTgsIm5iZiI6MTcwMTE2MTE5OCwicGF0aCI6Ii82NDY5NzI4My8yODYxMjM4OTYtMGQ4YTQ5ODUtNTRhMy00YzYwLTg1MDYtYTBhMGVlMjY1MjVlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzExMjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMTI4VDA4NDYzOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWIyMGU1OGE3YTQ0OTBkNzNhNTgzNDc3YTk3YTg5ZmVkYWM0MTg0NzFlNmQ2MTE2ZjMyYTc0OTUwYmI1Njk0MzMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.plAVHacwWmSAlXi9p7AuvLYOoqGadSY0ojYxd5ZOaKk)
> 
> partitions.csv该怎么修改？

请问您的工程是 chatgpt_demo 吗？如果是 chatgpt_demo，或许您可以尝试将 factory_nvs 中的分区表同步更改保证一致，然后按照 README 中步骤进行测试。

ZorkyZorky, 2023-11-28:
是的，chatgpt_demo。修改一致了，两个patitions.csv都还是同样的问题

ZorkyZorky, 2023-11-29:
[image]
烧录的地址与分区表不太一致，是因为这个原因吗

ZorkyZorky, 2023-11-30:
但是正常我是使用idf.py -p COMX flash，并没有指定烧录地址，应该不是这个原因吧？

espressif2022, 2023-11-30:
> 但是正常我是使用idf.py -p COMX flash，并没有指定烧录地址，应该不是这个原因吧？

这个 demo 是个 复合工程，不能用 flash 直接烧录，这样烧录只有 主目录下的 flash_args 信息，参考 README， 每个地址列出来烧录

ZorkyZorky, 2023-11-30:
可以了，竟然不是开箱即用的，README.md与partition.csv的地址不一样。

另外，开始运行后一致打印这个，我查了下有可能是空指针，但是没有给追溯位置，这个该怎么办？
[image]

espressif2022, 2023-12-01:
> 可以了，竟然不是开箱即用的，README.md与partition.csv的地址不一样。
>
> 另外，开始运行后一致打印这个，我查了下有可能是空指针，但是没有给追溯位置，这个该怎么办？ [...]

原始工程应该配置和 README 应该是一致的，前两天我有试过.

我看你这是改了 wakenet 唤醒词，我一会试下，是不是 srmodel 是不是没烧写成功？https://github.com/espressif/esp-box/blob/10d166a1d4db6e9ae459b65acf991408baec6eac/examples/chatgpt_demo/main/app/app_sr.c#L173

espressif2022, 2023-12-04:
> 可以了。按照README.md默认地址烧录。我是因为默认的flash，里面有OTA，老是不成功
> 
> 但是现在我用python脚本能跑的api key，box上提示not valid ![image](https://private-user-images.githubusercontent.com/64697283/287422650-472e0b63-ce4a-4dc1-bff0-5603da5dde74.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE2NjEzMzIsIm5iZiI6MTcwMTY2MTAzMiwicGF0aCI6Ii82NDY5NzI4My8yODc0MjI2NTAtNDcyZTBiNjMtY2U0YS00ZGMxLWJmZjAtNTYwM2RhNWRkZTc0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjA0VDAzMzcxMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg2OTExYTE5ZjYxNmJlNGI1NWE3Yzk1Y2MyYWExMWVhNzg1NGQwMmEzYjQ4NzMwNDM2N2NiZmE5ODliNTUwODUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.bK8QRVtS0gZqu7aY3vKJ-0ByRI4JtOO_DqLlKfRI8gs)

@alibukharai ，please take a look, thanks!

espressif2022, 2023-12-22:
> 可以了。按照README.md默认地址烧录。我是因为默认的flash，里面有OTA，老是不成功
>
> 但是现在我用python脚本能跑的api key，box上提示not valid [...]

这边内部已更新，马上会退出更新版本，我看了下 你的 log， 这边应该是你国内网络问题，无法打开 openAI 的 url。
后续版本有针对错误分类显示，之前所有报错均显示 key is invalid

espressif2022, 2024-01-18:
你好，新版本已经释放，我将关闭这个ISSUE，如果还有问题，可参考 https://github.com/espressif/esp-box/issues/123

closed 2024-01-18 by espressif2022