espressif/esp-box #119: [esp32-s3-box]OpenAI: Invalid URL (POST /v1/completionsaudio/transcriptions)

lomon-emerson, 2023-12-09:
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x42015c22
0x42015c22: WebRtcNs_AnalyzeCore at /home/sunxiangyu/workspace/esp_sr_lib/build/../components/esp_audio_processor/noise_suppression/webrtc_ns.c:1768

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x18dc
load:0x403c9700,len:0x4
load:0x403c9704,len:0xda4
load:0x403cc700,len:0x3084
entry 0x403c9934
I (26) boot: ESP-IDF v5.1.2 2nd stage bootloader
I (27) boot: compile time Dec  9 2023 12:08:25
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.1
I (34) qio_mode: Enabling default flash chip QIO
I (39) boot.esp32s3: Boot SPI Speed : 80MHz
I (44) boot.esp32s3: SPI Mode       : QIO
I (48) boot.esp32s3: SPI Flash Size : 16MB
I (53) boot: Enabling RNG early entropy source...
W (24841) app_sr: silence, frame:0
I (24894) Adev_Codec: Open codec device OK
E (24894) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (24896) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (24902) I2S_IF: STD Mode 0 bits:16/16 channel:2 sample_rate:16000 mask:3
I (24910) ES7210: Bits 16
I (24916) ES7210: Enable ES7210_INPUT_MIC1
I (24919) ES7210: Enable ES7210_INPUT_MIC2
I (24926) ES7210: Unmuted
I (24926) Adev_Codec: Open codec device OK
W (25000) app_sr: silence, frame:8
audio play end, 32034, 31 K
W (27802) app_sr: silence, frame:80
W (31836) app_sr: vad Time out
I (31836) app_audio: ESP_MN_STATE_TIMEOUT
I (31837) app_audio: ### record Stop, 118668 115K
audio load end, 11988, 11 K
E (31886) app_audio: [Start] start_openai, timestamp: 31886
I (31886) OpenAI: OpenAI create, version: 0.2.0
I (31889) ui_ctrl: Swich to panel[2]
I (31914) I2S_IF: Pending out channel for in channel running
E (31942) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (31942) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (31946) I2S_IF: STD Mode 1 bits:16/16 channel:2 sample_rate:16000 mask:3
I (32042) Adev_Codec: Open codec device OK
E (32042) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (32045) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (32052) I2S_IF: STD Mode 0 bits:16/16 channel:2 sample_rate:16000 mask:3
I (32062) ES7210: Bits 16
I (32065) ES7210: Enable ES7210_INPUT_MIC1
I (32067) ES7210: Enable ES7210_INPUT_MIC2
I (32074) ES7210: Unmuted
I (32074) Adev_Codec: Open codec device OK
I (32292) wifi:<ba-add>idx:0 (ifx:0, 48:7d:2e:8d:07:cb), tid:2, ssn:0, winSize:64
I (33417) esp-x509-crt-bundle: Certificate validated
I (36004) app_main: replay audio end
E (44714) OpenAI: Invalid URL (POST /v1/completionsaudio/transcriptions)
I (44714) ui_ctrl: update listen speak
E (44715) app_audio: [End] start_openai, +offset:12829

I use ESP32-S3-Box.My API Key is generated by my openAI account and is valid,  I use idf.py menuconfig to set the API key and Base_url ,Any hint why is is shown "API key is not valid",thank you!

alibukharai, 2023-12-11:
From the logs it seems like the OpenAI URL is not correct. please use the correct url 

```
 https://api.openai.com/v1/

```

lomon-emerson, 2023-12-11:
the "BASE_URL"  to set  “https://openai-proxy.pages.dev/v1” 
My API Key is generated by my openAI account and openai-proxy api is valid, but error:
I (21875) app_main: replay audio end
E (24202) OpenAI: Invalid URL (POST /v1audio/transcriptions)
I (24202) ui_ctrl: update listen speak
E (24202) app_audio: [End] start_openai, +offset:6412

thanks.

alibukharai, 2023-12-12:
The problem is not with the API key. Looks like the problem is with the server. Try to enable the debug logs.  you can find this line in sdkconfig file 
```
CONFIG_LOG_DEFAULT_LEVEL_DEBUG
```
and update this with 

```
CONFIG_LOG_DEFAULT_LEVEL_DEBUG=y
```
and build the example again

```
idf.py build
```
to flash the binaries use

```
python -m esptool -p /dev/ttyACM0 --chip esp32s3 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_size 16MB --flash_freq 80m 0x0 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0xd000 build/ota_data_initial.bin 0x10000 build/chatgpt_demo.bin 0x900000 build/storage.bin 0xb00000 build/srmodels/srmodels.bin 0x700000 factory_nvs/build/factory_nvs.bin

```

please put the log in this thread so i can see where does the logic breaks

lomon-emerson, 2023-12-13:
log:
I (4772) wifi:AP's beacon interval = 102400 us, DTIM period = 3
W (4869) app_sr: AFE Fetch Fail
W (5169) app_sr: AFE Fetch Fail
W (5469) app_sr: AFE Fetch Fail
I (5712) esp_netif_handlers: sta ip: 172.168.50.134, mask: 255.255.255.0, gw: 172.168.50.1
I (5712) wifi station: got ip:172.168.50.134
I (41626) ui-events: sr start once
I (41660) app_sr: AFE_FETCH_CHANNEL_VERIFIED, channel index: 2

I (41660) app_audio: ### record Start
W (41660) app_sr: silence, frame:1
I (41703) ui_ctrl: Swich to panel[1]
I (41799) app_audio: frame_rate= 16000, ch=2, width=16
I (41808) I2S_IF: Pending out channel for in channel running
E (41827) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (41827) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (41831) I2S_IF: STD Mode 1 bits:16/16 channel:2 sample_rate:16000 mask:3
I (41921) Adev_Codec: Open codec device OK
E (41924) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (41924) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (41930) I2S_IF: STD Mode 0 bits:16/16 channel:2 sample_rate:16000 mask:3
I (41938) ES7210: Bits 16
I (41944) ES7210: Enable ES7210_INPUT_MIC1
I (41947) ES7210: Enable ES7210_INPUT_MIC2
I (41953) ES7210: Unmuted
I (41954) Adev_Codec: Open codec device OK
W (42026) app_sr: silence, frame:9
audio play end, 32034, 31 K
W (43239) app_sr: silence, frame:27
W (46884) app_sr: silence, frame:87
W (47079) app_sr: silence, frame:3
W (47334) app_sr: silence, frame:2
W (50589) app_sr: vad Time out
I (50590) app_audio: ESP_MN_STATE_TIMEOUT
I (50590) app_audio: ### record Stop, 149358 145K
audio load end, 11988, 11 K
E (50758) app_audio: [Start] start_openai, timestamp: 50758
I (50758) OpenAI: OpenAI create, version: 0.2.0
I (50761) ui_ctrl: Swich to panel[2]
I (50780) I2S_IF: Pending out channel for in channel running
E (50801) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (50801) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (50806) I2S_IF: STD Mode 1 bits:16/16 channel:2 sample_rate:16000 mask:3
I (50915) Adev_Codec: Open codec device OK
E (50915) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (50919) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (50923) I2S_IF: STD Mode 0 bits:16/16 channel:2 sample_rate:16000 mask:3
I (50931) ES7210: Bits 16
I (50937) ES7210: Enable ES7210_INPUT_MIC1
I (50940) ES7210: Enable ES7210_INPUT_MIC2
I (50947) ES7210: Unmuted
I (50947) Adev_Codec: Open codec device OK
I (51928) esp-x509-crt-bundle: Certificate validated
I (54879) app_main: replay audio end
E (61524) OpenAI: cJSON_Parse failed!
I (61524) ui_ctrl: update listen speak
E (61525) app_audio: [End] start_openai, +offset:10767

alibukharai, 2023-12-13:
> log: I (4772) wifi:AP's beacon interval = 102400 us, DTIM period = 3 W (4869) app_sr: AFE Fetch Fail W (5169) app_sr: AFE Fetch Fail W (5469) app_sr: AFE Fetch Fail I (5712) esp_netif_handlers: sta [...]

Based on the logs, it appears there is an issue with the data received from the server. Have you implemented your own network station?

lomon-emerson, 2023-12-13:
esp-box settings: stored Base URL:https://e6bb1767.cloudflare-proxy-pages-cvq.pages.dev/v1

cloudflare pages _worker.js code:
export default {
  async fetch(request, env) {
    const url = new URL(request.url);
    url.host = "[api.openai.com](http://api.openai.com/)";
    // openai is already set all CORS heasders 
    return fetch(url, {
      headers: request.headers,
      method: request.method,
      body: request.body,
      redirect: 'follow'
    });
  }
}

or:
const TELEGRAPH_URL = 'https://api.openai.com';

export default {
  async fetch(request, env) {
      const NewResponse = await handleRequest(request)
      return NewResponse
  },

};

async function handleRequest(request) {
  const url = new URL(request.url);
  const headers_Origin = request.headers.get("Access-Control-Allow-Origin") || "*"
  url.host = TELEGRAPH_URL.replace(/^https?:\/\//, '');
  const modifiedRequest = new Request(url.toString(), {
    headers: request.headers,
    method: request.method,
    body: request.body,
    redirect: 'follow'
  });
  const response = await fetch(modifiedRequest);
  const modifiedResponse = new Response(response.body, response);
  modifiedResponse.headers.set('Access-Control-Allow-Origin', headers_Origin);
  return modifiedResponse;
}

but esp-box log error:
I (4639) wifi:new:<12,0>, old:<1,0>, ap:<255,255>, sta:<12,0>, prof:1
I (4640) wifi:state: init -> auth (b0)
I (4641) wifi:state: auth -> assoc (0)
I (4649) wifi:state: assoc -> run (10)
I (4656) wifi:<ba-add>idx:0 (ifx:0, 04:d4:c4:43:34:d8), tid:0, ssn:0, winSize:64
I (4665) wifi:connected with inspur5, aid = 2, channel 12, BW20, bssid = 04:d4:c4:43:34:d8
I (4666) wifi:security: WPA2-PSK, phy: bgn, rssi: -75
I (4669) wifi:pm start, type: 1

I (4670) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 25000, mt_pti: 0, mt_time: 10000
I (4690) wifi:<ba-add>idx:1 (ifx:0, 04:d4:c4:43:34:d8), tid:6, ssn:0, winSize:64
I (4750) wifi:AP's beacon interval = 102400 us, DTIM period = 3
W (4870) app_sr: AFE Fetch Fail
W (5170) app_sr: AFE Fetch Fail
W (5470) app_sr: AFE Fetch Fail
W (5770) app_sr: AFE Fetch Fail
W (6070) app_sr: AFE Fetch Fail
I (6179) esp_netif_handlers: sta ip: 172.168.50.134, mask: 255.255.255.0, gw: 172.168.50.1
I (6179) wifi station: got ip:172.168.50.134
I (27648) app_sr: wakeword detected
I (27648) app_audio: ### record Start
I (27648) ui_ctrl: Swich to panel[1]
I (27728) app_audio: frame_rate= 16000, ch=2, width=16
I (27731) I2S_IF: Pending out channel for in channel running
E (27745) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (27745) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (27751) I2S_IF: STD Mode 1 bits:16/16 channel:2 sample_rate:16000 mask:3
I (27783) app_sr: AFE_FETCH_CHANNEL_VERIFIED, channel index: 2

I (27854) Adev_Codec: Open codec device OK
E (27856) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (27858) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (27864) I2S_IF: STD Mode 0 bits:16/16 channel:2 sample_rate:16000 mask:3
I (27872) ES7210: Bits 16
I (27878) ES7210: Enable ES7210_INPUT_MIC1
I (27881) ES7210: Enable ES7210_INPUT_MIC2
I (27887) ES7210: Unmuted
I (27888) Adev_Codec: Open codec device OK
W (27960) app_sr: silence, frame:15
audio play end, 32034, 31 K
W (29488) app_sr: silence, frame:40
W (29624) app_sr: silence, frame:1
W (33208) app_sr: vad Time out
I (33209) app_audio: ESP_MN_STATE_TIMEOUT
I (33209) app_audio: ### record Stop, 95139 92K
audio load end, 11988, 11 K
E (33377) app_audio: [Start] start_openai, timestamp: 33377
I (33377) OpenAI: OpenAI create, version: 0.2.0
I (33380) ui_ctrl: Swich to panel[2]
I (33401) I2S_IF: Pending out channel for in channel running
E (33414) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (33416) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (33420) I2S_IF: STD Mode 1 bits:16/16 channel:2 sample_rate:16000 mask:3
I (33508) Adev_Codec: Open codec device OK
E (33508) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (33510) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (33516) I2S_IF: STD Mode 0 bits:16/16 channel:2 sample_rate:16000 mask:3
I (33530) ES7210: Bits 16
I (33534) ES7210: Enable ES7210_INPUT_MIC1
I (33536) ES7210: Enable ES7210_INPUT_MIC2
I (33540) ES7210: Unmuted
I (33540) Adev_Codec: Open codec device OK
I (34810) esp-x509-crt-bundle: Certificate validated
I (37463) app_main: replay audio end
E (41426) OpenAI: Invalid URL (POST /v1audio/transcriptions)
I (41426) ui_ctrl: update listen speak
E (41426) app_audio: [End] start_openai, +offset:8049

LHYhorion, 2023-12-14:
Based on the initial analysis of the logs, it is likely that the issue is related to the Base URL: https://e6bb1767.cloudflare-proxy-pages-cvq.pages.dev/v1. It is recommended to first test whether this URL is accessible in the script. If it is accessible, please provide more logs for further investigation of the issue.

lomon-emerson, 2023-12-15:
I used ngix for https://api.openai.com Interface configuration, settings: stored Base URL: https://openai.allsoftcould.xyz/v1 The esp box still displays "API Key is not valid", and the log update shows the following:

I (163984) app_audio: ### record Start
I (164034) ui_ctrl: Swich to panel[1]
I (164131) app_audio: frame_rate= 16000, ch=2, width=16
I (164134) I2S_IF: Pending out channel for in channel running
E (164153) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (164154) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (164158) I2S_IF: STD Mode 1 bits:16/16 channel:2 sample_rate:16000 mask:3
I (164248) Adev_Codec: Open codec device OK
E (164248) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (164250) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (164258) I2S_IF: STD Mode 0 bits:16/16 channel:2 sample_rate:16000 mask:3
I (164265) ES7210: Bits 16
I (164270) ES7210: Enable ES7210_INPUT_MIC1
I (164273) ES7210: Enable ES7210_INPUT_MIC2
I (164283) ES7210: Unmuted
I (164283) Adev_Codec: Open codec device OK
W (164355) app_sr: silence, frame:24
audio play end, 32034, 31 K
W (167486) app_sr: silence, frame:51
W (170741) app_sr: vad Time out
I (170741) app_audio: ESP_MN_STATE_TIMEOUT
I (170741) app_audio: ### record Stop, 114576 111K
audio load end, 11988, 11 K
E (170920) app_audio: [Start] start_openai, timestamp: 170920
I (170920) OpenAI: OpenAI create, version: 0.2.0
I (170923) ui_ctrl: Swich to panel[2]
I (170952) I2S_IF: Pending out channel for in channel running
E (170963) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (170965) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (170967) I2S_IF: STD Mode 1 bits:16/16 channel:2 sample_rate:16000 mask:3
I (171049) Adev_Codec: Open codec device OK
E (171049) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (171051) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (171058) I2S_IF: STD Mode 0 bits:16/16 channel:2 sample_rate:16000 mask:3
I (171066) ES7210: Bits 16
I (171072) ES7210: Enable ES7210_INPUT_MIC1
I (171075) ES7210: Enable ES7210_INPUT_MIC2
I (171081) ES7210: Unmuted
I (171082) Adev_Codec: Open codec device OK
I (171886) esp-x509-crt-bundle: Certificate validated
I (175011) app_main: replay audio end
I (180064) ui_ctrl: update listen speak
E (180064) app_audio: [End] start_openai, +offset:9144

LHYhorion, 2023-12-15:
From this log, it's challenging for me to pinpoint the issue. To address this problem, I suggest you first verify the availability of the URL in your script. Once you confirm that the URL is accessible, enable the debug log level in menuconfig, test again, and provide additional debug logs. This would greatly assist me in identifying the problem.

LHYhorion, 2023-12-18:
> I used ngix for https://api.openai.com Interface configuration, settings: stored Base URL: https://openai.allsoftcould.xyz/v1 The esp box still displays "API Key is not valid", and the log update shows the following:
> 
> chatgpt_demo % idf.py -p /dev/cu.usbmodem141201 monitor Executing action: monitor Running idf_monitor in directory /Users/xxx/esp-box/examples/chatgpt_demo Executing "/Users/xxx/.espressif/python_env/idf5.1_py3.10_env/bin/python /Users/xxx/esp/esp-idf/tools/idf_monitor.py -p /dev/cu.usbmodem141201 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 --revision 0 /Users/xxx/esp-box/examples/chatgpt_demo/build/chatgpt_demo.elf -m '/Users/xxx/.espressif/python_env/idf5.1_py3.10_env/bin/python' '/Users/xxx/esp/esp-idf/tools/idf.py' '-p' '/dev/cu.usbmodem141201'"... --- esp-idf-monitor 1.3.3 on /dev/cu.usbmodem141201 115200 --- --- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H --- ESP-ROM:esp32s3-20210327 Build:Mar 27 2021 rst:0x15 (USB_UART_CHIP_RESET),boot:0xa (SPI_FAST_FLASH_BOOT) Saved PC:0x4037d3a2 0x4037d3a2: esp_cpu_wait_for_intr at /Users/xxx/esp/esp-idf/components/esp_hw_support/cpu.c:121
> 
> SPIWP:0xee mode:DIO, clock div:1 load:0x3fce3818,len:0x18e4 load:0x403c9700,len:0x4 load:0x403c9704,len:0xda4 load:0x403cc700,len:0x3084 entry 0x403c9934 I (26) boot: ESP-IDF v5.1.2-dirty 2nd stage bootloader I (27) boot: compile time Dec 15 2023 09:48:15 I (27) boot: Multicore bootloader I (30) boot: chip revision: v0.1 I (34) qio_mode: Enabling default flash chip QIO I (39) boot.esp32s3: Boot SPI Speed : 80MHz I (44) boot.esp32s3: SPI Mode : QIO I (49) boot.esp32s3: SPI Flash Size : 16MB I (54) boot: Enabling RNG early entropy source... I (59) boot: Partition Table: I (63) boot: ## Label Usage Type ST Offset Length I (70) boot: 0 nvs WiFi data 01 02 00009000 00004000 I (77) boot: 1 otadata OTA data 01 00 0000d000 00002000 I (85) boot: 2 phy_init RF data 01 01 0000f000 00001000 I (92) boot: 3 ota_0 OTA app 00 10 00010000 00600000 I (100) boot: 4 uf2 factory app 00 00 00700000 00200000 I (107) boot: 5 storage Unknown data 01 82 00900000 00200000 I (115) boot: 6 model Unknown data 01 82 00b00000 003e8000 I (123) boot: End of partition table I (127) esp_image: segment 0: paddr=00010020 vaddr=3c120020 size=231cb4h (2301108) map I (484) esp_image: segment 1: paddr=00241cdc vaddr=3fc9ba00 size=071bch ( 29116) load I (490) esp_image: segment 2: paddr=00248ea0 vaddr=40374000 size=07178h ( 29048) load I (496) esp_image: segment 3: paddr=00250020 vaddr=42000020 size=11cfa4h (1167268) map I (675) esp_image: segment 4: paddr=0036cfcc vaddr=4037b178 size=10874h ( 67700) load I (689) esp_image: segment 5: paddr=0037d848 vaddr=600fe010 size=00004h ( 4) load I (700) boot: Loaded app from partition at offset 0x10000 I (700) boot: Disabling RNG early entropy source... I (711) cpu_start: Multicore app I (712) octal_psram: vendor id : 0x0d (AP) I (712) octal_psram: dev id : 0x02 (generation 3) I (715) octal_psram: density : 0x03 (64 Mbit) I (720) octal_psram: good-die : 0x01 (Pass) I (725) octal_psram: Latency : 0x01 (Fixed) I (731) octal_psram: VCC : 0x01 (3V) I (736) octal_psram: SRF : 0x01 (Fast Refresh) I (742) octal_psram: BurstType : 0x01 (Hybrid Wrap) I (747) octal_psram: BurstLen : 0x01 (32 Byte) I (753) octal_psram: Readlatency : 0x02 (10 cycles@Fixed) I (759) octal_psram: DriveStrength: 0x00 (1/1) I (765) MSPI Timing: PSRAM timing tuning index: 4 I (770) esp_psram: Found 8MB PSRAM device I (774) esp_psram: Speed: 80MHz I (856) mmu_psram: Instructions copied and mapped to SPIRAM I (857) cpu_start: Pro cpu up. I (857) cpu_start: Starting app cpu, entry point is 0x403758c4 0x403758c4: call_start_cpu1 at /Users/xxx/esp/esp-idf/components/esp_system/port/cpu_start.c:157
> 
> I (0) cpu_start: App cpu up. I (1104) esp_psram: SPI SRAM memory test OK I (1113) cpu_start: Pro cpu start user code I (1113) cpu_start: cpu freq: 240000000 Hz I (1113) cpu_start: Application information: I (1116) cpu_start: Project name: chatgpt_demo I (1121) cpu_start: App version: v0.5.0-32-g10d166a-dirty I (1128) cpu_start: Compile time: Dec 15 2023 09:48:04 I (1134) cpu_start: ELF file SHA256: f1ea9e484bccbe12... I (1140) cpu_start: ESP-IDF: v5.1.2-dirty I (1146) cpu_start: Min chip rev: v0.0 I (1150) cpu_start: Max chip rev: v0.99 I (1155) cpu_start: Chip rev: v0.1 I (1160) heap_init: Initializing. RAM available for dynamic allocation: I (1167) heap_init: At 3FCA8170 len 000415A0 (261 KiB): DRAM I (1174) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM I (1181) heap_init: At 600FE014 len 00001FD4 (7 KiB): RTCRAM I (1187) esp_psram: Adding pool of 7040K of PSRAM memory to heap allocator I (1195) spi_flash: detected chip: gd I (1199) spi_flash: flash io: qio I (1203) sleep: Configure to isolate all GPIO pins in sleep state I (1210) sleep: Enable automatic switching of GPIO sleep configuration I (1217) app_start: Starting scheduler on CPU0 I (1222) app_start: Starting scheduler on CPU1 I (1222) main_task: Started on CPU0 I (1232) esp_psram: Reserving pool of 8K of internal memory for DMA/internal allocations I (1240) main_task: Calling app_main() I (1261) settings: stored ssid:xxx I (1261) settings: stored password:xxx I (1261) settings: stored OpenAI:sk-xxx I (1268) settings: stored Base URL:https://openai.allsoftcould.xyz/v1 I (1400) ESP-BOX: Partition size: total: 1920401, used: 1038136 I (1401) LVGL: Starting LVGL task I (1401) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pul [... cut ...]  ES7210: Unmuted I (22078) Adev_Codec: Open codec device OK I (23208) esp-x509-crt-bundle: Certificate validated I (26008) app_main: replay audio end I (30067) ui_ctrl: update listen speak E (30068) app_audio: [End] start_openai, +offset:8158

Hello, after testing, it has been confirmed that your URL is indeed valid. Currently, there is an issue with changing the URL in the demo, and we have already fixed this issue. The solution will be released in the next version. Thank you for your feedback.

LHYhorion, 2023-12-21:
Hello, we have updated the version and tested it using the URL https://e6bb1767.cloudflare-proxy-pages-cvq.pages.dev/v1/. It seems to be working fine.

lomon-emerson, 2023-12-28:
I have updated using the URL https://e6bb1767.cloudflare-proxy-pages-cvq.pages.dev/v1/，but can't receive sound tts，only text:
I (23690) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (23697) I2S_IF: STD Mode 0 bits:16/16 channel:2 sample_rate:16000 mask:3
I (23704) ES7210: Bits 16
I (23710) ES7210: Enable ES7210_INPUT_MIC1
I (23713) ES7210: Enable ES7210_INPUT_MIC2
I (23720) ES7210: Unmuted
I (23720) Adev_Codec: Open codec device OK
I (23725) app_main: replay audio end
I (26925) ui_ctrl: update reply question
I (26925) ui_ctrl: update listen speak
I (28178) esp-x509-crt-bundle: Certificate validated
I (33369) ui_ctrl: update reply question
I (33369) ui_ctrl: update listen speak
I (33370) ui_ctrl: update reply content
I (33373) ui_ctrl: decode:[489, 489] 面线对象是指在计算机图形学中，由一系列有序的顶点和边构成的二维或三维图形。它们可以表示各种形状，如平面、多边形、曲线等。面线对象通常用于描述和绘制图形，例如在三维建模、动画和游戏开发中使用。通过定义顶点和边的连接关系，可以创建出复杂的图形，并进行各种操作，如平移、旋转、缩放等。面线对象在计算机图形学中是非常常见和重要的概念。
I (33421) ui_ctrl: reply scroll timer start
I (33423) ui_ctrl: Swich to panel[3]
I (34624) esp-x509-crt-bundle: Certificate validated
E (42047) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (42047) task_wdt:  - IDLE (CPU 0)
E (42047) task_wdt: Tasks currently running:
E (42047) task_wdt: CPU 0: SR Handler Task
E (42047) task_wdt: CPU 1: IDLE
E (42047) task_wdt: Print CPU 0 (current core) backtrace
Backtrace: 0x4207CE87:0x3FCA0190 0x40377679:0x3FCA01C0 0x400559DD:0x3FCD76C0 |<-CORRUPTED
0x4207ce87: task_wdt_timeout_handling at /Users/lomon/esp/esp-idf/components/esp_system/task_wdt/task_wdt.c:461
 (inlined by) task_wdt_isr at /Users/lomon/esp/esp-idf/components/esp_system/task_wdt/task_wdt.c:585
0x40377679: _xt_lowint1 at /Users/lomon/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1240
0x400559dd: _xtos_set_intlevel in ROM
E (47047) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (47047) task_wdt:  - IDLE (CPU 0)
E (47047) task_wdt: Tasks currently running:
E (47047) task_wdt: CPU 0: SR Handler Task
E (47047) task_wdt: CPU 1: IDLE
E (47047) task_wdt: Print CPU 0 (current core) backtrace
Backtrace: 0x4207CE87:0x3FCA0190 0x40377679:0x3FCA01C0 0x40376A4C:0x3FCD7710 0x40376BCF:0x3FCD7740 0x40387D2D:0x3FCD7760 0x4203563A:0x3FCD7780 0x42034CF7:0x3FCD7860 0x4200AE05:0x3FCD78A0 0x4200B53D:0x3FCD78D0
0x4207ce87: task_wdt_timeout_handling at /Users/lomon/esp/esp-idf/components/esp_system/task_wdt/task_wdt.c:461
 (inlined by) task_wdt_isr at /Users/lomon/esp/esp-idf/components/esp_system/task_wdt/task_wdt.c:585
0x40377679: _xt_lowint1 at /Users/lomon/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1240
0x40376a4c: find_containing_heap at /Users/lomon/esp/esp-idf/components/heap/heap_caps.c:365
 (inlined by) heap_caps_realloc_base at /Users/lomon/esp/esp-idf/components/heap/heap_caps.c:431
0x40376bcf: heap_caps_realloc_default at /Users/lomon/esp/esp-idf/components/heap/heap_caps.c:264
0x40387d2d: realloc at /Users/lomon/esp/esp-idf/components/newlib/heap.c:34
0x4203563a: OpenAI_Speech_Request at /Users/lomon/esp-box-master/examples/chatgpt_demo/managed_components/espressif__openai/OpenAI.c:2462
 (inlined by) OpenAI_Speech_Post at /Users/lomon/esp-box-master/examples/chatgpt_demo/managed_components/espressif__openai/OpenAI.c:2480
0x42034cf7: OpenAI_AudioSpeechMessage at /Users/lomon/esp-box-master/examples/chatgpt_demo/managed_components/espressif__openai/OpenAI.c:1933
0x4200ae05: start_openai at /Users/lomon/esp-box-master/examples/chatgpt_demo/main/main.c:114
0x4200b53d: sr_handler_task at /Users/lomon/esp-box-master/examples/chatgpt_demo/main/app/app_audio.c:320
E (52047) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (52047) task_wdt:  - IDLE (CPU 0)
E (52047) task_wdt: Tasks currently running:
E (52047) task_wdt: CPU 0: SR Handler Task
E (52047) task_wdt: CPU 1: IDLE
E (52047) task_wdt: Print CPU 0 (current core) backtrace

lomon-emerson, 2023-12-28:
try use https://openai.allsoftcould.xyz/v1/ is working fine,but Frequent restarts may occur,error is as follows:

I (529149) Adev_Codec: Open codec device OK
I (530035) esp-x509-crt-bundle: Certificate validated
I (532589) app_audio: Player IDLE
I (532592) I2S_IF: Pending out channel for in channel running
E (532607) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (532607) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (532613) I2S_IF: STD Mode 1 bits:16/16 channel:2 sample_rate:16000 mask:3
I (532701) Adev_Codec: Open codec device OK
E (532703) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (532706) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (532710) I2S_IF: STD Mode 0 bits:16/16 channel:2 sample_rate:16000 mask:3
I (532721) ES7210: Bits 16
I (532724) ES7210: Enable ES7210_INPUT_MIC1
I (532727) ES7210: Enable ES7210_INPUT_MIC2
I (532735) ES7210: Unmuted
I (532735) Adev_Codec: Open codec device OK
I (532739) app_main: replay audio end
I (532790) ui_ctrl: update reply question
I (532791) ui_ctrl: update listen speak
I (534034) esp-x509-crt-bundle: Certificate validated
I (543343) ui_ctrl: update reply question
I (543344) ui_ctrl: update listen speak
I (543346) ui_ctrl: update reply content
I (543358) ui_ctrl: decode:[1204, 1204] 
hello.
I (200971) ui_ctrl: reply scroll timer start
I (200971) ui_ctrl: Swich to panel[3]
I (202173) esp-x509-crt-bundle: Certificate validated
E (227797) OpenAI: ./managed_components/espressif__openai/OpenAI.c:1898 (OpenAI_SpeechResponseCreate):malloc failed!
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core  0 register dump:
PC      : 0x4200ae08  PS      : 0x00060d30  A0      : 0x8200b540  A1      : 0x3fcd7540  
0x4200ae08: start_openai at /Users/lomon/esp-box-master/examples/chatgpt_demo/main/main.c:115
A2      : 0x3fccaac4  A3      : 0x42034bd8  A4      : 0x3fcb78bc  A5      : 0x3c126dfc  
0x42034bd8: OpenAI_AudioSpeechMessage at /Users/lomon/esp-box-master/examples/chatgpt_demo/managed_components/espressif__openai/OpenAI.c:1914
A6      : 0x0001e387  A7      : 0x3fccdf94  A8      : 0x8200ae08  A9      : 0x3fcd7500  
A10     : 0x00000000  A11     : 0x3c597f2c  A12     : 0x3fcd8f04  A13     : 0x00260e80  
A14     : 0x3c591610  A15     : 0x0000005c  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffc  
0x400556d5: strlen in ROM
0x400556e5: strlen in ROM
Backtrace: 0x4200ae05:0x3fcd7540 0x4200b53d:0x3fcd7570
0x4200ae05: start_openai at /Users/lomon/esp-box-master/examples/chatgpt_demo/main/main.c:114
0x4200b53d: sr_handler_task at /Users/lomon/esp-box-master/examples/chatgpt_demo/main/app/app_audio.c:320

LHYhorion, 2023-12-28:
> 尝试使用 https://openai.allsoftcould.xyz/v1/ 工作正常，但可能会频繁重启，错误如下：
> 
> I （529149） Adev_Codec：打开编解码器设备 正常 I （530035） esp-x509-crt-bundle：证书验证 I （532589） app_audio：玩家空闲 I （532592） I2S_IF：运行 E （532607） i2s_common的通道的挂出通道： i2s_channel_disable（1030）：通道尚未启用 I （532607） I2S_IF：通道模式 0 位：16/16 通道：2 掩模：3 I （532613） I2S_IF： STD 模式 1 位：16/16 通道：2 sample_rate：16000 掩模：3 I （532701） Adev_Codec： 打开编解码器设备 OK E （532703） i2s_common： i2s_channel_disable（1030）： 通道尚未启用 I （532706） I2S_IF： 通道模式 0 位：16/16 通道：2 掩模：3 I （532710） I2S_IF： STD 模式 0 位：16/16 通道：2 sample_rate：16000 掩模：3 I （532721） ES7210 [美] 位 16 I （532724） ES7210：使能ES7210_INPUT_MIC1 I （532727） ES7210：使能 ES7210_INPUT_MIC2 I （532735） ES7210：未静音 I （532735） Adev_Codec：打开编解码器设备 OK I （532739） app_main：重播音频结束 I （532790） ui_ctrl：更新回复问题 I （532791） ui_ctrl：更新 listen speak I （534034） esp-x509-crt-bundle：证书验证 I （543343） ui_ctrl： 更新回复问题 I （543344） ui_ctrl：更新听 Speak I （543346） ui_ctrl：更新回复内容 I （543358） ui_ctrl：解码：[1204， 1204] 你好。 I （200971） ui_ctrl： 回复滚动计时器启动 I （200971） ui_ctrl： Swich to panel[3] I （202173） esp-x509-crt-bundle： Certificate validated E （227797） OpenAI： ./managed_components/espressif__openai/OpenAI.c：1898 （OpenAI_SpeechResponseCreate）：malloc failed！ Guru Meditation 错误：核心 0 恐慌 （LoadProhibited）。异常未处理。核心 0 寄存器转储： PC ： 0x4200ae08 PS ： 0x00060d30 A0 ： 0x8200b540 A1 ： 0x3fcd7540 0x4200ae08： start_openai at /Users/lomon/esp-box-master/examples/chatgpt_demo/main/main.c：115 A2 ： 0x3fccaac4 A3 ： 0x42034bd8 A4 ： 0x3fcb78bc A5 ： 0x3c126dfc 0x42034bd8： OpenAI_AudioSpeechMessage at /Users/lomon/esp-box-master/examples/chatgpt_demo/managed_components/espressif__openai/OpenAI.c：1914A6 ： 0x0001e387 A7 ： 0x3fccdf94 A8 ： 0x8200ae08 A9 ： 0x3fcd7500 A10 ： 0x00000000 A11 ： 0x3c597f2c A12 ： 0x3fcd8f04 A13 ： 0x00260e80 A14 ： 0x3c591610 A15 ： 0x0000005c SAR ： 0x0000001b 原因： 0x0000001c EXCVADDR： 0x00000000 LBEG ： 0x400556d5 LEND ： 0x400556e5 LCOUNT ： 0xfffffffc 0x400556d5： strlen in ROM 0x400556e5： strlen in ROM 回溯： 0x4200ae05：0x3fcd7540 0x4200b53d：0x3fcd75700x4200ae05：start_openai /Users/lomon/esp-box-master/examples/chatgpt_demo/main/main.c：114 0x4200b53d：sr_handler_task /Users/lomon/esp-box-master/examples/chatgpt_demo/main/app/app_audio.c：320

Hello, we have identified issues that may cause crashes when changing the URL. We have addressed this issue and will release a version in the near future. However, compatibility with changing URLs may not be optimal at the moment. In the next version, there may still be no sound alerts when changing the URL. However, if you use the official Open AI URL, everything should function normally.

espressif2022, 2024-01-18:
The new version has been released, and I will close this Issue. If you have new questions, please consult
Please refer to this https://github.com/espressif/esp-box/issues/123

closed 2024-01-18 by espressif2022