espressif/esp-box #120: ESP32-S3-BOX-3 Display & Touchscreen hardware

ophilli, 2023-12-14:
Hello, can you please confirm the display & touchscreen driver used by the ESP32-S3-BOX-3?

This [spec graphic](https://github.com/espressif/esp-box/blob/3bcc2a303f0d54738814a353ca30a172fc4eebbe/docs/_static/box_3_hardware_overview/specs_for_box_3.png) explains that the box-3 uses the ILI9342C display driver and does not mention provide much information about the touchscreen hardware, but [the bsp](https://github.com/espressif/esp-bsp/blob/a966ec308b4cd0c59400160dbc14bf60177b44b3/bsp/esp-box-3/esp-box-3.c#L25) seems to be using the ILI9341.

If I try to use examples that use the ILI9342C like https://github.com/sambenko/esp32s3box-display-and-publish and update the pins to match

```
diff --git a/src/main.rs b/src/main.rs
index 143653d..6bbf2c6 100644
--- a/src/main.rs
+++ b/src/main.rs
@@ -133,7 +133,7 @@ async fn main(spawner: Spawner) {
     let mosi = io.pins.gpio6;
 
     let dc = io.pins.gpio4.into_push_pull_output();
-    let mut backlight = io.pins.gpio45.into_push_pull_output();
+    let mut backlight = io.pins.gpio47.into_push_pull_output();
     let reset = io.pins.gpio48.into_push_pull_output();
 
     let spi = Spi::new_no_cs_no_miso(
```

I'm able to power on the backlight but I cannot display anything. Can you please provide confirmation about what hardware the box-3 comes with so that I can continue debugging?

espressif2022, 2023-12-14:
Maye be the reset GPIO level is not correct, you can check that first.
And the touch IC is GT911.

espressif2022, 2024-01-18:
I will turn off this issue, welcome to consult if you have new questions.

closed 2024-01-18 by espressif2022

UnexDev, 2024-04-03:
Any progress on this, @ophilli? Having the same issue. Cannot figure out which pins to use.

ophilli, 2024-04-18:
@UnexDev yes I got the LCD to work in a local repo based on https://github.com/sambenko/esp32s3box-display-and-publish but I don't have a driver for the GT911.

```
   let sclk = io.pins.gpio7;
    let mosi = io.pins.gpio6;

    let cs = io.pins.gpio5.into_push_pull_output();
    let dc = io.pins.gpio4.into_push_pull_output();
    let mut backlight = io.pins.gpio47.into_push_pull_output();
    let mut reset = io.pins.gpio48.into_push_pull_output();

    reset.internal_pull_up(true);

    let spi = Spi::new_no_cs_no_miso(
        peripherals.SPI2,
        sclk,
        mosi,
        40u32.MHz(),
        SpiMode::Mode0,
        &clocks,
    );

    let di = SPIInterface::new(spi, dc, cs);
    delay.delay_ms(500u32);

    let mut display_struct = EmbassyTaskDisplay {
        display: match mipidsi::Builder::ili9342c_rgb565(di)
            .with_display_size(320, 240)
            .with_framebuffer_size(320, 240)
            .with_orientation(mipidsi::Orientation::PortraitInverted(false))
            .with_color_order(mipidsi::ColorOrder::Bgr)
            .init(&mut delay, None)
        {
            Ok(display) => {
                println!("Display initialization succeeded!");
                display
            }
            Err(e) => {
                println!("Display initialization failed: {:?}", e);
                panic!("Display initialization failed");
            }
        },
    };

    backlight.set_high().unwrap();
```

opsnull, 2024-05-11:
@ophilli could you please show the complete code? I am also experiencing the issue of a  backlight white screen on power up, with no images being displayed.

opsnull, 2024-05-11:
@UnexDev Do you resolve that problem?

opsnull, 2024-05-11:
@espressif2022 I am also experiencing the issue of a backlight white screen on power up, with no images being displayed.

My Code are following:
```
        let sclk = io.pins.gpio7;
        let mosi = io.pins.gpio6;
        let miso = io.pins.gpio9;
        let cs = io.pins.gpio5;
        let dc = io.pins.gpio4.into_push_pull_output();
        let mut backlight = io.pins.gpio47.into_push_pull_output();
        let mut reset = io.pins.gpio48.into_push_pull_output();
        reset.internal_pull_up(true);

        let spi = Spi::new(
            peripherals.SPI3,
            sclk,
            mosi,
            miso,
            cs,
            40u32.MHz(),
            SpiMode::Mode0,
            &mut system.peripheral_clock_control,
            &clocks,
        );

        let di = display_interface_spi::SPIInterfaceNoCS::new(spi, dc);
        delay.delay_ms(500u32);

        let mut display = mipidsi::Builder::ili9341_rgb565(di)
            .with_display_size(320, 240)
            .with_framebuffer_size(320, 240)
            .with_orientation(mipidsi::Orientation::PortraitInverted(false))
            .with_color_order(mipidsi::ColorOrder::Bgr)
            .init(&mut delay, Some(reset))
            .unwrap();

        backlight.set_high().unwrap();
        display.clear(Rgb565::WHITE).unwrap();

        let default_style =
            MonoTextStyleBuilder::new().font(&FONT_10X20).text_color(RgbColor::BLACK).build();

        let espressif_style : MonoTextStyle<'_, Rgb565>=
            MonoTextStyleBuilder::new().font(&FONT_10X20).text_color(RgbColor::CYAN).build();

        for position_y in (25..240).step_by(28) {
            for position_x in 0..320 {
                Text::with_alignment(
                    "O",
                    Point::new(position_x, position_y),
                    default_style,
                    Alignment::Center,
                )
                .draw(&mut display)
                .unwrap();
            }
        }

```

ophilli, 2024-05-13:
@opsnull you can reference https://github.com/ophilli/esp32-s3-box-3/commit/0fb34fe9d7d90bdd8c73f54247a9edaab002130f - I am able to flash this on my esp32 s3-box-3 and display a simple UI using that commit.

I see a few differences between my snippet and yours. I'm using the SPI2 peripheral and you're using SPI3. I'm not using the MISO at all, and I'm passing the cs line to the [`display_interface_spi::SPIInterface()`](https://docs.rs/display-interface-spi/0.4.1/display_interface_spi/struct.SPIInterface.html) instead of the SPI struct from the esp hal.

To be honest I don't currently understand what struct the CS line should really belong to, or what the difference is between those SPI peripherals, but I am pretty sure that there is no MISO line because [it is not shown in the schematic](https://github.com/espressif/esp-box/blob/master/hardware/SCH_ESP32-S3-BOX-3_V1.0/SCH_ESP32-S3-BOX-3-MB_V1.1_20230808.pdf).

opsnull, 2024-05-13:
@ophilli Thanks very much to show your code! I have a hard time to figure out its a problem of init method `init(&mut delay, Some(reset))`, for ESP32-S3-Box-3, it should be something like `init(&mut delay, None)`, and I create to struct NoPin to implement embedded_hal::digital::v2::OutputPin trait to avoid compiler error. Please reference code comment at  #143.