espressif/esp-box #121: esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0x900000 for file: storage.bin

GYY-110, 2023-12-14:
[3/3] cmd.exe /C "cd /D G:\project\ESP-IDF\code\esp-box\esp-box\examples\chatgpt_demo\b...storage.bin --page-size=256 --obj-name-len=32 --meta-len=4 --use-magic --use-magic-len" 

 *  正在执行任务: g:\project\ESP-IDF\TOOLS\python_env\idf5.1_py3.11_env\Scripts\python.exe g:\project\ESP-IDF\esp-idf\components\esptool_py\esptool\esptool.py -p COM4 -b 460800 --before default_reset --after hard_reset --chip esp32s3 write_flash --flash_mode dio --flash_freq 80m --flash_size 16MB 0x0 bootloader/bootloader.bin 0x700000 chatgpt_demo.bin 0x8000 partition_table/partition-table.bin 0xd000 ota_data_initial.bin 0xb00000 srmodels/srmodels.bin 0x900000 storage.bin 

usage: esptool write_flash [-h] [--erase-all] [--flash_freq {keep,80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}] [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size {detect,keep,256KB,512KB,1MB,2MB,2MB-c1,4MB,4MB-c1,8MB,16MB,32MB,64MB,128MB}] [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--encrypt] [--encrypt-files <address> <filename> [<address> <filename> ...]] [--ignore-flash-encryption-efuse-setting] [--force]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0x900000 for file: storage.bin
 *  终端进程“g:\project\ESP-IDF\TOOLS\python_env\idf5.1_py3.11_env\Scripts\python.exe 'g:\project\ESP-IDF\esp-idf\components\esptool_py\esptool\esptool.py', '-p', 'COM4', '-b', '460800', '--before', 'default_reset', '--after', 'hard_reset', '--chip', 'esp32s3', 'write_flash', '--flash_mode', 'dio', '--flash_freq', '80m', '--flash_size', '16MB', '0x0', 'bootloader/bootloader.bin', '0x700000', 'chatgpt_demo.bin', '0x8000', 'partition_table/partition-table.bin', '0xd000', 'ota_data_initial.bin', '0xb00000', 'srmodels/srmodels.bin', '0x900000', 'storage.bin'”已终止，退出代码: 2。

bearpaddy, 2023-12-25:
@espressif2022 What do you mean, I get this error when running `idf.py flash`

LHYhorion, 2023-12-26:
> @espressif2022 What do you mean, I get this error when running `idf.py flash`

Hello, the flashing method for the ChatGPT demo is different from other demos. For specific flashing steps, please refer to https://github.com/espressif/esp-box/blob/master/examples/chatgpt_demo/README.md.

espressif2022, 2024-01-18:
I will turn off this issue, welcome to consult if you have new questions

closed 2024-01-18 by espressif2022