espressif/esp-box #122: can not find bsp_board.h when I open the example 《mp3_demo》with VSCODE

yirenonege, 2023-12-15:
[image]

I can not find the heardfile as  show in picture

yirenonege, 2023-12-15:
[image]

LHYhorion, 2023-12-15:
> <img alt="image" width="619" src="https://private-user-images.githubusercontent.com/62145396/290782311-47e16397-cb3c-45b0-a913-3f311247f385.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI2MzQ0OTQsIm5iZiI6MTcwMjYzNDE5NCwicGF0aCI6Ii82MjE0NTM5Ni8yOTA3ODIzMTEtNDdlMTYzOTctY2IzYy00NWIwLWE5MTMtM2YzMTEyNDdmMzg1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjE1VDA5NTYzNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQ2ODVkMmJiNDcxYThiM2VmNDc2Yjg2NzViMGM1ZmNjYzVlODU3N2NmOWUwOTg4ZjllODI1NDgxMGRkOTQ3MjEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Fv8i-MRCjEEzGRNXYgeJm0jk7-UQc0qagN3pAPQ_nok">

Hello, is the .h file not found during compilation, or is it just a prompt in VSCode?

LHYhorion, 2023-12-15:
You can try adding the path to the compile_commands.json file in the "build" folder to the browsing path in the C/C++ configuration. Additionally, select the option to limit symbols to included headers. This may help resolve the prompting issue in your vscode.

espressif2022, 2024-01-18:
I will turn off this issue, welcome to consult if you have new questions

closed 2024-01-18 by espressif2022