espressif/esp-box #123: [ESP32 BOX S3] OpenAI  提示 cJSON_Parse failed!

xiaoshanggys, 2023-12-16:
没改动代码，正常烧录下载
sr handle task, mute:1
successfully created record_audio_buffer with a size: 256000
audio_rx_buffer with a size: 1048576
I (2780) file_iterator: File : echo_cn_end.wav
W (2783) app_sr: AFE Fetch Fail
I (2793) file_iterator: File : echo_cn_ok.wav
I (2799) file_iterator: File : echo_cn_wake.wav
I (2804) file_iterator: File : echo_en_end.wav
I (2817) file_iterator: File : echo_en_ok.wav
I (2824) file_iterator: File : echo_en_wake.wav
I (2828) file_iterator: File : Hi.mp3
I (2836) file_iterator: File : Hi.wav
I (2854) file_iterator: File : input.wav
I (2855) file_iterator: File : waitPlease.mp3
W (3083) app_sr: AFE Fetch Fail
W (3383) app_sr: AFE Fetch Fail
W (3683) app_sr: AFE Fetch Fail
W (3983) app_sr: AFE Fetch Fail
W (4283) app_sr: AFE Fetch Fail
W (4583) app_sr: AFE Fetch Fail
I (4768) wifi station: Total APs scanned = 9, ret:0
I (4768) wifi station: SSID             TP-LINK_39E8
I (4768) wifi station: SSID             CMCC-Zs9z
I (4772) wifi station: SSID             HONOR-GuiErGuiEr
I (4777) wifi station: SSID             HONOR-GuiErGuiEr_Wi-Fi5
I (4783) wifi station: SSID             HWCC5
I (4787) wifi station: SSID             Amor
I (4791) wifi station: SSID             CU_hRjE
I (4795) wifi station: SSID             太快了吧
I (4800) wifi station: SSID             401
I (4809) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (4810) wifi:state: init -> auth (b0)
I (4815) wifi:state: auth -> assoc (0)
I (4822) wifi:state: assoc -> run (10)
I (4831) wifi:connected with TP-LINK_39E8, aid = 4, channel 1, 40U, bssid = 34:96:72:26:39:e8
I (4831) wifi:security: WPA2-PSK, phy: bgn, rssi: -53
I (4835) wifi:pm start, type: 1

I (4836) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 25000, mt_pti: 0, mt_time: 10000
I (4850) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (4851) wifi:<ba-add>idx:0 (ifx:0, 34:96:72:26:39:e8), tid:0, ssn:0, winSize:64
W (4883) app_sr: AFE Fetch Fail
W (5183) app_sr: AFE Fetch Fail
W (5483) app_sr: AFE Fetch Fail
W (5783) app_sr: AFE Fetch Fail
I (5845) esp_netif_handlers: sta ip: 192.168.1.110, mask: 255.255.255.0, gw: 192.168.1.1
I (5846) wifi station: got ip:192.168.1.110
I (12146) app_sr: wakeword detected
I (12147) app_audio: ### record Start
I (12147) ui_ctrl: act_scr:0x3fcb5794, ui_ScreenSetup:0x3fcb5794
I (12152) ui_ctrl: Swich to panel[1]
I (12225) app_audio: frame_rate= 16000, ch=2, width=16
I (12227) I2S_IF: Pending out channel for in channel running
E (12245) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (12246) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (12250) I2S_IF: STD Mode 1 bits:16/16 channel:2 sample_rate:16000 mask:3
I (12282) app_sr: AFE_FETCH_CHANNEL_VERIFIED, channel index: 2

I (12350) Adev_Codec: Open codec device OK
E (12350) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (12353) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (12359) I2S_IF: STD Mode 0 bits:16/16 channel:2 sample_rate:16000 mask:3
I (12367) ES7210: Bits 16
I (12372) ES7210: Enable ES7210_INPUT_MIC1
I (12375) ES7210: Enable ES7210_INPUT_MIC2
I (12382) ES7210: Unmuted
I (12382) Adev_Codec: Open codec device OK
W (12454) app_sr: silence, frame:13
audio play end, 32034, 31 K
W (14750) app_sr: silence, frame:58
W (18333) app_sr: vad Time out
I (18333) app_audio: ESP_MN_STATE_TIMEOUT
I (18333) app_audio: ### record Stop, 104346 101K
audio load end, 11988, 11 K
E (18382) app_audio: [Start] start_openai, timestamp: 18382
I (18382) OpenAI: OpenAI create, version: 0.2.0
I (18385) ui_ctrl: Swich to panel[2]
I (18398) wifi:<ba-add>idx:1 (ifx:0, 34:96:72:26:39:e8), tid:6, ssn:0, winSize:64
I (18407) I2S_IF: Pending out channel for in channel running
E (18439) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (18439) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (18443) I2S_IF: STD Mode 1 bits:16/16 channel:2 sample_rate:16000 mask:3
I (18538) Adev_Codec: Open codec device OK
E (18538) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (18540) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (18546) I2S_IF: STD Mode 0 bits:16/16 channel:2 sample_rate:16000 mask:3
I (18554) ES7210: Bits 16
I (18560) ES7210: Enable ES7210_INPUT_MIC1
I (18563) ES7210: Enable ES7210_INPUT_MIC2
I (18571) ES7210: Unmuted
I (18572) Adev_Codec: Open codec device OK
I (18773) esp-x509-crt-bundle: Certificate validated
E (20002) OpenAI: cJSON_Parse failed!
I (20002) ui_ctrl: update listen speak
E (20003) app_audio: [End] start_openai, +offset:1621
I (22501) app_main: replay audio end

LHYhorion, 2023-12-17:
您好， 使用的是原有的 URL 还是作了更改呢？如果作了更改，请先确保该 URL 可用。

xiaoshanggys, 2023-12-17:
> 您好， 使用的是原有的 URL 还是作了更改呢？如果作了更改，请先确保该 URL 可用。

这个代码不是要自己添加URL和API吗
https://api.chatanywhere.com.cn
这是我使用的URL，验证过没问题
[image]

LHYhorion, 2023-12-18:
您好，经过测试发现您的 URL 应该是无效的。同时，我们发现目前确实 demo 更换 URL 会出现问题，我们已经修复了这个问题，会在下个版本发布。您可以使用一个有效 URL，在下个版本进行测试使用。

LHYhorion, 2023-12-21:
您好，我们进行了版本更新，并对一些 URL 进行了测试，目前已经支持更改 URL。但是对您的 URL 测试发现不可用，如果您想使用这个 demo，请尝试更换 URL，并再次进行测试。

tangyiyong, 2024-01-11:
你好，这个问题解决了吗？我git clone的最新版本的esp-idf和esp-box,还是碰到这个问题，
openai的语音识别成功了，但是执行到Chat Complletion的时候显示解析错误；所以对openai的访问应该能正常访问吧？
出错的位置：
    // OpenAI Chat Completion
    result = chatCompletion->message(chatCompletion, text, false);
    if (NULL == result) {
        ret = ESP_ERR_INVALID_RESPONSE;
        ESP_GOTO_ON_ERROR(ret, err, TAG, "[chatCompletion]: invalid response");
    }

输出如下：

I (56232) ES7210: Unmuted
I (56232) Adev_Codec: Open codec device OK
I (58079) esp-x509-crt-bundle: Certificate validated
I (59666) app_audio: Player IDLE
I (59669) I2S_IF: Pending out channel for in channel running
E (59690) i2s_common: i2s_channel_disable(1022): the channel has not been enabled yet
I (59690) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (59694) I2S_IF: STD Mode 1 bits:16/16 channel:2 sample_rate:16000 mask:3
I (59780) Adev_Codec: Open codec device OK
E (59782) i2s_common: i2s_channel_disable(1022): the channel has not been enabled yet
I (59783) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (59789) I2S_IF: STD Mode 0 bits:16/16 channel:2 sample_rate:16000 mask:3
I (59797) ES7210: Bits 16
I (59803) ES7210: Enable ES7210_INPUT_MIC1
I (59806) ES7210: Enable ES7210_INPUT_MIC2
I (59813) ES7210: Unmuted
I (59813) Adev_Codec: Open codec device OK
I (59817) app_main: replay audio end
I (66679) ui_ctrl: update reply question
I (66679) ui_ctrl: update listen speak
I (67403) esp-x509-crt-bundle: Certificate validated
E (72296) OpenAI: Error: cJSON_Parse failed!
E (72296) app_main: start_openai(109): [chatCompletion]: invalid response

我使用的openai的base_url是github上找的一个，在其他应用里可以使用：https://render.openai-forward.com//v1/
我自己搭建的的http的base_url访问出错；
api_key是自己申请的key，其他应用也能正常使用。

所以可能是什么问题？需要下载openAI这部分的接口代码分析？

espressif2022, 2024-01-12:
> 你好，这个问题解决了吗？我git clone的最新版本的esp-idf和esp-box,还是碰到这个问题， openai的语音识别成功了，但是执行到Chat Complletion的时候显示解析错误；所以对openai的访问应该能正常访问吧？ 出错的位置： // OpenAI Chat Completion result = [...]

不是很确定具体原因，上述 url 我本地测试确实是可以的，在设备端回复无效包，可尝试如下代理 url， 我测试是可以的
https://api.xiaoyukefu.com/v1/

tangyiyong, 2024-01-13:
> 不是很确定具体原因，上述 url 我本地测试确实是可以的，在设备端回复无效包，可尝试如下代理 url， 我测试是可以的 https://api.xiaoyukefu.com/v1/

这个地址的确可以正确访问，不知道是什么原因。有没有这个代理服务器部署的方法，或者部署的源代码之类？我部署一个。。。我有一台新加坡那边的服务器可以用。

espressif2022, 2024-01-16:
> 这个地址的确可以正确访问，不知道是什么原因。有没有这个代理服务器部署的方法，或者部署的源代码之类？我部署一个。。。我有一台新加坡那边的服务器可以用。

抱歉，上述 url 也是其他渠道看到的，并不是我们自己搭的。

ESP-Mars, 2024-03-14:
Temporarily closed due to the extended period of time with no response. Should you have additional questions or concerns, don't hesitate to reopen the issue.

closed 2024-03-14 by ESP-Mars