espressif/esp-box #124: Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

GYY-110, 2023-12-20:
当我在运行espbox中的例子chatgpt_demo时，唤醒esp后问了问题会回复：I think about this question，please... ,然后就一直重连，显示下面的日志，这个如何处理啊！

I (186668) app_sr: wakeword detected
I (186668) app_audio: ### record Start
I (186668) ui_ctrl: act_scr:0x3fcb559c, ui_ScreenSetup:0x3fcb559c
I (186674) ui_ctrl: Swich to panel[1]
I (186745) app_audio: frame_rate= 16000, ch=2, width=16
I (186747) I2S_IF: Pending out channel for in channel running
E (186759) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (186759) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (186764) I2S_IF: STD Mode 1 bits:16/16 channel:2 sample_rate:16000 mask:3
I (186782) app_sr: AFE_FETCH_CHANNEL_VERIFIED, channel index: 2

I (186791) Adev_Codec: Open codec device OK
E (186791) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (186794) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (186800) I2S_IF: STD Mode 0 bits:16/16 channel:2 sample_rate:16000 mask:3
I (186809) Adev_Codec: Open codec device OK
I (195339) app_audio: ESP_MN_STATE_TIMEOUT
I (195339) app_audio: ### record Stop, 139128 135K
I (195371) OpenAI: OpenAI create, version: 0.3.0
I (195372) app_audio: Player PLAYING
I (195372) ui_ctrl: Swich to panel[2]
I (195411) I2S_IF: Pending out channel for in channel running
E (195422) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (195423) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (195429) I2S_IF: STD Mode 1 bits:16/16 channel:2 sample_rate:16000 mask:3
I (195456) Adev_Codec: Open codec device OK
E (195456) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (195459) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (195465) I2S_IF: STD Mode 0 bits:16/16 channel:2 sample_rate:16000 mask:3
I (195474) Adev_Codec: Open codec device OK
I (196225) esp-x509-crt-bundle: Certificate validated
I (198915) app_audio: Player IDLE
I (198916) I2S_IF: Pending out channel for in channel running
E (198928) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (198928) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (198933) I2S_IF: STD Mode 1 bits:16/16 channel:2 sample_rate:16000 mask:3
I (198970) Adev_Codec: Open codec device OK
E (198970) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (198973) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (198979) I2S_IF: STD Mode 0 bits:16/16 channel:2 sample_rate:16000 mask:3
I (198988) Adev_Codec: Open codec device OK
I (198991) app_main: replay audio end
I (209025) ui_ctrl: update reply question
I (209026) ui_ctrl: update listen speak
I (209931) esp-x509-crt-bundle: Certificate validated
E (215592) OpenAI: Error: cJSON_Parse failed!
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x42009d8a  PS      : 0x00060130  A0      : 0x8200a4b8  A1      : 0x3fcd6450
0x42009d8a: start_openai at G:/project/ESP-IDF/code/yuanchu/esp-box-yanchu/esp-box-master/examples/chatgpt_demo/main/main.c:91

A2      : 0x3fcd8de8  A3      : 0x42062e60  A4      : 0x42064288  A5      : 0x3c126e00
0x42062e60: OpenAI_ChatCompletionMessage at G:/project/ESP-IDF/code/yuanchu/esp-box-yanchu/esp-box-master/examples/chatgpt_demo/managed_components/espressif__openai/OpenAI.c:1081

0x42064288: OpenAI_AudioTranscriptionFile at G:/project/ESP-IDF/code/yuanchu/esp-box-yanchu/esp-box-master/examples/chatgpt_demo/managed_components/espressif__openai/OpenAI.c:2028

A6      : 0x3fca3290  A7      : 0x00000000  A8      : 0x82009d8a  A9      : 0x3fcd6410
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3fcd82b0
A14     : 0x3fcd9208  A15     : 0x4206231c  SAR     : 0x00000017  EXCCAUSE: 0x0000001c
0x4206231c: OpenAI_Post at G:/project/ESP-IDF/code/yuanchu/esp-box-yanchu/esp-box-master/examples/chatgpt_demo/managed_components/espressif__openai/OpenAI.c:2489

EXCVADDR: 0x00000008  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffa
0x400556d5: strlen in ROM

0x400556e5: strlen in ROM

Backtrace: 0x42009d87:0x3fcd6450 0x4200a4b5:0x3fcd6480
0x42009d87: start_openai at G:/project/ESP-IDF/code/yuanchu/esp-box-yanchu/esp-box-master/examples/chatgpt_demo/main/main.c:90

0x4200a4b5: sr_handler_task at G:/project/ESP-IDF/code/yuanchu/esp-box-yanchu/esp-box-master/examples/chatgpt_demo/main/app/app_audio.c:320

ELF file SHA256: 5fbe5f49abb3ac5e

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375d8b
0x40375d8b: esp_restart_noos_dig at G:/Espressif/frameworks/esp-idf-v5.1.2/components/esp_system/port/esp_system_chip.c:57 (discriminator 1)

GYY-110, 2023-12-20:
[image]

LHYhorion, 2023-12-21:
请问您使用的是默认的 URL 吗？网络环境是国内吗？

GYY-110, 2023-12-21:
@LHYhorion 
最开始使用的时默认的URL，连不上，换了下面的URL后就这样了
[image]

LHYhorion, 2023-12-22:
> @LHYhorion 最开始使用的时默认的URL，连不上，换了下面的URL后就这样了 [...]

感谢反馈。目前使用默认 openai 的 URL 是可以的，但是国内网络环境不可用。经过测试发现更换 URL 后，可能会发生崩溃的情况，后续我们会修复这个问题。

GYY-110, 2023-12-22:
> 感谢反馈。目前使用默认 openai 的 URL 是可以的，但是国内网络环境不可用。经过测试发现更换 URL 后，可能会发生崩溃的情况，后续我们会修复这个问题。

感谢回复，我试着更换了URL，成功连接到了，但是只能持续3到4次提问，会报以下的错误，请问一下这个是为什么呀
[image]

LHYhorion, 2023-12-22:
> 感谢回复，我试着更换了URL，成功连接到了，但是只能持续3到4次提问，会报以下的错误，请问一下这个是为什么呀 [...]

这个 demo 使用 openai 官方 URL 不会出现这个问题，但是由于国内网络环境不能使用，因此我们支持用户自己配置 URL。但是因为目前代理 URL 的兼容没有做好，所以可能会发生这个问题。后续的版本会修复这个问题，再次感谢您的反馈。

espressif2022, 2024-01-18:
你好，我将关闭此 ISSUE， 如还有问题，可参考 https://github.com/espressif/esp-box/issues/123
谢谢！

closed 2024-01-18 by espressif2022