espressif/esp-box #127: Development with Arduino IDE

tahunus, 2024-01-03:
What libraries are available to access the on-board audio input/output and accelerometer?
What libraries for the sensors on the ESP32-S3-BOX-3-SENSOR (Temp&Hum , IR Emitter and Receiver, Radar )

LHYhorion, 2024-01-09:
Hello, currently there is no official Arduino library available for audio and sensor functionalities. Updates in this regard may be released in the future. If you are interested in developing such content, you can explore relevant libraries in the open-source community.

tahunus, 2024-01-11:
@LHYhorion thanks for your response.

To start looking for relevant libraries, it would be very helpful to understand a couple of things. 

1. What is the implemented pinout of ESP32S3 in the Box-3?  There's this info on[ esp-bsp/bsp/esp-box-3](https://github.com/espressif/esp-bsp/blob/master/bsp/esp-box-3/include/bsp/esp-box-3.h), then there's[ this variant](https://github.com/espressif/arduino-esp32/blob/master/variants/esp32s3box/pins_arduino.h) in the arduino-esp32 core which differ from each other. 

2. It is not clear from the [hardware specs of the BOX-3](https://github.com/espressif/esp-box/blob/master/docs/hardware_overview/esp32_s3_box_3/hardware_overview_for_box_3.md#technical-specification) if the sensors (ADC, DAC) for audio IN/OUT and the accel/gyro have direct pins or are connected via I2C, and if so, what is their I2C address

Many thanks!

LHYhorion, 2024-01-11:
You can find all the hardware information about ESP-S3-BOX-3 at https://github.com/espressif/esp-box/tree/master/hardware/SCH_ESP32-S3-BOX-3_V1.0. For audio input/output, we use ES8311 and ES7210, and we do not use the S3's own ADC.

ESP-Mars, 2024-03-14:
Temporarily closed due to the extended period of time with no response. Should you have additional questions or concerns, don't hesitate to reopen the issue.

closed 2024-03-14 by ESP-Mars