espressif/esp-box #128: esp32-box-lite  chatgpt-demo

gzlwz, 2024-01-10:
show "invalid_request_error"
===============log===========
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x18e4
load:0x403c9700,len:0x4
load:0x403c9704,len:0xda4
load:0x403cc700,len:0x3084
entry 0x403c9934
I (26) boot: ESP-IDF v5.1.2-dirty 2nd stage bootloader
I (27) boot: compile time Jan 10 2024 19:27:30
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.1
I (34) qio_mode: Enabling default flash chip QIO
I (39) boot.esp32s3: Boot SPI Speed : 80MHz
I (44) boot.esp32s3: SPI Mode       : QIO
I (49) boot.esp32s3: SPI Flash Size : 16MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (77) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (85) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (92) boot:  3 ota_0            OTA app          00 10 00010000 00600000
I (100) boot:  4 uf2              factory app      00 00 00700000 00200000
I (107) boot:  5 storage          Unknown data     01 82 00900000 00200000
I (115) boot:  6 model            Unknown data     01 82 00b00000 003e8000
I (123) boot: End of partition table
I (127) esp_image: segment 0: paddr=00010020 vaddr=3c120020 size=2306c0h (2295488) map
I (483) esp_image: segment 1: paddr=002406e8 vaddr=3fc9b900 size=07198h ( 29080) load
I (489) esp_image: segment 2: paddr=00247888 vaddr=40374000 size=08790h ( 34704) load
I (496) esp_image: segment 3: paddr=00250020 vaddr=42000020 size=11b59ch (1160604) map
I (673) esp_image: segment 4: paddr=0036b5c4 vaddr=4037c790 size=0f128h ( 61736) load
I (686) esp_image: segment 5: paddr=0037a6f4 vaddr=600fe010 size=00004h (     4) load
I (696) boot: Loaded app from partition at offset 0x10000
I (697) boot: Disabling RNG early entropy source...
I (708) cpu_start: Multicore app
I (708) octal_psram: vendor id    : 0x0d (AP)
I (709) octal_psram: dev id       : 0x02 (generation 3)
I (712) octal_psram: density      : 0x03 (64 Mbit)
I (717) octal_psram: good-die     : 0x01 (Pass)
I (722) octal_psram: Latency      : 0x01 (Fixed)
I (728) octal_psram: VCC          : 0x01 (3V)
I (733) octal_psram: SRF          : 0x01 (Fast Refresh)
I (739) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (744) octal_psram: BurstLen     : 0x01 (32 Byte)
I (750) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (756) octal_psram: DriveStrength: 0x00 (1/1)
I (762) MSPI Timing: PSRAM timing tuning index: 4
I (767) esp_psram: Found 8MB PSRAM device
I (771) esp_psram: Speed: 80MHz
I (853) mmu_psram: Instructions copied and mapped to SPIRAM
I (854) cpu_start: Pro cpu up.
I (854) cpu_start: Starting app cpu, entry point is 0x403758cc
0x403758cc: call_start_cpu1 at D:/Espressif50-ide/frameworks/esp-idf-v5.1.2/components/esp_system/port/cpu_start.c:157

I (0) cpu_start: App cpu up.
I (1101) esp_psram: SPI SRAM memory test OK
I (1110) cpu_start: Pro cpu start user code
I (1110) cpu_start: cpu freq: 240000000 Hz
I (1110) cpu_start: Application information:
I (1113) cpu_start: Project name:     chatgpt_demo
I (1118) cpu_start: App version:      v0.5.0-44-g88cca17
I (1124) cpu_start: Compile time:     Jan 10 2024 19:27:10
I (1131) cpu_start: ELF file SHA256:  7563b3c8abc0db92...
I (1137) cpu_start: ESP-IDF:          v5.1.2-dirty
I (1142) cpu_start: Min chip rev:     v0.0
I (1147) cpu_start: Max chip rev:     v0.99
I (1152) cpu_start: Chip rev:         v0.1
I (1157) heap_init: Initializing. RAM available for dynamic allocation:
[... 177 lines cut ...]
E (25102) OpenAI: ./managed_components/espressif__openai/OpenAI.c:2080 (OpenAI_AudioTranscriptionFile):Empty result!

I (25113) ui_ctrl: update listen speak

E (25117) app_main: start_openai(85): [audioTranscription]: invalid url

I (27873) app_audio: Player IDLE
I (27876) I2S_IF: Pending out channel for in channel running

E (27884) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet

I (27884) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (27888) I2S_IF: STD Mode 1 bits:16/16 channel:2 sample_rate:16000 mask:3
I (27915) Adev_Codec: Open codec device OK

E (27915) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet

I (27917) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (27924) I2S_IF: STD Mode 0 bits:16/16 channel:2 sample_rate:16000 mask:3
I (27939) Adev_Codec: Open codec device OK
I (27939) app_main: replay audio end
=====================================
api key and base url has pass through chatbox and  esp32 Audio

but can't run in this case.

espressif2022, 2024-01-11:
can you try this url with `curl` ? I tried it, and the result is  `401 Authorization Required`

```
request:
curl https://api.chatanywhere.com.cn/v1/chat/completions \
  -H "Content-Type: application/json" \
  -H "Authorization: Bearer  $OPENAI_API_KEY" \
  -d '{
    "model": "gpt-3.5-turbo",
    "messages": [
      {
        "role": "system",
        "content": "You are a helpful assistant."
      },
      {
        "role": "user",
        "content": "Hello!"
      }
    ]
  }'
result:
<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html>
<head><title>401 Authorization Required</title></head>
<body>
<h1>401 Authorization Required</h1>
<p>This server could not verify that you are authorized to access the document requested.<hr/>Powered by Tengine</body>
</html>
```

```
request:
curl https://api.openai.com/v1/chat/completions \
  -H "Content-Type: application/json" \
  -H "Authorization: Bearer  $OPENAI_API_KEY" \
  -d '{
    "model": "gpt-3.5-turbo",
    "messages": [
      {
        "role": "system",
        "content": "You are a helpful assistant."
      },
      {
        "role": "user",
        "content": "Hello!"
      }
    ]
  }'
result:
{
  "id": "chatcmpl-8fetdWnNxgrsXs5JknHW5W4SZLNUh",
  "object": "chat.completion",
  "created": 1704939345,
  "model": "gpt-3.5-turbo-0613",
  "choices": [
    {
      "index": 0,
      "message": {
        "role": "assistant",
        "content": "Hello! How may I assist you today?"
      },
      "logprobs": null,
      "finish_reason": "stop"
    }
  ],
  "usage": {
    "prompt_tokens": 19,
    "completion_tokens": 9,
    "total_tokens": 28
  },
  "system_fingerprint": null
}
```

gzlwz, 2024-01-11:
It is not opai api key 
the url & api key come from the next link:
https://github.com/chatanywhere/GPT_API_free?tab=readme-ov-file

免费版支持gpt-3.5-turbo, embedding, gpt-4。其中gpt-4由于价格过高，每24小时限制10次调用，且不支持流式传输。需要更稳定快速的gpt-4请使用付费版。
转发Host1: https://api.chatanywhere.tech (国内中转，延时更低，host1和host2二选一)
转发Host2: https://api.chatanywhere.com.cn (国内中转，延时更低，host1和host2二选一)
转发Host3: https://api.chatanywhere.cn (国外使用,国内需要全局代理)
[Issues](https://github.com/chatanywhere/GPT_API_free/issues)
该API Key用于转发API，需要将Host改为api.chatanywhere.tech(国内首选)或者api.chatanywhere.cn(国外使用，国内需要全局代理)。

gzlwz, 2024-01-11:
Because we can't direct access the openai in mainland.

gzlwz, 2024-01-11:
I (473880) I2S_IF: Pending out channel for in channel running
E (473892) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (473892) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (473896) I2S_IF: STD Mode 1 bits:16/16 channel:2 sample_rate:16000 mask:3
I (473930) Adev_Codec: Open codec device OK

E (473930) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet

I (473932) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (473939) I2S_IF: STD Mode 0 bits:16/16 channel:2 sample_rate:16000 mask:3
I (473965) Adev_Codec: Open codec device OK
I (473983) esp-x509-crt-bundle: Certificate validated
I (476177) ui_ctrl: update listen speak

E (476178) app_main: start_openai(85): [audioTranscription]: invalid url

I (477400) app_audio: Player IDLE
I (477402) I2S_IF: Pending out channel for in channel running
E (477417) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (477417) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (477421) I2S_IF: STD Mode 1 bits:16/16 channel:2 sample_rate:16000 mask:3
I (477449) Adev_Codec: Open codec device OK
E (477449) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (477451) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (477459) I2S_IF: STD Mode 0 bits:16/16 channel:2 sample_rate:16000 mask:3
I (477470) Adev_Codec: Open codec device OK
I (477471) app_main: replay audio end

espressif2022, 2024-01-12:
> Because we can't direct access the openai in mainland.

I know this, so you need to make sure that the proxy server is accessible on the PC first, I tested it locally, and your url curl commands are inaccessible.
It's definitely not accessible on the device.

gzlwz, 2024-01-12:
yeah,　Ｉ　ｈａｄ　ａｌｒｅａｄｙ　．

curl https://api.chatanywhere.com.cn/v1/chat/completions \
> -H "Content-Type: application/json" \
>   -H "Authorization: Bearer $OPENAI_API_KEY" \
>   -d '{
>     "model": "gpt-3.5-turbo",
>     "messages": [
>       {
>         "role": "system",
>         "content": "You [...]
{"id":"chatcmpl-8gEDp3nAMzqGh05Lh15SokKAb2s7e",
"object":"chat.completion",
"created":1705075137,
"model":"gpt-3.5-turbo-1106",
"choices":[
{"index":0,"
message":{"role":"assistant","content":"Hello! How can I assist you today?"},
"logprobs":nu

dujianmin, 2024-01-13:
E (22066) HTTP_CLIENT: Error parse url audio/transcriptions
E (22071) HTTP_CLIENT: Failed to set URL

LHYhorion, 2024-01-30:
> yeah,　Ｉ　ｈａｄ　ａｌｒｅａｄｙ　．
>
> curl https://api.chatanywhere.com.cn/v1/chat/completions \

I tried your URL, and it showed a 401 error message

LHYhorion, 2024-01-30:
> yeah,　Ｉ　ｈａｄ　ａｌｒｅａｄｙ　．
>
> curl https://api.chatanywhere.com.cn/v1/chat/completions \

Hello, I tested your URL and received a 401 warning. I suggest you try using a different URL for testing.

LHYhorion, 2024-01-30:
Perhaps you can try using the URL https://api.xiaoyukefu.com/v1/. It has been tested and is functional.

gzlwz, 2024-02-07:
无法访问此网站网址为 https://api.xiaoyukefu.com/v1/ 的网页可能暂时无法连接，或者它已永久性地移动到了新网址。

LHYhorion, 2024-02-08:
You need to add this URL to CONFIG.INI or configure it in menuconfig. You cannot directly open and use https://api.xiaoyukefu.com/v1/.

ESP-Mars, 2024-03-14:
Temporarily closed due to the extended period of time with no response. Should you have additional questions or concerns, don't hesitate to reopen the issue.

closed 2024-03-14 by ESP-Mars