espressif/esp-box #130: [Question] Does it work with ZX3D95CE01S-TR-4848. board?

gschanuel, 2024-01-17:
Hello there
I recently bought this board from Aliexpess https://www.aliexpress.us/item/3256805699742641.htm
http://doc.panel-tag.cn/_static/espboard/ESP32S3/ZX3D95CE01S-TR-4848.pdf

I would like to know how to make this firmware work with it
When I upload the firmware to it I cannot make it boot

```
sudo esptool.py -p /dev/ttyUSB0 -b 1152000 --chip esp32s3 write_flash 0x0 ESP-BOX_Demo_EN_V0.5.0.bin
$ sudo esptool.py -p /dev/ttyUSB0 -b 1152000 --chip esp32s3 write_flash 0x0 ESP-BOX_Demo_EN_V0.5.0.bin 
esptool.py v4.6.2
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32-S3 (revision v0.2)
Features: WiFi, BLE
Crystal is 40MHz
MAC: ec:da:3b:9b:8b:fc
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 1152000
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00b7ffff...
Compressed 12057588 bytes to 6269421...
Wrote 12057588 bytes (6269421 compressed) at 0x00000000 in 98.7 seconds (effective 977.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```
After the reset, minicom displays this:
```
Welcome to minicom 2.9

OPTIONS: I18n 
Compiled on Sep 23 2023, 19:55:08.
Port /dev/ttyUSB0, 13:24:10

Press CTRL-A Z for help on special keys                                                                    
                                                                                                           
ESP-ROM:esp32s3-20210327                                                                                   
Build:Mar 27 2021                                                                                          
rst:0x1 (POWERON),boot:0x2 (DOWNLOAD(USB/UART0))                                                           
waiting for download
```

espressif2022, 2024-01-18:
I'm sorry, this board can't use this fw, this is not ESP-BOX.

espressif2022, 2024-02-29:
Hello，I'll close this PR. If you have any question of esp-box, you can create a new one, thanks.

closed 2024-02-29 by espressif2022