espressif/esp-box #131: Feb 2024, cannot buy one anywhere, when will you have them?

dac2020, 2024-02-21:
I was going to order one from Amazon but it shows nothing available until April. I looked all over and see no stick anywhere. 
When will these be available for sale again?

ESP-Mars, 2024-02-22:
If you don't need many accessories, you can purchase the [ESP32-S3-BOX-3B](https://www.amazon.com/Espressif-ESP32-S3-BOX-3B-Development-Board/dp/B0CSJC5BX6/ref=sr_1_1?crid=3E9INZGU82PSO&dib=eyJ2IjoiMSJ9.a_tBpTdNBu4KhZ2SPdlEMhAZOJ9isPbL9wC3OTFVJCrRWoYPrJ4EMreKFIo5GMUP1usnBHOmaZd_RruBOQXFhbh2ekSEvF8pdyrH56V3JyhowcOl1bX3_yTnROW6XcqA9eu8-Vekhe-198E363FtZt3qO-Jb9FX3QimRcIsV7vOg6zXnOCFPaJL0UxccYqePYAi8bjcbSAyw-Crrum1jR5OXBW-PXzVLU7uOb-KYUws.ByTcTETB6QkIEnXqIQtxakhrw6HtfV9AG0WyS184u2Y&dib_tag=se&keywords=ESP32-S3-BOX-3B%2BDevelopment%2BBoard&qid=1708570377&sprefix=%2Caps%2C459&sr=8-1&th=1) as an alternative. The ESP32-S3-BOX-3 comes in two different versions: ESP32-S3-BOX-3 and ESP32-S3-BOX-3B. The ESP32-S3-BOX-3 represents the standard edition, equipped with four blue accessories, while the ESP32-S3-BOX-3B provides fewer accessories to meet the needs of beginners who don't require too much unnecessary hardware. Here are the distinctions between the two versions:
[image]

Please click [here](https://www.espressif.com/en/products/devkits), scroll down, and go to ESP32-S3-BOX-3 to explore all available purchase channels.

dac2020, 2024-02-22:
I want an ESP32 WiFi display that can receive and display messages from MQTT. 
However, it's not clear if I need the bracket or not for this basic function.

ESP-Mars, 2024-02-23:
@dac2020 Considering the requirements you've described, the ESP32-S3-BOX-3B should be sufficient. Could you please specify your country? We can assist you in finding the appropriate purchasing channels.

dac2020, 2024-02-23:
USA, typically order from Amazon.

ESP-Mars, 2024-02-27:
@dac2020 We apologize for the insufficient availability of the product. We are actively expediting production. Please keep an eye on platforms such as AliExpress, Mouser, and Adafruit for updates.

1544k, 2024-02-27:
Not sure if I just got particularly lucky, although  higher lead time was indicated, had my backorder on digikey fullfilled within a month, got it last week shipped within two days to Germany.

dac2020, 2024-03-03:
Personally, I only purchase from Amazon because it's very fast delivery and returns if there are problems. 
Buying overseas takes much longer and returning an item not working is much harder and longer.

ESP-Mars, 2024-04-08:
The BOX-3 is in urgent production, and a batch of goods will be available on AliExpress and Amazon next week.

raminf, 2024-04-18:
I am also looking for the BOX-3. Specifically need one with the microphone and a speaker for a voice recognition application. 

It's not clear from the table above if the 3B has those or not. I went ahead and ordered a 3B on AliExpress, just in case. Will the 3B be enough or should I wait for the BOX-3? Thanks!

ESP-Mars, 2024-04-19:
@raminf If you specifically need one with a microphone and speaker, the BOX-3B is sufficient, as the BOX-3 only has some additional sensor accessories and adapter boards compared to the BOX-3B.

ESP-Mars, 2024-04-19:
> USA, typically order from Amazon.

@dac2020 Arrival on Amazon: https://www.amazon.com/dp/B0CL6HD8JX?ref=myi_title_dp&th=1

touchgadget, 2024-04-19:
@ESP-Mars All sold?
[image]

ESP-Mars, 2024-04-22:
@touchgadget  My Amazon shows that there is stock available for delivery to "Deliver to Houston 77022.
[image]

touchgadget, 2024-04-22:
@ESP-Mars 
Thanks, I see the boards are available now!

ESP-Mars, 2024-07-23:
Temporarily closed the issue. Should you have additional questions or concerns, don't hesitate to reopen the issue.

closed 2024-07-23 by ESP-Mars