espressif/esp-box #132: idf flashed factory_demo crashes

1544k, 2024-02-26:
The box 3 is a fantastic product, keep up the good work!
I'm having an issue flashing the factory_demo to the box 3 with idf, the esp crashes. When flashing from launchpad it works fine. Also I'm able to flash the other box or skainet examples with idf without issues, so I guess its a config issue. CONFIG_BSP_BOARD_ESP32_S3_BOX_3=y
is correctly set.

I'd be grateful for any pointers that may resolve the issue.

<details>
    <summary> idf-monitor output </summary>

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x40056f64
0x40056f64: memcpy in ROM

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x1870
load:0x403c9700,len:0x4
load:0x403c9704,len:0xd88
load:0x403cc700,len:0x3070
entry 0x403c9934
I (19) boot: ESP-IDF v5.1-dirty 2nd stage bootloader
I (19) boot: compile time Feb 26 2024 17:52:24
I (19) boot: Multicore bootloader
I (19) boot: chip revision: v0.2
I (20) qio_mode: Enabling QIO for flash chip GD
I (20) boot.esp32s3: Boot SPI Speed : 80MHz
I (20) boot.esp32s3: SPI Mode       : QIO
I (20) boot.esp32s3: SPI Flash Size : 16MB
I (21) boot: Enabling RNG early entropy source...
I (21) boot: Partition Table:
I (21) boot: ## Label            Usage          Type ST Offset   Length
I (21) boot:  0 sec_cert         Unknown data     01 06 0000d000 00003000
I (22) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (22) boot:  2 otadata          OTA data         01 00 00016000 00002000
I (23) boot:  3 phy_init         RF data          01 01 00018000 00001000
I (23) boot:  4 fctry            WiFi data        01 02 00019000 00006000
I (23) boot:  5 ota_0            OTA app          00 10 00020000 0041a000
I (24) boot:  6 storage          Unknown data     01 82 0043a000 0028a000
I (24) boot:  7 model            Unknown data     01 82 006c4000 00866000
I (25) boot: End of partition table
I (25) boot: No factory image, trying OTA 0
I (25) esp_image: segment 0: paddr=00020020 vaddr=3c170020 size=22e16ch (2285932) map
I (372) esp_image: segment 1: paddr=0024e194 vaddr=3fca3c00 size=01e84h (  7812) load
I (374) esp_image: segment 2: paddr=00250020 vaddr=42000020 size=16ac20h (1485856) map
I (599) esp_image: segment 3: paddr=003bac48 vaddr=3fca5a84 size=05f7ch ( 24444) load
I (604) esp_image: segment 4: paddr=003c0bcc vaddr=40378000 size=1bbdch (113628) load
I (627) esp_image: segment 5: paddr=003dc7b0 vaddr=50000000 size=00004h (     4) load
I (627) esp_image: segment 6: paddr=003dc7bc vaddr=600fe000 size=0005ch (    92) load
I (639) boot: Loaded app from partition at offset 0x20000
I (660) boot: Set actual ota_seq=1 in otadata[0]
I (660) boot: Disabling RNG early entropy source...
I (661) cpu_start: Multicore app
I (661) octal_psram: vendor id    : 0x0d (AP)
I (661) octal_psram: dev id       : 0x03 (generation 4)
I (662) octal_psram: density      : 0x05 (128 Mbit)
I (662) octal_psram: good-die     : 0x01 (Pass)
I (662) octal_psram: Latency      : 0x01 (Fixed)
I (662) octal_psram: VCC          : 0x00 (1.8V)
I (662) octal_psram: SRF          : 0x01 (Fast Refresh)
I (663) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (663) octal_psram: BurstLen     : 0x01 (32 Byte)
I (663) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (664) octal_psram: DriveStrength: 0x00 (1/1)
I (665) esp_psram: Found 16MB PSRAM device
I (665) esp_psram: Speed: 80MHz
I (665) cpu_start: Pro cpu up.
I (665) cpu_start: Starting app cpu, entry point is 0x40379a9c
0x40379a9c: call_start_cpu1 at /home/user/esp/esp-idf/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.
I (1308) esp_psram: SPI SRAM memory test OK
I (1317) cpu_start: Pro cpu start user code
I (1317) cpu_start: cpu freq: 240000000 Hz
I (1317) cpu_start: Application information:
I (1317) cpu_start: Project name:     factory_demo
I (1317) cpu_start: App version:      v0.5.0-55-gee2160d
I (1318) cpu_start: Compile time:     Feb 26 2024 17:57:39
I (1318) cpu_start: ELF file SHA256:  8189a328...
I (1318) cpu_start: ESP-IDF:          v5.1-dirty
I (1318) cpu_start: Min chip rev:     v0.0
I (1318) cpu_start: Max chip rev:     v0.99 
I (1318) cpu_start: Chip rev:         v0.2
I (1319) heap_init: Initializing. RAM available for dynamic allocation:
I (1319) heap_init: At 3FCB2240 len 000374D0 (221 KiB): DRAM
I (1319) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (1319) heap_init: At 600FE05C len 00001FA4 (7 KiB): RTCRAM
I (1320) esp_psram: Adding pool of 16384K of PSRAM memory to heap allocator
I (1320) spi_flash: detected chip: gd
I (1321) spi_flash: flash io: qio
I (1321) sleep: Configure to isolate all GPIO pins in sleep state
I (1322) sleep: Enable automatic switching of GPIO sleep configuration
I (1322) pm: Frequency switching config: CPU_MAX: 240, APB_MAX: 80, APB_MIN: 40, Light sleep: DISABLED
I (1322) coexist: coex firmware version: ebddf30
I (1323) coexist: coexist rom version e7ae62f
I (1323) app_start: Starting scheduler on CPU0
I (0) app_start: Starting scheduler on CPU1
I (1323) main_task: Started on CPU0
I (1324) esp_psram: Reserving pool of 8K of internal memory for DMA/internal allocations
I (1324) main_task: Calling app_main()
I (1325) main: Compile time: Feb 26 2024 17:57:56
W (1489) settings: Not found, Set to default
I (1489) settings: Saving settings
I (1700) ESP-BOX-3: Partition size: total: 2439971, used: 1376986
I (1700) LVGL: Starting LVGL task
I (1701) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1702) gpio: GPIO[48]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1702) ili9341: LCD panel create success, version: 1.2.0
W (1823) ili9341: The 36h command has been used and will be overwritten by external initialization sequence
W (1823) ili9341: The 3Ah command has been used and will be overwritten by external initialization sequence

assert failed: lvgl_port_add_disp esp_lvgl_port.c:258 (disp_cfg->buffer_size > 0)

Backtrace: 0x4037a012:0x3fcb5c70 0x4038b30d:0x3fcb5c90 0x40390add:0x3fcb5cb0 0x420baf67:0x3fcb5dd0 0x420432e1:0x3fcb5e10 0x4204360b:0x3fcb5e60 0x42008e25:0x3fcb5e80 0x4216748a:0x3fcb5ee0 0x4038ba8e:0x3fcb5f10
0x4037a012: panic_abort at /home/user/esp/esp-idf/components/esp_system/panic.c:452
0x4038b30d: esp_system_abort at /home/user/esp/esp-idf/components/esp_system/port/esp_system_chip.c:84
0x40390add: __assert_func at /home/user/esp/esp-idf/components/newlib/assert.c:81
0x420baf67: lvgl_port_add_disp at /home/user/esp/esp-box/examples/factory_demo/managed_components/espressif__esp_lvgl_port/esp_lvgl_port.c:258 (discriminator 1)
0x420432e1: bsp_display_lcd_init at /home/user/esp/esp-box/examples/factory_demo/managed_components/espressif__esp-box-3/esp-box-3.c:460 (discriminator 2)
0x4204360b: bsp_display_start_with_config at /home/user/esp/esp-box/examples/factory_demo/managed_components/espressif__esp-box-3/esp-box-3.c:576 (discriminator 2)
0x42008e25: app_main at /home/user/esp/esp-box/examples/factory_demo/main/main.c:114 (discriminator 2)
0x4216748a: main_task at /home/user/esp/esp-idf/components/freertos/app_startup.c:208 (discriminator 13)
0x4038ba8e: vPortTaskWrapper at /home/user/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

ELF file SHA256: 8189a328

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x40379e98
0x40379e98: esp_restart_noos_dig at /home/user/esp/esp-idf/components/esp_system/port/esp_system_chip.c:57 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x1870
load:0x403c9700,len:0x4
load:0x403c9704,len:0xd88
load:0x403cc700,len:0x3070
entry 0x403c9934
</details>

espressif2022, 2024-02-27:
[image]

Hello, I'm sorry. The BSP data structure has recently been adjusted, and the project has not been adapted in time. 
You can write this version to 1.1.2 first, and 1.1.3 will adapt later.
Thanks!

1544k, 2024-02-27:
That works, thanks for the quick answer

closed 2024-02-27 by 1544k