espressif/esp-box #133: Support for new I2C master driver.

ammaree, 2024-03-01:
When is support planned for the new I2C master driver?

espressif2022, 2024-03-05:
Hello, do you mean the file `esp_lcd_panel_io_i2c_v1.c` and `esp_lcd_panel_io_i2c_v2.c`. 
This interface should not be used in the application, compatibility has been done at the io_handle layer. 
Can I learn more about your needs?

ammaree, 2024-03-06:
I am building our application for the ESP-BOX using latest VSCode, IDF Master (v5.3-dev-2381-g9f4e8eb0cd).
Application compiles and links, only complains about:
`
/Users/andremaree/DevSpace/z-comp/gui_basic/gui_lvgl.cpp:87:33: warning: invalid suffix on literal; C++11 requires a space between literal and string macro [-Wliteral-suffix]
   87 |     lv_label_set_text(lv_label, "#FF0000 "LV_SYMBOL_BELL" Hello world Espressif and LVGL "LV_SYMBOL_BELL"#\n#FF9400 "LV_SYMBOL_WARNING" For simplier initialization, use BSP "LV_SYMBOL_WARNING" #");
      |                                 ^
/Users/andremaree/DevSpace/z-comp/gui_basic/gui_lvgl.cpp:87:57: warning: invalid suffix on literal; C++11 requires a space between literal and string macro [-Wliteral-suffix]
   87 |     lv_label_set_text(lv_label, "#FF0000 "LV_SYMBOL_BELL" Hello world Espressif and LVGL "LV_SYMBOL_BELL"#\n#FF9400 "LV_SYMBOL_WARNING" For simplier initialization, use BSP "LV_SYMBOL_WARNING" #");
      |                                                         ^
/Users/andremaree/DevSpace/z-comp/gui_basic/gui_lvgl.cpp:87:105: warning: invalid suffix on literal; C++11 requires a space between literal and string macro [-Wliteral-suffix]
   87 |     lv_label_set_text(lv_label, "#FF0000 "LV_SYMBOL_BELL" Hello world Espressif and LVGL "LV_SYMBOL_BELL"#\n#FF9400 "LV_SYMBOL_WARNING" For simplier initialization, use BSP "LV_SYMBOL_WARNING" #");
      |                                                                                                         ^
/Users/andremaree/DevSpace/z-comp/gui_basic/gui_lvgl.cpp:87:135: warning: invalid suffix on literal; C++11 requires a space between literal and string macro [-Wliteral-suffix]
   87 |     lv_label_set_text(lv_label, "#FF0000 "LV_SYMBOL_BELL" Hello world Espressif and LVGL "LV_SYMBOL_BELL"#\n#FF9400 "LV_SYMBOL_WARNING" For simplier initialization, use BSP "LV_SYMBOL_WARNING" #");
      |                                                                                                                                       ^
In file included from /Users/andremaree/DevSpace/z-projects/irmacs/managed_components/lvgl__lvgl/src/core/lv_obj.h:140,
                 from /Users/andremaree/DevSpace/z-projects/irmacs/managed_components/lvgl__lvgl/lvgl.h:35,
                 from /Users/andremaree/DevSpace/z-projects/irmacs/managed_components/espressif__esp-box/include/bsp/esp-box.h:18,
                 from /Users/andremaree/DevSpace/z-projects/irmacs/managed_components/espressif__esp-box/include/bsp/esp-bsp.h:8,
                 from /Users/andremaree/DevSpace/z-comp/gui_basic/gui_lvgl.cpp:16:
/Users/andremaree/DevSpace/z-projects/irmacs/managed_components/lvgl__lvgl/src/core/lv_obj_style.h: In function 'void lv_obj_remove_style_all(_lv_obj_t*)':
/Users/andremaree/DevSpace/z-projects/irmacs/managed_components/lvgl__lvgl/src/core/lv_obj_style.h:94:48: warning: bitwise operation between different enumeration types '<unnamed enum>' and '<unnamed enum>' is deprecated [-Wdeprecated-enum-enum-conversion]
   94 |     lv_obj_remove_style(obj, NULL, LV_PART_ANY | LV_STATE_ANY);
      |                                    ~~~~~~~~~~~~^~~~~~~~~~~~~~
`
On boot complains in log about:
`
boot i2c This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
`

ammaree, 2024-03-06:
@espressif2022 

The way I understand it the build should automatically use the new i2c_master functionality when compiled with IDF v5.3 ie Master?

espressif2022, 2024-03-11:
@ammaree , hello, sorry for my delayed reply.

- I think it'll still use the old driver.

  ```
  #define esp_lcd_new_panel_io_i2c(bus, io_config, ret_io) _Generic((bus),  \
              i2c_master_bus_handle_t : esp_lcd_new_panel_io_i2c_v2, \
              default : esp_lcd_new_panel_io_i2c_v1) (bus, io_config, ret_io) \
			  
  esp_err_t esp_lcd_new_panel_io_i2c_v2(i2c_master_bus_handle_t bus, const esp_lcd_panel_io_i2c_config_t *io_config, esp_lcd_panel_io_handle_t *ret_io)
  esp_err_t esp_lcd_new_panel_io_i2c_v1(uint32_t bus, const esp_lcd_panel_io_i2c_config_t *io_config, esp_lcd_panel_io_handle_t *ret_io)
  ```
  It can be seen that IDF internally handles compatibility between V1 and V2, mainly to determine the parameter type. 
  
  If the parameter of new_panel is `i2c_master_bus_handle_t`, V2 will be selected; 
  If it is `uint32_t` type, the driver of V1 will be selected.
  
  I saw the following definition in `esp_lcd/esp_lcd_panel_io.h`, so the V1 driver should still be used.
  `typedef uint32_t esp_lcd_i2c_bus_handle_t;                    /*!< Type of LCD I2C bus handle */`

- About warnning.

  There are also some compilation warnning mentioned above, this example code is from esp_bsp?
  And it should be caused by some differences between C++ and C compilers.

ammaree, 2024-03-14:
@espressif2022 

I just tried to rebuild the app using latest esp_box v3.0.5 with esl_lvgl_port v2.0.0 and lvgl v9 build the build fails, unable to find a working combination.

The last working combination was using box/v3.0.4 with port/v1.4.0 and lvgl 8.3.11

We are not too worried about tsp-box or lvgl or esp_lvgl_port versions, what is critical is to use the new (5.1.x. onwards) I2C master driver since this allows us to intercept the I2C API calls and transparently route them throughout our dedicated I2C task.

How can we achieve this?

As far as the "invalid suffix" compile warnings/errors are concerned, the code was lifted from a BSP example but we have fixed those.

Help on making the I2C master driver interface work will be much appreciated.

espressif2022, 2024-03-15:
`what is critical is to use the new (5.1.x. onwards) I2C master driver since this allows us to intercept the I2C API calls and transparently route them throughout our dedicated I2C task.`
I'm sorry. This description above is not very clear to me. 
According to the existing driver, if we use our esp_lcd/esp_lcd_panel_io interface, the old IIC driver will be called, which requires modification by IDF.
Does the current driver not meet the requirements?

ESP-Mars, 2024-07-02:
Temporarily closed due to the extended period of time with no response. Should you have additional questions or concerns, don't hesitate to reopen the issue.

closed 2024-07-02 by ESP-Mars

ammaree, 2024-09-19:
Sorry about the long delay but this problem still persists.
In order to simplify the situation we have:
- changed to the ESP32-S3-LCD-EV-BOARD-2
- building a GUI application using ESP-IDF Master (v5.4)
- using the latest BSP for the devboard
- also using latest LVGL component

The I2C controller is not used by anything in our application but the BSP includes I2C support for the touch controller.

At startup the following message still occurs:
`boot i2c This driver is an old driver, please migrate your application code to adapt driver/i2c_master.h`

What do we have to change in the configuration, BSP, LVGL or IDF to force the use of the new V2 I2C driver?

The comments on Mar 14 were just for background, if we can force the use of the V2 driver the problem will go away...

ammaree, 2024-09-20:
@ESP-Mars 
I have also check with the just released lv_port_espressif_esp32-s3-lcd-ev-board demo/example from the LVGL Project Creator extension for VS-Code and the exact same situation there.

So it seems all these projects are missing something to force the use of the I2C V2 drivers

espressif2022, 2024-09-20:
> @ESP-Mars I have also check with the just released lv_port_espressif_esp32-s3-lcd-ev-board demo/example from the LVGL Project Creator extension for VS-Code and the exact same situation there.
> 
> So it seems all these projects are missing something to force the use of the I2C V2 drivers

Yes, all our current BSPs are not compatible with the new I2C driver. When you mix the old and new drivers in your code, the program will automatically detect crashes.

Why do you need the new driver，and how can we help you?

ammaree, 2024-09-24:
We have an I2C task that allows us to combine multiple read and write operations with timers, delays and events to streamline I2C operation.

In order to support libraries that use I2C APIs directly we wrap the IDF I2C APIs and redirect them to our I2C task.

All we need is that the BSP's support the new I2C master interface.

espressif2022, 2024-09-25:
> We have an I2C task that allows us to combine multiple read and write operations with timers, delays and events to streamline I2C operation.
>
> In order to support libraries that use I2C APIs directly [...]

Hello, you can refer to this https://github.com/espressif/esp-bsp/issues/391, we'll support bsp in this  respository.