espressif/esp-box #134: lvgl_v8.3播放spiffs 中的gif动图，很卡很迟钝

huangdiajd, 2024-03-02:
lvgl_v8.3播放spiffs 中的gif动图，很卡很迟钝。请问我应该怎么优化迟钝现象？或者有没有使用过box主机播放gif动图的事例？

LHYhorion, 2024-03-04:
您可以尝试将 gif 转换成 C 数组形式，LVGL 8.3 支持使用 gif。
转换网址：https://lvgl.io/tools/imageconverter
使用方法：https://docs.lvgl.io/8.3/libs/gif.html

huangdiajd, 2024-03-04:
> 您可以尝试将 gif 转换成 C 数组形式，LVGL 8.3 支持使用 gif。 转换网址：https://lvgl.io/tools/imageconverter （注意需要转换成 RAW 格式） 使用方法：https://docs.lvgl.io/8.3/libs/gif.html

我刚尝试您说的这种方法去实现，对于320x240分辨率的gif 效果也是有卡顿。我想问一下，除了gif可以实现动图，还有什么方式方法可以做一些流畅的动图展示在我的esp32 s3 box上运行。

LHYhorion, 2024-03-05:
您可以尝试开启以下配置项，观察效果是否有所改善

SPI RAM config
- Move Instructions in Flash to PSRAM
- Move Read-Only Data in Flash to PSRAM

Cache config -- Data cache line size
- 64 Bytes

LVGL configuration -- Memory settingg
- If true use custom malloc/free, otherwise use the built-in `lv_mem_alloc()` and `lv_mem_free()`
- Use the standard memcpy and memset instead of LVGL's own functions

LVGL configuration -- Feature configuration -- Compiler settings
- Set IRAM as LV_ATTRIBUTE_FAST_MEM

stelldogone, 2024-08-27:
我也是box播放320*240的gif很卡顿，请问大哥是否解决了这个问题，如果是的话能联系帮助我吗，可以加我的qq：2272271或者微信：13695877373