espressif/esp-box #135: ESP32-S3-BOX-Lite 运行chatgpt demo 时偶尔出现无法唤醒的问题

Wvirgil123, 2024-03-02:
环境：
1、IDF: ESP-IDF v5.1
2、[esp-box](https://github.com/espressif/esp-box) commit: ee2160d233a5fe0dcc8e7fe728360a91b7f424e1
3、 设备 ESP32-S3-BOX-Lite 
4、WIFI 连接正常

问题:
每次上电有一定概率出现无法唤醒的情况，按键唤醒之后，也会一直卡在listening的状态； 设备重新上电之后， 有概率可以正常唤醒， 只要能唤醒之后， 后面都是正常的。

(1) 测试
当出现异常时， 获取wakeup_state和vad_state状态。当说唤醒词时， wakeup_state 也一直也为0； 当通过按键唤醒时， 周围没有说话时，vad_state一直为1，导致无法发出超时事件.

[image]

(2) 异常时启动log（ 正常和异常唤醒时,日志无差别，启动阶段都会报出“AFE Fetch Fail”）
```
I (27) boot: ESP-IDF v5.1 2nd stage bootloader
I (27) boot: compile time Mar  2 2024 09:40:50
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.1
I (34) qio_mode: Enabling default flash chip QIO
I (39) boot.esp32s3: Boot SPI Speed : 80MHz
I (44) boot.esp32s3: SPI Mode       : QIO
I (48) boot.esp32s3: SPI Flash Size : 16MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (77) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (84) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (92) boot:  3 factory          factory app      00 00 00010000 00600000
I (99) boot:  4 ota_0            OTA app          00 10 00700000 00200000
I (107) boot:  5 storage          Unknown data     01 82 00900000 00200000
I (114) boot:  6 model            Unknown data     01 82 00b00000 003e8000
I (122) boot: End of partition table
I (126) boot: Defaulting to factory image
I (131) esp_image: segment 0: paddr=00010020 vaddr=3c100020 size=21a998h (2206104) map
I (474) esp_image: segment 1: paddr=0022a9c0 vaddr=3fc9b800 size=05658h ( 22104) load
I (478) esp_image: segment 2: paddr=00230020 vaddr=42000020 size=f5984h (1005956) map
I (632) esp_image: segment 3: paddr=003259ac vaddr=3fca0e58 size=01998h (  6552) load
I (634) esp_image: segment 4: paddr=0032734c vaddr=40374000 size=17750h ( 96080) load
I (667) boot: Loaded app from partition at offset 0x10000
I (667) boot: Disabling RNG early entropy source...
I (678) cpu_start: Multicore app
I (678) octal_psram: vendor id    : 0x0d (AP)
I (679) octal_psram: dev id       : 0x02 (generation 3)
I (682) octal_psram: density      : 0x03 (64 Mbit)
I (687) octal_psram: good-die     : 0x01 (Pass)
I (692) octal_psram: Latency      : 0x01 (Fixed)
I (698) octal_psram: VCC          : 0x01 (3V)
I (703) octal_psram: SRF          : 0x01 (Fast Refresh)
I (709) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (714) octal_psram: BurstLen     : 0x01 (32 Byte)
I (720) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (726) octal_psram: DriveStrength: 0x00 (1/1)
I (732) esp_psram: Found 8MB PSRAM device
I (736) esp_psram: Speed: 80MHz
I (809) mmu_psram: Instructions copied and mapped to SPIRAM
I (810) cpu_start: Pro cpu up.
I (810) cpu_start: Starting app cpu, entry point is 0x40375870
0x40375870: call_start_cpu1 at /esp-idf/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.
I (1061) esp_psram: SPI SRAM memory test OK
I (1071) cpu_start: Pro cpu start user code
I (1071) cpu_start: cpu freq: 240000000 Hz
I (1071) cpu_start: Application information:
I (1074) cpu_start: Project name:     chatgpt_demo
I (1079) cpu_start: App version:      v0.5.0-55-gee2160d-dirty
I (1086) cpu_start: Compile time:     Mar  2 2024 09:40:39
[... 120 lines cut ...]
W (3925) app_sr: AFE Fetch Fail
W (4225) app_sr: AFE Fetch Fail
I (4497) wifi station: Total APs scanned = 31, ret:0
I (4497) wifi station: xxxx
I (4596) wifi:new:<6,2>, old:<1,0>, ap:<255,255>, sta:<6,2>, prof:1
I (4597) wifi:state: init -> auth (b0)
I (4603) wifi:state: auth -> assoc (0)
I (4645) wifi:state: assoc -> run (10)
W (4833) app_sr: AFE Fetch Fail
I (4851) wifi:<ba-add>idx:0 (ifx:0, xxx), tid:0, ssn:134, winSize:64
I (4940) wifi:connected with M2-TEST, aid = 2, channel 6, 40D, bssid = xxx
I (4940) wifi:security: WPA2-PSK, phy: bgn, rssi: -29
I (4944) wifi:pm start, type: 1

I (4945) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 25000, mt_pti: 0, mt_time: 10000
I (4988) wifi:AP's beacon interval = 102400 us, DTIM period = 1
W (5120) wifi:[ADDBA]rx delba, code:39, delete tid:0
I (5120) wifi:<ba-del>idx:0, tid:0
W (5121) wifi:[ADDBA]rx delba, code:39, delete tid:0
W (5133) app_sr: AFE Fetch Fail
W (5433) app_sr: AFE Fetch Fail
W (5733) app_sr: AFE Fetch Fail
I (5954) esp_netif_handlers: sta ip: 192.168.8.101, mask: 255.255.255.0, gw: 192.168.8.1
I (5954) wifi station: got ip:192.168.8.101
```

LHYhorion, 2024-03-04:
已经根据您的环境进行了测试并反复上电，没有复现问题。建议您可以尝试更新一下最新版本，交叉比对，观察是否还会有相同问题出现。

Wvirgil123, 2024-03-05:
> 已经根据您的环境进行了测试并反复上电，没有复现问题。建议您可以尝试更新一下最新版本，交叉比对，观察是否还会有相同问题出现。

@LHYhorion  不好意思，我测试的时候修改了 esp-sr 的版本， 改成了1.6.1 (当前最近版本， 因为支持更多的唤醒词)
https://components.espressif.com/components/espressif/esp-sr

将esp-sr改回1.3.3， 测试多次都可以正常唤醒。

LHYhorion, 2024-03-05:
好的，不用客气，有问题可以随时提出。

closed 2024-03-05 by ESP-Mars

Wvirgil123, 2024-03-05:
> 好的，不用客气，有问题可以随时提出。

能否支持支持最新版本的esp-sr 呢

PengyuYan703, 2024-03-06:
我现在正在用 esp-sr 的1.6.1，能支持的