espressif/esp-box #137: ESP-IDF 5.2.1 and sensor sensor bottom `intr_alloc` factory_demo problem

MacWyznawca, 2024-03-21:
The latest version of ESP-BOX factory_demo on the stable wewrsion ESP-IDF 5.2.1 does not have enough interrupts. 
The error occurs only when using the sensor cradle. 

```
I (19) boot: compile time Mar 20 2024 21:28:14
I (19) boot: Multicore bootloader
I (19) boot: chip revision: v0.2
I (20) qio_mode: Enabling QIO for flash chip GD
I (20) boot.esp32s3: Boot SPI Speed : 80MHz
I (20) boot.esp32s3: SPI Mode       : QIO
I (20) boot.esp32s3: SPI Flash Size : 16MB
I (21) boot: Enabling RNG early entropy source...
I (21) boot: Partition Table:
I (21) boot: ## Label            Usage          Type ST Offset   Length
I (21) boot:  0 sec_cert         Unknown data     01 06 0000d000 00003000
I (22) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (22) boot:  2 otadata          OTA data         01 00 00016000 00002000
I (23) boot:  3 phy_init         RF data          01 01 00018000 00001000
I (23) boot:  4 fctry            WiFi data        01 02 00019000 00006000
I (23) boot:  5 ota_0            OTA app          00 10 00020000 0041a000
I (24) boot:  6 storage          Unknown data     01 82 0043a000 0028a000
I (24) boot:  7 model            Unknown data     01 82 006c4000 00866000
I (25) boot: End of partition table
I (25) boot: No factory image, trying OTA 0
I (25) esp_image: segment 0: paddr=00020020 vaddr=3c180020 size=2314b0h (2299056) map
I (374) esp_image: segment 1: paddr=002514d8 vaddr=3fca4400 size=08354h ( 33620) load
I (380) esp_image: segment 2: paddr=00259834 vaddr=40378000 size=067e4h ( 26596) load
I (386) esp_image: segment 3: paddr=00260020 vaddr=42000020 size=172ae4h (1518308) map
I (616) esp_image: segment 4: paddr=003d2b0c vaddr=4037e7e4 size=15b78h ( 88952) load
I (633) esp_image: segment 5: paddr=003e868c vaddr=50000000 size=00004h (     4) load
I (634) esp_image: segment 6: paddr=003e8698 vaddr=600fe000 size=0005ch (    92) load
I (646) boot: Loaded app from partition at offset 0x20000
I (670) boot: Set actual ota_seq=1 in otadata[0]
I (670) boot: Disabling RNG early entropy source...
I (670) cpu_start: Multicore app
I (671) octal_psram: vendor id    : 0x0d (AP)
I (671) octal_psram: dev id       : 0x03 (generation 4)
I (671) octal_psram: density      : 0x05 (128 Mbit)
I (672) octal_psram: good-die     : 0x01 (Pass)
I (672) octal_psram: Latency      : 0x01 (Fixed)
I (672) octal_psram: VCC          : 0x00 (1.8V)
I (672) octal_psram: SRF          : 0x01 (Fast Refresh)
I (673) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (673) octal_psram: BurstLen     : 0x01 (32 Byte)
I (673) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (674) octal_psram: DriveStrength: 0x00 (1/1)
I (675) MSPI Timing: PSRAM timing tuning index: 5
I (675) esp_psram: Found 16MB PSRAM device
I (675) esp_psram: Speed: 80MHz
I (1317) esp_psram: SPI SRAM memory test OK
I (1326) cpu_start: Pro cpu start user code
I (1326) cpu_start: cpu freq: 240000000 Hz
I (1326) cpu_start: Application information:
I (1327) cpu_start: Project name:     factory_demo
I (1327) cpu_start: App version:      v0.5.0-57-g9b28a35
I (1327) cpu_start: Compile time:     Mar 20 2024 21:51:21
I (1327) cpu_start: ELF file SHA256:  5adf1f32a...
I (1327) cpu_start: ESP-IDF:          v5.2.1-dirty
I (1328) cpu_start: Min chip rev:     v0.0
I (1328) cpu_start: Max chip rev:     v0.99 
I (1328) cpu_start: Chip rev:         v0.2
I (1328) heap_init: Initializing. RAM available for dynamic allocation:
I (1328) heap_init: At 3FCB3308 len 00036408 (217 KiB): RAM
I (1328) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (1329) heap_init: At 600FE05C len 00001F8C (7 KiB): RTCRAM
I (1329) esp_psram: Adding pool of 16384K of PSRAM memory to heap allocator
I (1330) spi_flash: detected chip: gd
I (1330) spi_flash: flash io: qio
W (1330) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
I (1331) sleep: Configure to isolate all GPIO pins in sleep state
[... 142 lines cut ...]
I (7530) esp_claim: Initialising Self Claiming. This may take time.
W (7540) esp_claim: Generating the private key. This may take time.
E (7559) intr_alloc: No free interrupt inputs for RSA interrupt (flags 0xE)
HINT: For troubleshooting instructions related to interrupt allocation, run 'idf.py docs -sp api-reference/system/intr_alloc.html'
E (7561) bignum: Failed to allocate RSA interrupt 261

abort() was called at PC 0x42105452 on core 0
0x42105452: esp_mpi_isr_initialise at /Users/jkk/esp/esp-idf/components/mbedtls/port/bignum/esp_bignum.c:98

Backtrace: 0x40379f1a:0x3fce3080 0x4038aa01:0x3fce30a0 0x4039091d:0x3fce30c0 0x42105452:0x3fce3130 0x421055bb:0x3fce3150 0x421056cd:0x3fce3190 0x420f80eb:0x3fce31b0 0x420f84d9:0x3fce3210 0x420f8635:0x3fce3240 0x42102f0d:0x3fce3280 0x42085a65:0x3fce32d0 0x42085e51:0x3fce34f0 0x42085f78:0x3fce3560 0x4038af7e:0x3fce3580
0x40379f1a: panic_abort at /Users/jkk/esp/esp-idf/components/esp_system/panic.c:472
0x4038aa01: esp_system_abort at /Users/jkk/esp/esp-idf/components/esp_system/port/esp_system_chip.c:93
0x4039091d: abort at /Users/jkk/esp/esp-idf/components/newlib/abort.c:38
0x42105452: esp_mpi_isr_initialise at /Users/jkk/esp/esp-idf/components/mbedtls/port/bignum/esp_bignum.c:98
0x421055bb: esp_mpi_exp_mod at /Users/jkk/esp/esp-idf/components/mbedtls/port/bignum/esp_bignum.c:412
0x421056cd: mbedtls_mpi_exp_mod at /Users/jkk/esp/esp-idf/components/mbedtls/port/bignum/esp_bignum.c:473
0x420f80eb: mpi_miller_rabin at /Users/jkk/esp/esp-idf/components/mbedtls/mbedtls/library/bignum.c:2428
0x420f84d9: mbedtls_mpi_is_prime_ext at /Users/jkk/esp/esp-idf/components/mbedtls/mbedtls/library/bignum.c:2501
0x420f8635: mbedtls_mpi_gen_prime at /Users/jkk/esp/esp-idf/components/mbedtls/mbedtls/library/bignum.c:2571
0x42102f0d: mbedtls_rsa_gen_key at /Users/jkk/esp/esp-idf/components/mbedtls/mbedtls/library/rsa.c:712
0x42085a65: esp_rmaker_claim_generate_key at /Users/jkk/esp/esp-box/examples/factory_demo/managed_components/espressif__esp_rainmaker/src/core/esp_rmaker_claim.c:214 (discriminator 1)
0x42085e51: __esp_rmaker_claim_init at /Users/jkk/esp/esp-box/examples/factory_demo/managed_components/espressif__esp_rainmaker/src/core/esp_rmaker_claim.c:843
0x42085f78: esp_rmaker_claim_task at /Users/jkk/esp/esp-box/examples/factory_demo/managed_components/espressif__esp_rainmaker/src/core/esp_rmaker_claim.c:896
0x4038af7e: vPortTaskWrapper at /Users/jkk/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134
```

LHYhorion, 2024-03-21:
You can also manually disable the 'Enable hardware MPI acceleration' configuration option to mitigate this issue.

MacWyznawca, 2024-03-21:
Thanks @LHYhorion! Workaround works!

closed 2024-05-08 by ESP-Mars