espressif/esp-box #138: atgpt demo获取chatgpt结果后，一直没有tts语音合成播放，一直返回：E (580156) audio: unknown file type, cleaning up

shantai57, 2024-03-26:
查看openai官方文档默认返回**MP3**格式音频，但是接收回来的数据一直解析不出来是什么格式。
[image]
将音频数据保存到文件，然后打开文件进行播放也是报一样的错误：**E (580156) audio: unknown file type, cleaning up**
[image]
请问上述问题怎么解决，谢谢！

espressif2022, 2024-03-26:
可以使用在线命令，先获取下音频文件，导到板子和本地分别看下？
```
curl https://api.openai.com/v1/audio/speech \
  -H "Authorization: Bearer $OPENAI_API_KEY" \
  -H "Content-Type: application/json" \
  -d '{
    "model": "tts-1",
    "input": "Today is a wonderful day to build something people love!",
    "voice": "alloy"
  }' \
  --output speech.mp3
```

shantai57, 2024-03-26:
可以的，都能播放出来。

[image]

shantai57, 2024-03-26:
可以的，PC和板子都能播放出来。

[image]

shantai57, 2024-03-26:
正常导入的音频和预置的音频播放都没问题，就是通过源码从openai获取的音频解析不成功。

espressif2022, 2024-03-26:
> 正常导入的音频和预置的音频播放都没问题，就是通过源码从openai获取的音频解析不成功。

之前在 audio 结尾会出问题，curl 获取的，和本地获取的现象一致，现已修复；
头上直接出错的，且只有这边 TTS 拿到的出问题，没遇到，晚点测试下。

shantai57, 2024-03-26:
> > 正常导入的音频和预置的音频播放都没问题，就是通过源码从openai获取的音频解析不成功。
> 
> 之前在 audio 结尾会出问题，curl 获取的，和本地获取的现象一致，现已修复； 头上直接出错的，且只有这边 TTS 拿到的出问题，没遇到，晚点测试下。

好的，谢谢

LHYhorion, 2024-04-03:
请问您的网络环境是怎样的？如果是在国内的话，可能直接访问 OpenAI 会有问题，您可以尝试更换 URL 为：https://api.xiaoyukefu.com/v1/
并再次测试是否有同样问题出现？我刚刚进行了测试没有出现与您相似的情况。

closed 2024-04-11 by shantai57