espressif/esp-box #139: gpt demo运行时，经常在播放tts的时候自动重启。

welkinchan, 2024-03-28:
将gpt demo烧录到espbox3后，使用xiaoyukefu的api进行对话，
当屏幕显示了回复文本，即将开始播放回复语音时，设备就自动重启。
试了几次，感觉是回复文本比较长的时候就会自动重启，如果比较短的时候多数都可以播放回复语音。
FYI: 我用的是idf 5.0。menuconfig里边打开了中文命令词识别。

LHYhorion, 2024-04-02:
可以提供一下自动重启的 log 吗？以方便我们定位问题。

LHYhorion, 2024-04-03:
> 将gpt demo烧录到espbox3后，使用xiaoyukefu的api进行对话， 当屏幕显示了回复文本，即将开始播放回复语音时，设备就自动重启。 试了几次，感觉是回复文本比较长的时候就会自动重启，如果比较短的时候多数都可以播放回复语音。 FYI: 我用的是idf 5.0。menuconfig里边打开了中文命令词识别。

开中文命令识别是没有作用的。如果有中文识别的需要，您可以在代码中进行修改，具体参考：https://github.com/espressif/esp-box/tree/master/examples/chatgpt_demo#known-issues。所以建议您使用原有配置，在代码中进行修改，并检查是否还会有重启问题出现。

welkinchan, 2024-04-12:
> > 将gpt demo烧录到espbox3后，使用xiaoyukefu的api进行对话， 当屏幕显示了回复文本，即将开始播放回复语音时，设备就自动重启。 试了几次，感觉是回复文本比较长的时候就会自动重启，如果比较短的时候多数都可以播放回复语音。 FYI: 我用的是idf 5.0。menuconfig里边打开了中文命令词识别。
> 
> 开中文命令识别是没有作用的。如果有中文识别的需要，您可以在代码中进行修改，具体参考：https://github.com/espressif/esp-box/tree/master/examples/chatgpt_demo#known-issues 所以建议您使用原有配置，在代码中进行修改，并检查是否还会有重启问题出现。

1. 我近期使用的时候，发现了另外一个问题，就是使用新版的esp-sr组件后运行，
在menuconfig里边修改唤醒词，有很多唤醒词无法唤醒设备，包括sophia，甚至hi esp。
我印象里边，只有使用hi,lexin都可以正常唤醒。

2. 请问在运行GPT_DEMO例程的时候，我可以打开multinet里的命令词识别吗？
在正常使用gpt语音对话的功能同时，打开multinet的话也能够识别到命令词列表中的命令词吗？
我早先试了下，好像也没成功。

LHYhorion, 2024-04-12:
您好，目前没有针对 chatgpt_demo 进行最新 esp-sr 组件的测试，因此可能存在一些问题（我在刚才测试了 1.7.0 版本的 esp-sr， Hi, lexin 和 Hi, ESP 是可以识别的），后续可能会测试并更新。同时，在目前 chatgpt_demo 中，暂时是无法使用 multinet 里的命令词识别的。
       针对长文本重启问题，后续会对该 Demo 进行优化。但是我测试并没有出现重启现象，目前已知的问题是，长文本的等待时间过长，具体原因在于目前的交互流程（语音转文本，文本回答转 TTS）在长文本的情况下等待时间会更长。后续会优化这个流程。

welkinchan, 2024-04-18:
> 您好，目前没有针对 chatgpt_demo 进行最新 esp-sr 组件的测试，因此可能存在一些问题（我在刚才测试了 1.7.0 版本的 esp-sr， Hi, lexin 和 Hi, ESP 是可以识别的），后续可能会测试并更新。同时，在目前 chatgpt_demo 中，暂时是无法使用 multinet 里的命令词识别的。 针对长文本重启问题，后续会对该 Demo 进行优化。但是我测试并没有出现重启现象，目前已知的问题是，长文本的等待时间过长，具体原因在于目前的交互流程（语音转文本，文本回答转 TTS）在长文本的情况下等待时间会更长。后续会优化这个流程。

1. 感谢，我确认了下，是sophia这种词的识别率很低导致的，并不是模型没能成功加载导致的。
2. 针对新出的唤醒词模型，我通过直接覆盖其他模型的方法也成功实现了。
3. multinet与在线对话的功能我也成功实现了。
非常感谢您的答疑解惑，帮我解决了不少问题。@LHYhorion

closed 2024-04-18 by welkinchan