espressif/esp-box #141: gpt demo运行ASR时，语音丢失问题。

welkinchan, 2024-04-13:
在运行gpt demo的时候，如果录入的语音较短，大约10s以内，那么这10s的语音的内容都会被ASR识别到。
但是，当录入的语音较长时，比如从1数数到30，或者到20的时候，语音里边大约末尾5-7s的语音就会丢失，丢失的语音部分无法被ASR识别到，导致语音识别内容不完整。
测试了几次，感觉问题应该是出在录音文件的保存环节，而不是ASR环节。
能否帮忙分析看下是什么原因导致的？
我的IDF版本是5.1.0，esp-box版本是master版本。

附件是完整的日志：

[gpt log_20240413_2300.txt](https://github.com/espressif/esp-box/files/14967976/gpt.log_20240413_2300.txt)

数数从1到20时，录音停止时间正常，但是ASR识别出来的内容丢失了一部分，以下是部分日志节选：

W (44064) app_audio: CWG================================sizeof wav_header: 44
W (44064) app_audio: CWG================================MAX_FILE_SIZE - sizeof(wav_header_t): 1048532
W (44071) app_audio: CWG================================record_save===record_total_len: 218922
W (44138) app_audio: CWG================================sizeof wav_header: 44
W (44138) app_audio: CWG================================MAX_FILE_SIZE - sizeof(wav_header_t): 1048532
W (44145) app_audio: CWG================================record_save===record_total_len: 219945
W (44198) app_audio: CWG================================sizeof wav_header: 44
W (44198) app_audio: CWG================================MAX_FILE_SIZE - sizeof(wav_header_t): 1048532
W (44205) app_audio: CWG================================record_save===record_total_len: 220968
I (44200) app_audio: ESP_MN_STATE_TIMEOUT
W (44219) app_audio: CWG================================record_total_len
I (44226) app_audio: ### record Stop,record_total_len: 221991 216K
W (44233) app_audio: CWG================================file_total_len: 222035
W (44244) app_audio: CWG================================record_total_len: 221991
I (44244) app_audio: Player PLAYING
I (44271) stt_api: connected to web-server
I (44281) I2S_IF: Pending out channel for in channel running
E (44303) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (44303) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (44308) I2S_IF: STD Mode 1 bits:16/16 channel:2 sample_rate:16000 mask:3
I (44386) Adev_Codec: Open codec device OK
E (44386) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (44389) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (44395) I2S_IF: STD Mode 0 bits:16/16 channel:2 sample_rate:16000 mask:3
I (44403) ES7210: Bits 16
I (44409) ES7210: Enable ES7210_INPUT_MIC1
I (44411) ES7210: Enable ES7210_INPUT_MIC2
I (44418) ES7210: Unmuted
I (44419) Adev_Codec: Open codec device OK
I (45453) stt_api: finished a request and response!
E (45454) stt_api: ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~result_data: ���一���二���三���四���五���六���七���八���九���十���，���十���十���，23���。

I (45458) stt_api: disconnected to web-server
E (45464) app_main: cwg===================text: ���一���二���三���四���五���六���七���八���九���十���，���十���十���，23���。

I (45610) esp-x509-crt-bundle: Certificate validated
E (47197) app_main: cwg==========================GPT resonse content:
E (47197) app_main: cwg==========================GPT resonse content: ���您���好���！���看���起���来���您���在���数���数���。���需���要���我���帮���您���做���些���什���么���吗���？

E (47206) app_main: cwg=========================start tts

E (47212) TTS-Api: [Start] create_TTS_request, timestamp:47212
I (47244) TTS-Api: HTTP_EVENT_ON_CONNECTED
I (47245) TTS-Api: HTTP_EVENT_HEADER_SENT
I (47498) TTS-Api: HTTP_EVENT_ON_HEADER
I (47498) TTS-Api: HTTP_EVENT_ON_HEADER
I (47499) TTS-Api: HTTP_EVENT_ON_HEADER
I (47501) TTS-Api: HTTP_EVENT_ON_HEADER
I (47505) TTS-Api: HTTP_EVENT_ON_HEADER
I (47510) TTS-Api: HTTP_EVENT_ON_HEADER
I (47514) TTS-Api: HTTP_EVENT_ON_HEADER
I (47519) TTS-Api: HTTP_EVENT_ON_HEADER
I (47523) TTS-Api: HTTP_EVENT_ON_HEADER
I (47528) TTS-Api: HTTP_EVENT_ON_HEADER
I (47532) TTS-Api: HTTP_EVENT_ON_DATA, len=(0 + 958) [1048576]
I (47540) TTS-Api: HTTP_EVENT_ON_DATA, len=(958 + 5744) [1048576]
I (47556) TTS-Api: HTTP_EVENT_ON_DATA, len=(6702 + 2824) [1048576]
I (47560) TTS-Api: HTTP_EVENT_ON_DATA, len=(9526 + 2920) [1048576]
I (47571) TTS-Api: HTTP_EVENT_ON_DATA, len=(12446 + 2824) [1048576]
I (47577) TTS-Api: HTTP_EVENT_ON_DATA, len=(15270 + 2920) [1048576]
I (47583) TTS-Api: HTTP_EVENT_ON_DATA, len=(18190 + 1220) [1048576]
I (47585) TTS-Api: HTTP_EVENT_ON_DATA, len=(19410 + 1604) [1048576]
I (47595) TTS-Api: HTTP_EVENT_ON_DATA, len=(21014 + 1220) [1048576]
I (47604) TTS-Api: HTTP_EVENT_ON_DATA, len=(22234 + 1700) [1048576]
I (47618) TTS-Api: HTTP_EVENT_ON_DATA, len=(23934 + 1124) [1048576]
I (47626) TTS-Api: HTTP_EVENT_ON_DATA, len=(25058 + 1412) [1048576]
I (47629) TTS-Api: HTTP_EVENT_ON_DATA, len=(26470 + 3208) [1048576]
I (47646) TTS-Api: HTTP_EVENT_ON_DATA, len=(29678 + 1028) [1048576]
I (47649) TTS-Api: HTTP_EVENT_ON_DATA, len=(30706 + 4716) [1048576]
I (47662) TTS-Api: HTTP_EVENT_ON_DATA, len=(35422 + 932) [1048576]
I (47663) TTS-Api: HTTP_EVENT_ON_DATA, len=(36354 + 1412) [1048576]
I (47666) TTS-Api: HTTP_EVENT_ON_DATA, len=(37766 + 3400) [1048576]
I (47682) TTS-Api: HTTP_EVENT_ON_DATA, len=(41166 + 5744) [1048576]
I (47711) TTS-Api: HTTP_EVENT_ON_DATA, len=(46910 + 34) [1048576]
I (47711) TTS-Api: HTTP_EVENT_ON_FINISH:46944, 45 K
E (47712) TTS-Api: [End] create_TTS_request, + offset:500
I (47718) TTS-Api: HTTP_EVENT_DISCONNECTED
I (47750) app_audio: Player NEXT
I (47755) I2S_IF: Pending out channel for in channel running
E (47772) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (47772) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (47776) I2S_IF: STD Mode 1 bits:16/16 channel:2 sample_rate:16000 mask:3
I (47863) Adev_Codec: Open codec device OK
E (47863) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (47866) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (47872) I2S_IF: STD Mode 0 bits:16/16 channel:2 sample_rate:16000 mask:3
I (47884) ES7210: Bits 16
I (47890) ES7210: Enable ES7210_INPUT_MIC1
I (47892) ES7210: Enable ES7210_INPUT_MIC2
I (47895) ES7210: Unmuted
I (47896) Adev_Codec: Open codec device OK
W (47900) app_main:     Description     Internal        SPIRAM
W (47905) app_main: Current Free Memory 79519           12969100
W (47927) app_main: Min. Ever Free Size 60335           12825884
W (52927) app_main:     Description     Internal        SPIRAM
W (52927) app_main: Current Free Memory 80623           12969100
W (52927) app_main: Min. Ever Free Size 60335           12825884
I (53754) app_audio: Player IDLE
I (53756) I2S_IF: Pending out channel for in channel running
E (53771) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (53772) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (53775) I2S_IF: STD Mode 1 bits:16/16 channel:2 sample_rate:16000 mask:3
I (53859) Adev_Codec: Open codec device OK
E (53859) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (53861) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (53867) I2S_IF: STD Mode 0 bits:16/16 channel:2 sample_rate:16000 mask:3
I (53875) ES7210: Bits 16
I (53881) ES7210: Enable ES7210_INPUT_MIC1
I (53884) ES7210: Enable ES7210_INPUT_MIC2
I (53891) ES7210: Unmuted
I (53891) Adev_Codec: Open codec device OK
I (53896) app_main: replay audio end
W (57940) app_main:     Description     Internal        SPIRAM
W (57940) app_main: Current Free Memory 80883           12969100
W (57940) app_main: Min. Ever Free Size 60335           12825884

LHYhorion, 2024-04-18:
好的，稍后我会进行一个测试，该 demo 后续也会进行优化。您近距离也会发生这个情况吗？或者这个情况是必现吗？因为录音这也就是存在一个 Buffer 中，您测试的时候是把这个 buffer 中的数据播放出来然后发现会末尾丢失吗？

welkinchan, 2024-04-18:
> 好的，稍后我会进行一个测试，该 demo 后续也会进行优化。您近距离也会发生这个情况吗？或者这个情况是必现吗？因为录音这也就是存在一个 Buffer 中，您测试的时候是把这个 buffer 中的数据播放出来然后发现会末尾丢失吗？

这个问题的复现率很高，几乎都是100%的。从一数数到三十会非常明显。近距离的时候也是会丢失语音。
1. 我用源代码，使用直连openai的方式进行测试，ASR出来的文字会有末尾语音丢失的问题。
2. 我将源代码改成使用国内的ASR的API，像是百度的语音识别，末尾语音也是丢的。
3. 我将录音缓存写入到spiff中的rec.wav，放出来的声音会加速（我不理解为什么会加速），但是加速放出来的声音也是末尾丢失的。

我问了下其他群里的人，他们说没注意到。
搞得我有点怀疑是不是我的编译环境不一样导致的。
我用的是offline的5.1.0的IDF，master版本的esp-box项目，还装了2.6的ADF并打了补丁（感觉好像没用到ADF）。

LHYhorion, 2024-04-18:
您可以尝试更新一下 IDF，我没有理解是哪个阶段的语音会丢失，是您录入的语音会丢失还是回答的语音会丢失。具体声音会加速，可能原因就在于音频文件的位宽或者采样率与您设置的不同。

welkinchan, 2024-04-18:
> 您可以尝试更新一下 IDF，我没有理解是哪个阶段的语音会丢失，是您录入的语音会丢失还是回答的语音会丢失。具体声音会加速，可能原因就在于音频文件的位宽或者采样率与您设置的不同。

1. 其他群里边玩gpt_demo的人比较少，他们也没有仔细测过这个问题，但我是测过很多遍。包括用box3，boxlite，新的idf版本都测过，都有这个问题。
2. 我所说的语音丢失是指，在gpt_demo的代码逻辑里边，当唤醒词唤醒设备之后，record_start就会运行，开始保存录音，直到mn_detect_timeout之后，record_stop函数停止，这个时候缓存里边会有录音的，加了wav头之后的audio_buffer。就是这个audio_buffer的内容会比实际的说话的内容少了末尾的7s左右的内容。当用户说话少于10s的时候没有这个问题，但是说话时间较长就会有语音丢失。
3. 我还尝试过把timeout逻辑里边，flag的延时加长，但还是会语音丢失。

LHYhorion, 2024-04-19:
好的，我明白您的意思了。您一般说话时长有多久？我会根据您的情况进行一个同步测试。但是我猜测可能是 buffer 不够大了。

LHYhorion, 2024-04-19:
经过检查发现确实是由于 Buffer 不足导致的语音丢失，您可以将 https://github.com/espressif/esp-box/blob/78531c7a26a68eab6c2e6788b10d9eeaaf97e23d/examples/chatgpt_demo/main/app/app_audio.h#L11 调大并再次进行测试，因为之前没有考虑到输入语音会很久。后续优化会补充上一个长度警告。感谢反馈。

welkinchan, 2024-04-19:
> 经过检查发现确实是由于 Buffer 不足导致的语音丢失，您可以将
> 
> https://github.com/espressif/esp-box/blob/78531c7a26a68eab6c2e6788b10d9eeaaf97e23d/examples/chatgpt_demo/main/app/app_audio.h#L11
> 
> 调大并再次进行测试，因为之前没有考虑到输入语音会很久。后续优化会补充上一个长度警告。感谢反馈。

非常感谢。
能否帮忙测试一下从0数数到30，看一下录音是否确实能将30s的录音完整缓存下来？
我印象里边这个地方我是调试过的，效果并不好，还是会丢音。
（不过我后来在调试别的问题的时候，发现我在别的地方又重复定义了这个变量，所以我不能完整确认这个地方的缓存空间加大后是否真的有效。）

另外，还有一个原因让我并不认可这个结论，
就是我昨晚在调试timeout时间的时候，我将flag的检测时间从100调整到60，然后发现一个问题。
就是平时的时候，录音10s都可以正常缓存录音，但是flag时间从100调整到60后，也发生了末尾缓存录音丢失的问题，
所以我认为肯定不是file_size不够大导致的。

另外，在30s的录音时间的情况下，我也试过将timeout的flag从100调整到150，录音丢失的问题也是一样存在的。

LHYhorion, 2024-04-20:
您好，参考您的建议，我进行了读数测试。并在 https://github.com/espressif/esp-box/blob/78531c7a26a68eab6c2e6788b10d9eeaaf97e23d/examples/chatgpt_demo/main/app/app_audio.c#L156 处增添 else 打印。发现在读到 20 个数以上时，buffer 确实会不够，从而导致数据丢失。所以您可以根据您的需求酌情增大 buffer 大小。

LHYhorion, 2024-04-20:
经过测试查到原因了，因为每次保存的是一个 16 位的音频样本，所以这里应该处理为 https://github.com/espressif/esp-box/blob/78531c7a26a68eab6c2e6788b10d9eeaaf97e23d/examples/chatgpt_demo/main/app/app_audio.c#L150 record_total_len += 2；

welkinchan, 2024-04-22:
> 经过测试查到原因了，因为每次保存的是一个 16 位的音频样本，所以这里应该处理为
> 
> https://github.com/espressif/esp-box/blob/78531c7a26a68eab6c2e6788b10d9eeaaf97e23d/examples/chatgpt_demo/main/app/app_audio.c#L150
> 
> record_total_len += 2；

我把代码改成record_total_len += 2；之后，ASR识别出来的结果全都不正确。
是不是别的哪个地方也要配合着改？

LHYhorion, 2024-04-22:
抱歉我没有考虑周全 https://github.com/espressif/esp-box/blob/78531c7a26a68eab6c2e6788b10d9eeaaf97e23d/examples/chatgpt_demo/main/app/app_audio.c#L185 正确的改法应该是改为 record_total_len *= 2; 同时建议将 audio_buffer 增大，因为测试发现会有溢出的情况发生。

welkinchan, 2024-04-28:
> 抱歉我没有考虑周全
> 
> https://github.com/espressif/esp-box/blob/78531c7a26a68eab6c2e6788b10d9eeaaf97e23d/examples/chatgpt_demo/main/app/app_audio.c#L185
> 
> 正确的改法应该是改为 record_total_len *= 2; 而不是之前的更改。同时建议将 audio_buffer 增大，因为测试发现会有溢出的情况发生。

今天测试了一下，问题确实解决了！
太给力了！非常感谢！

closed 2024-04-28 by welkinchan

Pan-yonghong, 2024-04-28:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

opsnull, 2024-05-13:
> 抱歉我没有考虑周全
> 
> https://github.com/espressif/esp-box/blob/78531c7a26a68eab6c2e6788b10d9eeaaf97e23d/examples/chatgpt_demo/main/app/app_audio.c#L185
> 
> 正确的改法应该是改为 record_total_len *= 2; 而不是之前的更改。同时建议将 audio_buffer 增大，因为测试发现会有溢出的情况发生。

没看到这个修复代码提交到仓库，我来提一个？