espressif/esp-box #143: ESP32-S3-BOX-3 上电后显示白屏

opsnull, 2024-05-11:
HI，我在用 Rust 开发 ESP32 应用，使用类似于下面的代码来初始化 LCD SPI，但是烧写后 LCD 显示白屏。求助下该如何解决？
```rust
        let sclk = io.pins.gpio7;
        let mosi = io.pins.gpio6;
        let miso = io.pins.gpio9;
        let cs = io.pins.gpio5;
        let dc = io.pins.gpio4.into_push_pull_output();
        let mut backlight = io.pins.gpio47.into_push_pull_output();
        let mut reset = io.pins.gpio48.into_push_pull_output();
        reset.internal_pull_up(true);

        let spi = Spi::new(
            peripherals.SPI3,
            sclk,
            mosi,
            miso,
            cs,
            40u32.MHz(),
            SpiMode::Mode0,
            &mut system.peripheral_clock_control,
            &clocks,
        );

        let di = display_interface_spi::SPIInterfaceNoCS::new(spi, dc);
        delay.delay_ms(500u32);

        let mut display = mipidsi::Builder::ili9341_rgb565(di)
            .with_display_size(320, 240)
            .with_framebuffer_size(320, 240)
            .with_orientation(mipidsi::Orientation::PortraitInverted(false))
            .with_color_order(mipidsi::ColorOrder::Bgr)
            .init(&mut delay, Some(reset))
            .unwrap();

        backlight.set_high().unwrap();
        display.clear(Rgb565::WHITE).unwrap();

        let default_style =
            MonoTextStyleBuilder::new().font(&FONT_10X20).text_color(RgbColor::BLACK).build();

        let espressif_style : MonoTextStyle<'_, Rgb565>=
            MonoTextStyleBuilder::new().font(&FONT_10X20).text_color(RgbColor::CYAN).build();

        for position_y in (25..240).step_by(28) {
            for position_x in 0..320 {
                Text::with_alignment(
                    "O",
                    Point::new(position_x, position_y),
                    default_style,
                    Alignment::Center,
                )
                .draw(&mut display)
                .unwrap();
            }
        }

```

[image]

opsnull, 2024-05-11:
这里的图片 https://github.com/espressif/esp-box/blob/master/docs/hardware_overview/esp32_s3_box_3/hardware_overview_for_box_3.md#technical-specification-1 显示 esp32_s3_box_3 使用的 Display Driver IC 是 ILI9342C，但是 esp-bsp 的代码 [esp-box-3.c](https://github.com/espressif/esp-bsp/blob/master/bsp/esp-box-3/esp-box-3.c#L36) 使用却是 ILI9341，两者不一致。

opsnull, 2024-05-12:
经过测试，白屏问题出在调用 ili9341 的 init() 方法时，不能传入 Some(reset) 参数，而是应该传入 None：

```
// https://github.com/almindor/mipidsi/issues/73
struct NoPin;
impl embedded_hal::digital::v2::OutputPin for NoPin {
    type Error = core::convert::Infallible;

    fn set_high(&mut self) -> Result<(), Self::Error> {
        Ok(())
    }

    fn set_low(&mut self) -> Result<(), Self::Error> {
        Ok(())
    }
}

// 引脚定义参考：https://github.com/espressif/esp-bsp/blob/master/bsp/esp-box-3/include/bsp/esp-box-3.h
        let sclk = io.pins.gpio7;
        let mosi = io.pins.gpio6;
        let cs = io.pins.gpio5.into_push_pull_output();
        let dc = io.pins.gpio4.into_push_pull_output();
        let mut backlight = io.pins.gpio47.into_push_pull_output();
        let mut reset = io.pins.gpio48.into_push_pull_output();
        reset.internal_pull_up(true); // 将 reset pin 设置为正常运行所需要的 high

        let spi = Spi::new_no_cs_no_miso(
            peripherals.SPI2,
            sclk,
            mosi,
            40u32.MHz(),
            SpiMode::Mode0,
            &mut system.peripheral_clock_control,
            &clocks,
        );

        let di = SPIInterface::new(spi, dc, cs);
        delay.delay_ms(500u32);

        // let mut display = mipidsi::Builder::ili9342c_rgb565(di)
        //     .with_display_size(320, 240)
        //     .with_framebuffer_size(320, 240)
        //     .with_orientation(mipidsi::Orientation::PortraitInverted(false))
        //     .with_color_order(mipidsi::ColorOrder::Bgr)
        //     .init(&mut delay, None::<NoPin>) 
        //     .unwrap();
        // backlight.set_high().unwrap();

        // LCD 配置参数：https://github.com/espressif/esp-bsp/blob/master/bsp/esp-box-3/include/bsp/display.h
        let mut display = mipidsi::Builder::ili9341_rgb565(di)
            .with_display_size(320, 240)
            .with_framebuffer_size(320, 240)
            .with_orientation(mipidsi::Orientation::PortraitInverted(false))
            .with_color_order(mipidsi::ColorOrder::Bgr)
            .init(&mut delay, None::<NoPin>) // 关键：不能传入 Some(reset) 来自动设置 reset pin ，否则重启后显示白屏。
            .unwrap();

        backlight.set_high().unwrap();
        display.clear(Rgb565::WHITE).unwrap();
```

另外，上面的 ili9341_rgb565 和 ili9342c_rgb565 都可以驱动 LCD 正常显示。

closed 2024-05-12 by opsnull

espressif2022, 2024-05-13:
> 经过测试，白屏问题出在调用 ili9341 的 init() 方法时，不能传入 Some(reset) 参数，而是应该传入 None：
>
> ```
> // https://github.com/almindor/mipidsi/issues/73
> struct NoPin;
> impl embedded_hal::digital::v2::OutputPin for NoPin {
>     type [...]

屏幕具体型号问题不大，主要是屏幕初始化传入的参数，所以我们并未详细将每个 IC 型号区分开来，而是通过用户导入自己的 vendor_config 初始化变量。

opsnull, 2024-05-13:
> 屏幕具体型号问题不大，主要是屏幕初始化传入的参数，所以我们并未详细将每个 IC 型号区分开来，而是通过用户导入自己的 vendor_config 初始化变量。

明白了，多谢解释！