espressif/esp-box #145: Blank Screen

yycguy123, 2024-05-13:
Just purchased this ESP32-S3_BOX-3B
powered up and screen is blank, its receiving power and top yellow light is on and off with mute button, it does connect and write when flashed, but screen is still blank

espressif2022, 2024-05-14:
Hello, can you provide the log after device poweron?

yycguy123, 2024-05-16:
When I connect with ESPLaunchpad I see this on the console if I select erase I get "invalid header: 0xffffffff"

esptool.js
Serial port WebSerial VendorID 0x303a ProductID 0x1001
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3
Features: Wi-Fi,BLE
Crystal is 40MHz
MAC: 84:fc:e6:64:08:f4
Uploading stub...
Running stub...
Stub running...
Changing baudrate to 921600
Changed
Manufacturer: c8
Device: 6018
Detected flash size: 16MB

espressif2022, 2024-05-16:
If it is "invalid header: 0xffffffff", your device should not have valid firmware.

yycguy123, 2024-05-16:
How do I resolve this ?

yycguy123, 2024-05-16:
So how to I fix this or is it broken I just got this Sent from my iPhone

yycguy123, 2024-05-17:
Attached is  my latest log file, I was able to flash and reset the device with launchpad, it appears the flash was successful but I can’t really tell, the display screen is blank, like its not on, no display no anything, and that’s the way its been from the start 
oel

> 
> 
> I've test my esp-box-3-B with launchpad  <https://espressif.github.io/esp-launchpad/?flashConfigURL=https://espressif.github.io/esp-box/launchpad.toml>, it's ok.
> Did you try some other bin, and the flash result is successfully?
> Writing at 0xe70e9e... (100%) Wrote 15157656 bytes (9477414 compressed) at 0x0 in 104.036 seconds.
￼
I (673) esp_image: segment 4: paddr=0036de8c vaddr=4037b160 size=114b0h ( 70832) load
I (688) esp_image: segment 5: paddr=0037f344 vaddr=600fe010 size=00004h (     4) load
I (699) boot: Loaded app from partition at offset 0x10000
I (699) boot: Disabling RNG early entropy source...
I (710) cpu_start: Multicore app
I (711) octal_psram: vendor id    : 0x0d (AP)
I (711) octal_psram: dev id       : 0x03 (generation 4)
I (714) octal_psram: density      : 0x05 (128 Mbit)
I (719) octal_psram: good-die     : 0x01 (Pass)
I (725) octal_psram: Latency      : 0x01 (Fixed)
I (730) octal_psram: VCC          : 0x00 (1.8V)
I (735) octal_psram: SRF          : 0x01 (Fast Refresh)
I (741) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (747) octal_psram: BurstLen     : 0x01 (32 Byte)
I (752) octal_psram: Readlatency  : 0x02 (10 ***@***.***)
I (758) octal_psram: DriveStrength: 0x00 (1/1)
I (764) MSPI Timing: PSRAM timing tuning index: 5
I (769) esp_psram: Found 16MB PSRAM device
I (774) esp_psram: Speed: 80MHz
I (856) mmu_psram: Instructions copied and mapped to SPIRAM
I (856) cpu_start: Pro cpu up.
I (856) cpu_start: Starting app cpu, entry point is 0x403758b4
I (0) cpu_start: App cpu up.
I (1391) esp_psram: SPI SRAM memory test OK
I (1400) cpu_start: Pro cpu start user code
I (1400) cpu_start: cpu freq: 240000000 Hz
I (1400) cpu_start: Application information:
I (1404) cpu_start: Project name:     chatgpt_demo
I (1409) cpu_start: App version:      1
I (1414) cpu_start: Compile time:     Apr 26 2024 14:20:26
I (1420) cpu_start: ELF file SHA256:  3cd987036b5f6482...
I (1426) cpu_start: ESP-IDF:          v5.1.3-470-g2fd087b246
I (1432) cpu_start: Min chip rev:     v0.0
I (1437) cpu_start: Max chip rev:     v0.99 
I (1442) cpu_start: Chip rev:         v0.2
I (1447) heap_init: Initializing. RAM available for dynamic allocation:
I (1454) heap_init: At 3FCA8F28 len 000407E8 (257 KiB): DRAM
I (1460) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (1467) heap_init: At 600FE014 len 00001FD4 (7 KiB): RTCRAM
I (1473) esp_psram: Adding pool of 15232K of PSRAM memory to heap allocator
I (1481) spi_flash: detected chip: gd
I (1485) spi_flash: flash io: qio
I (1490) sleep: Configure to isolate all GPIO pins in sleep state
I (1496) sleep: Enable automatic switching of GPIO sleep configuration
I (1504) app_start: Starting scheduler on CPU0
I (1509) app_start: Starting scheduler on CPU1
I (1509) main_task: Started on CPU0
I (1519) esp_psram: Reserving pool of 8K of internal memory for DMA/internal allocations
I (1527) main_task: Calling app_main()
I (1538) settings: stored ssid:My Network SSID
I (1538) settings: stored password:My Password
I (1542) settings: stored OpenAI:sk-xxxxxxxx
I (1547) settings: stored Base URL:https://api.openai.com/v1/
I (1678) ESP-BOX: Partition size: total: 1920401, used: 1072774
I (1678) LVGL: Starting LVGL task
I (1679) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1687) gpio: GPIO[48]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1817) gpio: GPIO[3]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:2 
E (1818) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed
E (1824) TT21100: esp_lcd_touch_tt21100_read_data(198): I2C read error!
E (1831) TT21100: esp_lcd_touch_new_i2c_tt21100(110): TT21100 init failed
[... 1146 lines cut ...]
I (482) esp_image: segment 1: paddr=00241d1c vaddr=3fc9c700 size=07194h ( 29076) load
I (487) esp_image: segment 2: paddr=00248eb8 vaddr=40374000 size=07160h ( 29024) load
I (493) esp_image: segment 3: paddr=00250020 vaddr=42000020 size=11de64h (1171044) map
I (673) esp_image: segment 4: paddr=0036de8c vaddr=4037b160 size=114b0h ( 70832) load
I (688) esp_image: segment 5: paddr=0037f344 vaddr=600fe010 size=00004h (     4) load
I (699) boot: Loaded app from partition at offset 0x10000
I (699) boot: Disabling RNG early entropy source...
I (710) cpu_start: Multicore app
I (711) octal_psram: vendor id    : 0x0d (AP)
I (711) octal_psram: dev id       : 0x03 (generation 4)
I (714) octal_psram: density      : 0x05 (128 Mbit)
I (719) octal_psram: good-die     : 0x01 (Pass)
I (725) octal_psram: Latency      : 0x01 (Fixed)
I (730) octal_psram: VCC          : 0x00 (1.8V)
I (735) octal_psram: SRF          : 0x01 (Fast Refresh)
I (741) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (747) octal_psram: BurstLen     : 0x01 (32 Byte)
I (752) octal_psram: Readlatency  : 0x02 (10 ***@***.***)
I (758) octal_psram: DriveStrength: 0x00 (1/1)
I (764) MSPI Timing: PSRAM timing tuning index: 5
I (769) esp_psram: Found 16MB PSRAM device
I (774) esp_psram: Speed: 80MHz
I (856) mmu_psram: Instructions copied and mapped to SPIRAM
I (856) cpu_start: Pro cpu up.
I (85

espressif2022, 2024-05-17:
`file: "./managed_components/espressif__esp-box/esp-box.c" line 387`
It seemd that you flash the wrong bin, your devide is box-3, you should choose this file` chatgpt_demo-ESP-BOX-3-1_0_0`.
Please try again.

yycguy123, 2024-05-17:
Ok that worked !!!! Thank you for the assistance, much appreciated Sent from my iPhone

closed 2024-06-11 by espressif2022