espressif/esp-box #146: example编译报错

nuomin0708, 2024-05-14:
在编译esp32-s3-box  v0.5.0版本的examples下的factory_demo报错，报错如下
[image]
像是组件版本不兼容，可是我没修改官方的代码，有什么方法解决吗？

espressif2022, 2024-05-14:
拉一下最新的 master 试下，部分组件可能更新了导致不匹配

nuomin0708, 2024-05-14:
> 拉一下最新的 master 试下，部分组件可能更新了导致不匹配

拉的就是最新的0.5.0版本，刚刚又试了下还是不行。

Horion0415, 2024-05-17:
您这边可以试下拉取 master 分支，编译试下。稍后我们会测试 0.5.0 是否可以正常使用

opsnull, 2024-05-17:
@nuomin0708 我看你用的是最近发布的 esp-idf-5.2.1 版本，可能与 esp-box 0.5.0 不兼容的, esp-box master 分支最近有几个 commit 来修复了与 esp-idf-5.2 兼容的问题，比如这个 
[example(chatgpt): fix build with idf 5.2](https://github.com/espressif/esp-box/commit/24219f7100722d62d9acd7956c44c90063bb1ec5) ，我试了下用  esp-idf-5.2.1 和 esp-box master 分支是可以正常编译的。

nuomin0708, 2024-05-17:
> 您这边可以试下拉取 master 分支，编译试下。稍后我们会测试 0.5.0 是否可以正常使用
n抱歉，我说的0.5.0可能就是master分支。
[image]

nuomin0708, 2024-05-17:
> 我看你用的是最近发布的 esp-idf-5.2.1 版本，可能与 esp-box 0.5.0 不兼容的， esp-box master 分支最近有几个 commit 来修复了与 esp-idf-5.2 兼容的问题，比如这个 [example（chatgpt）： fix build with idf 5.2](https://github.com/espressif/esp-box/commit/24219f7100722d62d9acd7956c44c90063bb1ec5) ，我试了下用 esp-idf-5.2.1 和 esp-box master 分支是可以正常编译的。

谢谢你的回复，我用的是5.2.1的，chatgpt_demo我可以正常编译。是factory_demo不可以编译成功。

espressif2022, 2024-07-01:
This issue is related to #154

ESP-Mars, 2024-07-02:
Is there any progress on this issue?

nuomin0708, 2024-07-02:
> Is there any progress on this issue?
我升级了idf manager的版本，解决了这个问题

closed 2024-07-02 by ESP-Mars

Quantriple, 2024-10-27:
最新的5.3.1版本，我编译了三四个例子 都没过去，包括 factory   image    mp3 这些例子