espressif/esp-box #148: esp s3 box cant be seen on windows machine

waitsmd41, 2024-05-27:
The s3 box was able to be seen just a few hours ago. after trying to run chatgpt demo i couldnt see the folder for esp box. it finally showed up and i changed the config.ini file with ssd and password and api key for chatgpt. 

it was flashing saying it was connecting to network. then it would flash black back and forth. 
[image]

i tried to reflash it with launch pad but now it wont even see the device. 
[image]
and windows doesnt even see it. 
[image]

someone please help 

is this box dead?

ESP-Mars, 2024-05-27:
If the downloading through USB fails, please hold down the Boot button and then press the Reset button to initiate Firmware Download mode manually.

ESP-Mars, 2024-07-02:
Is there any progress on this issue?

waitsmd41, 2024-07-02:
I gave up sometime ago.  Guess this hobby is by of my ability.  Sent from my iPhone

ESP-Mars, 2024-07-03:
Temporarily closed the issue. Should you have additional questions or concerns, don't hesitate to reopen the issue.

closed 2024-07-03 by ESP-Mars