espressif/esp-box #151: ESP-Box-3 showing white screen regardless of firmware, no boot animation

teknetia, 2024-06-12:
I have an ESP32-S3-BOX-3 that has since it came out of the box only shown a white screen, never showing anything as indicated in the manual, including the boot animation. To ensure it wasn't a firmware issue I have tried using the [launchpad tool](https://espressif.github.io/esp-launchpad/?flashConfigURL=https://espressif.github.io/esp-box/launchpad.toml) to re-flash it with a number of different potential images though all end up in the same place.

I notice with the factory demo that the logs indicate it seems stuck at the wifi initialisation step, if I press the mute/unmute button though I will see some output (includes in the detail blocks below). I've left it connected to the launch pad for over an hour in case there were just slow things going on but no change.

I've also tried the manual option (press and hold boot then press reset) but that doesn't result in any different behaviour. I've also tried it with different machines (macOS and Windows with Chrome) and with the main box docked and not docked (in case the sensors in the sensor dock were contributing at all).

Is there anything else I can troubleshoot to ensure there isn't a hardware fault?

<details><summary>Flash logs - chatgpt_demo-ESP-BOX-3-1_0_0</summary>

```
esptool.js
Serial port WebSerial VendorID 0x303a ProductID 0x1001
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3
Features: Wi-Fi,BLE
Crystal is 40MHz
MAC: 3c:84:27:05:a7:c8
Uploading stub...
Running stub...
Stub running...
Changing baudrate to 921600
Changed
Manufacturer: c8
Device: 6018
Detected flash size: 16MB
Warning: Image file at 0x0 doesn't look like an image file, so not changing any flash settings.
Compressed 14002698 bytes to 4682680...
Writing at 0x0... (0%)
Writing at 0x10f78... (0%)
Writing at 0x1aa05... (1%)
Writing at 0x25e2a... (1%)
Writing at 0x3505a... (1%)
Writing at 0x3d472... (2%)
Writing at 0x44102... (2%)
Writing at 0x4a8fe... (2%)
Writing at 0x5100d... (3%)
Writing at 0x580db... (3%)
Writing at 0x5e849... (3%)
Writing at 0x64c07... (4%)
Writing at 0x6b6c0... (4%)
Writing at 0x72097... (4%)
Writing at 0x78a45... (5%)
Writing at 0x7f4a5... (5%)
Writing at 0x8594e... (5%)
Writing at 0x8c0b1... (6%)
Writing at 0x9250b... (6%)
Writing at 0x9896d... (6%)
Writing at 0x9f329... (7%)
Writing at 0xa5b6e... (7%)
Writing at 0xabea0... (8%)
Writing at 0xb282e... (8%)
Writing at 0xb8b52... (8%)
Writing at 0xbed04... (9%)
Writing at 0xc519c... (9%)
Writing at 0xcb4f7... (9%)
Writing at 0xd1aca... (10%)
Writing at 0xd7e4e... (10%)
Writing at 0xde65c... (10%)
Writing at 0xe4a6d... (11%)
Writing at 0xeab46... (11%)
Writing at 0xf0c5b... (11%)
Writing at 0xf6f38... (12%)
Writing at 0xfd9bc... (12%)
Writing at 0x104278... (12%)
Writing at 0x10a53c... (13%)
Writing at 0x11062f... (13%)
Writing at 0x116e9f... (13%)
Writing at 0x11d563... (14%)
[... 1411 lines cut ...]
I (781) esp_psram: Adding pool of 16384K of PSRAM memory to heap allocator
I (782) spi_flash: detected chip: gd
I (782) spi_flash: flash io: qio
I (782) sleep: Configure to isolate all GPIO pins in sleep state
I (783) sleep: Enable automatic switching of GPIO sleep configuration
I (783) app_start: Starting scheduler on CPU0
I (784) app_start: Starting scheduler on CPU1
I (784) main_task: Started on CPU0
I (794) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (794) main_task: Calling app_main()
I (794) LVGL: Starting LVGL task
I (794) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (794) gpio: GPIO[48]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (794) ili9341: LCD panel create success, version: 1.2.0
W (914) ili9341: The 36h command has been used and will be overwritten by external initialization sequence
W (914) ili9341: The 3Ah command has been used and will be overwritten by external initialization sequence
I (914) gpio: GPIO[3]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:2 
I (914) GT911: TouchPad_ID:0x39,0x31,0x31
I (914) GT911: TouchPad_Config_Version:65
I (914) ESP-BOX-3: Setting LCD backlight: 100%
I (1004) ESP-BOX-3: Partition size: total: 956561, used: 170429
I (1044) main_task: Returned from app_main()
```

</details>

espressif2022, 2024-06-12:
Is the factory firmware normal before you flash it?

ESP-Mars, 2024-06-12:
@teknetia 
1. Please refer to the disassembly Tutorial to check if the fpc cable of LCD has is loose. https://github.com/espressif/esp-box/blob/master/docs/hardware_overview/esp32_s3_box_3/disassembly_tutorial_3.md
2. To further troubleshoot the potential hardware issue, please flash the following factory test firmware (Sorry that the firmware user language is in Chinese only). Normally, each device undergoes rigorous factory testing before leaving the factory, but in order to facilitate troubleshooting, please use this firmware to conduct additional testing.

https://cowtransfer.com/s/e104ceb4637544 click link to check [ esp32_s3_box_3_factory_test_firmware.bin ] or access cowtransfer.com and input extract code: iqqtft to check;

teknetia, 2024-06-16:
@espressif2022 No, this white screen issue has been since it came out of the box.

@ESP-Mars No loose cables, but I reseated them all jut in case. Nothing changed as a result. I flashed the factory test firmware and log are below but the screen still lights up entirely white only.

<details><summary><b>Factory test logs</b></summary>

```
esptool.js
Serial port WebSerial VendorID 0x303a ProductID 0x1001
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3
Features: Wi-Fi,BLE
Crystal is 40MHz
MAC: 3c:84:27:05:a7:c8
Uploading stub...
Running stub...
Stub running...
Changing baudrate to 921600
Changed
Manufacturer: c8
Device: 6018
Detected flash size: 16MB
Warning: Image file at 0x0 doesn't look like an image file, so not changing any flash settings.
Compressed 1091024 bytes to 657553...
Writing at 0x0... (2%)
Writing at 0x1358f... (4%)
Writing at 0x208ae... (7%)
Writing at 0x26c37... (9%)
Writing at 0x3074f... (12%)
Writing at 0x3aa99... (14%)
Writing at 0x45b4f... (17%)
Writing at 0x52bf8... (19%)
Writing at 0x5830d... (21%)
Writing at 0x5eab8... (24%)
Writing at 0x64354... (26%)
Writing at 0x699f4... (29%)
Writing at 0x6e8fe... (31%)
Writing at 0x73dfd... (34%)
Writing at 0x79554... (36%)
Writing at 0x7ea2a... (39%)
Writing at 0x837e7... (41%)
Writing at 0x89b70... (43%)
Writing at 0x8eda9... (46%)
Writing at 0x93fd6... (48%)
Writing at 0x98f05... (51%)
Writing at 0x9e3b3... (53%)
Writing at 0xa3331... (56%)
Writing at 0xa9154... (58%)
Writing at 0xaf3ad... (60%)
Writing at 0xb48f2... (63%)
Writing at 0xbabd9... (65%)
Writing at 0xc017e... (68%)
Writing at 0xc65fc... (70%)
Writing at 0xcef00... (73%)
Writing at 0xd6595... (75%)
Writing at 0xdb74e... (78%)
Writing at 0xe0400... (80%)
Writing at 0xe4b30... (82%)
Writing at 0xe92db... (85%)
Writing at 0xed984... (87%)
Writing at 0xf213c... (90%)
Writing at 0xf9077... (92%)
Writing at 0xfe2a1... (95%)
Writing at 0x103bf8... (97%)
Writing at 0x109945... (100%)
Wrote 1091024 bytes (657553 compressed) at 0x0 in 7.205 seconds.
Leaving...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x40378c1b
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x4f0
load:0x403c9700,len:0x4
load:0x403c9704,len:0xb5c
load:0x403cc700,len:0x2c08
entry 0x403c98fc
I (180) cpu_start: Multicore app
I (181) octal_psram: vendor id    : 0x0d (AP)
I (181) octal_psram: dev id       : 0x03 (generation 4)
I (181) octal_psram: density      : 0x05 (128 Mbit)
I (181) octal_psram: good-die     : 0x01 (Pass)
I (182) octal_psram: Latency      : 0x01 (Fixed)
I (182) octal_psram: VCC          : 0x00 (1.8V)
I (182) octal_psram: SRF          : 0x01 (Fast Refresh)
I (182) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (183) octal_psram: BurstLen     : 0x01 (32 Byte)
I (183) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (183) octal_psram: DriveStrength: 0x00 (1/1)
W (184) MSPI Timing: tuning fail, best point is fallen back to index 5
I (185) esp_psram: Found 16MB PSRAM device
I (185) esp_psram: Speed: 80MHz
I (223) mmu_psram: Instructions copied and mapped to SPIRAM
I (238) mmu_psram: Read only data copied and mapped to SPIRAM
I (238) cpu_start: Pro cpu up.
I (238) cpu_start: Starting app cpu, entry point is 0x40375678
I (0) cpu_start: App cpu up.
I (780) esp_psram: SPI SRAM memory test OK
I (789) cpu_start: Pro cpu start user code
I (789) cpu_start: cpu freq: 240000000 Hz
I (789) cpu_start: Application information:
I (789) cpu_start: Project name:     production_firmware
I (789) cpu_start: App version:      1
I (789) cpu_start: Compile time:     Sep 14 2023 14:47:57
I (790) cpu_start: ELF file SHA256:  d0cc9af20...
I (790) cpu_start: ESP-IDF:          v5.2-dev-1010-g213504238f-dirty
I (790) cpu_start: Min chip rev:     v0.0
I (790) cpu_start: Max chip rev:     v0.99 
I (790) cpu_start: Chip rev:         v0.2
I (791) heap_init: Initializing. RAM available for dynamic allocation:
I (791) heap_init: At 3FCC4B78 len 00024B98 (146 KiB): DRAM
I (791) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (791) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (792) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (792) esp_psram: Adding pool of 15552K of PSRAM memory to heap allocator
I (792) spi_flash: detected chip: gd
I (792) spi_flash: flash io: qio
I (793) sleep: Configure to isolate all GPIO pins in sleep state
I (793) sleep: Enable automatic switching of GPIO sleep configuration
I (794) app_start: Starting scheduler on CPU0
I (794) app_start: Starting scheduler on CPU1
I (794) main_task: Started on CPU0
I (795) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (795) main_task: Calling app_main()
I (796) gpio: GPIO[44]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (799) LVGL: Starting LVGL task
I (800) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (800) gpio: GPIO[48]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (923) ESP-BOX-3: Addr 5D
I (923) gpio: GPIO[3]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:2 
I (924) GT911: TouchPad_ID:0x39,0x31,0x31
I (924) GT911: TouchPad_Config_Version:65
I (927) ES8311: Work in Slave mode
I (929) gpio: GPIO[46]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (930) ES7210: Work in Slave mode
I (933) ES7210: Enable ES7210_INPUT_MIC1
I (934) ES7210: Enable ES7210_INPUT_MIC2
I (937) ESP-BOX-3: Setting LCD backlight: 70%
I (1001) button: IoT Button Version: 2.5.6
I (1001) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1001) ESP-BOX: ESP-BOX initialization done.
```

</details>

teknetia, 2024-06-27:
Just checking in on this as I am still unable to get any functional firmware/screen on this device.

ESP-Mars, 2024-07-02:
@teknetia We apologize for the issues you're experiencing with the hardware you received. The logs indicate that the LCD on your device is abnormal. Please contact customer service to initiate the return and exchange process.
In the manufacturer, each device should undergoe rigorous factory testing before leaving the factory, we will investigate the production records for your device to identify the issue and continually improve our product quality.

Please accept our sincere apologies for any inconvenience this has caused you.

teknetia, 2024-07-02:
Will do, thanks for confirming!

closed 2024-07-02 by teknetia