espressif/esp-box #154: Cannot build factory demo

rjk100, 2024-06-22:
Using component placed at C:\git\esp-box\examples\factory_demo\components\espressif__esp_schedule for dependency espressif/esp_schedule(~1.0.0)(introduced by component espressif/esp_rainmaker), specified in C:/git/esp-box/examples/factory_demo/main\idf_component.yml
....................................HINT: Please check manifest file of the following component(s): main
CMake Error at C:/Espressif/frameworks/esp-idf-v5.2.1/tools/cmake/build.cmake:544 (message):
  ERROR: Because espressif/esp_rainmaker (1.0.0) depends on
  espressif__esp_schedule (1.1.0)
 
   and espressif__esp_schedule (1.1.0) depends on espressif/rmaker_common (~1.4.2), espressif/esp_rainmaker (1.0.0) requires espressif/rmaker_common (~1.4.2).
 
  So, because no versions of espressif/rmaker_common match ~1.4.2
 
   and project depends on espressif/esp_rainmaker (1.0.0), version solving failed.
 
Call Stack (most recent call first):
  C:/Espressif/frameworks/esp-idf-v5.2.1/tools/cmake/project.cmake:605 (idf_build_process)
  CMakeLists.txt:16 (project)
 
 
-- Configuring incomplete, errors occurred!
See also "C:/git/esp-box/examples/factory_demo/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the C:\git\esp-box\examples\factory_demo\build\log\idf_py_stderr_output_24636 and C:\git\esp-box\examples\factory_demo\build\log\idf_py_stdout_output_24636

espressif2022, 2024-06-24:
Hello you can use this command `pip install idf-component-manager==2.0.0.dev0` to update your idf-component-manage version.

ESP-Mars, 2024-07-23:
Temporarily closed the issue. Should you have additional questions or concerns, don't hesitate to reopen the issue.

closed 2024-07-23 by ESP-Mars