espressif/esp-box #156: ESP-Box-3 showing white screen regardless of firmware

Xatinni, 2024-07-01:
I received my ESP32-S3-BOX-3 for Mouser Electonics, Invoice 79666586, They have advised to open an issue on github.

I believe the TFT screen is faulty, Ive tried to reset and get the following error.. I've also tried to flash with a couple off different official firmware, see YouTube clips  

https://www.youtube.com/watch?v=jQN2anmFOQU
https://www.youtube.com/shorts/Gdjw_0Y07HA

I get the following error when resetting device E (1573) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed E (1574) TT21100: esp_lcd_touch_tt21100_read_data(173): I2C read error! E (1574) TT21100: esp_lcd_touch_new_i2c_tt21100(103): TT21100 init failed E (1574) TT21100: Error (0xffffffff)! Touch controller TT21100 initialization failed! I (1574) gpio: GPIO[3]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 ESP_ERROR_CHECK failed: esp_err_t 0xffffffff at 0x4203a70e file: "./managed_components/espressif__esp-box/esp-box.c" line 389 func: bsp_display_indev_init expression: bsp_touch_new(((void *)0), &tp) abort() was called at PC 0x403889a3 on core 0

I have dismantled and reseated all cables and this has not resolved the issue.

[image]
[image]

espressif2022, 2024-07-01:
This log is not normal `espressif__esp-box/esp-box.c`
I think you flash the wrong fw, can you try this [link](https://espressif.github.io/esp-launchpad/?flashConfigURL=https://espressif.github.io/esp-box/launchpad.toml) ?

Xatinni, 2024-07-01:
Thank you, but this did not resolve the issue 

https://youtu.be/kbZrwSpZujs?si=ekVPu7W_UGxVaBJB

espressif2022, 2024-07-01:
Hello, your box is ESP-box-3, I think you should choose this fw `chatgpt_demo-ESP-BOX-3_1_0_0`

Xatinni, 2024-07-01:
Thank you, but this did not resolve the issue

https://www.youtube.com/watch?v=vfQcZ09mVLo

ESP-Mars, 2024-07-01:
@Xatinni Could you try to flash 'factory_demo-ESP-BOX-3-1_2_4'? Thanks.

Xatinni, 2024-07-01:
Thanks, unfortunately this did not resolve the issue,

https://youtu.be/PS-tMefJ2NI?si=xlCdNFqBuOXPfHC0

ESP-Mars, 2024-07-02:
@Xatinni Is the display normal since it came out of the box?

Xatinni, 2024-07-02:
No. I suspect it's an RMA.

ESP-Mars, 2024-07-02:
@Xatinni We apologize for the issues you're experiencing with the hardware you received. Please contact Mouser Electonics customer service to initiate the return and exchange process.
In the manufacturer, each device should undergoe rigorous factory testing before leaving the factory, we will investigate the production records for your device to identify the issue and continually improve our product quality.

Please accept our sincere apologies for any inconvenience this has caused you.

closed 2025-01-04 by Xatinni