espressif/esp-box #157: Atecc608A I2C address in ESP32-S3-BOX-3 device

DeppLearning, 2024-07-01:
Hey there, 

I'm trying to use the atecc608a-tngtls chip built in the ESP32-S3-BOX-3. I am using GPIO8 for SDA and GPIO18 for SCL but can't find the device in an I2C scan, nor does it respond to my driver. From the schematic it looks like VCC of the atecc is just connected to MCU_VDD. Is there anything that I am missing, e.g. any extra pin I should be driving apart SDA/SCL? 

What would be the default address of the device? 

I'd expect the default I2C address to be 0x60, maybe 0x35 (since it's the tngtls variant) but these don't work. When scanning, I am finding devices at I2C addresses 0x19 (probably es8311 codec), 0x69 (probably icm-42607 imu). Based on the schematics I'd have expected a es7210 - dac  at 0x20 or 0x21 but not seeing that one. Also I'm finding some device at 0x5e, not sure what that is, and I'm using the touch controller at 0x5d with no issues. This is at 400/100khz bus frequency. At 1khz bus frequency I'm finding 0x18, 0x40 and 0x68, but neither of those respond to my driver either.

When connecting an Adafruit Atecc608A Breakout board to the DOCK accessory  on GPIO40 and 41 I'm having no issues scanning or talking to the device with the same driver. The adafruit board has external pull-up resistors in place, so I also tried a Atecct608B without pull-up resistors, with no issues. So pull-up resistors shouldn't be the issue I guess. Afaik the atecc608a-tngtls variant should just be a normal atecc608a that is pre-provisioned and thus should respond to a "normal" atecc608x driver.

I would be very glad for any pointers what I might be doing wrong :)

espressif2022, 2024-07-02:
Hello，Pull-up resistors should have an impact, and when so many I2C slaves are connected to the bus.
When we were designing the development board, we also found that the pull-up resistors did not match, causing some slaves to not be found.

ESP-Mars, 2025-04-01:
This issue has remained inactive for an extended period. Closing it now.

closed 2025-04-01 by ESP-Mars