espressif/esp-box #158: 能用Slint编写esp-box的UI界面吗

jianjiabailu, 2024-07-06:
我跑通了官方的Slint例子，编译并烧录到esp-box中去了，但是屏幕没有显示，也找不到原因，谁能告诉我该怎么做？
https://components.espressif.com/components/slint/slint

ESP-Mars, 2024-07-07:
是否有烧录后的log？我们看一下。
看官方的Slint还是用的老版本的 EP32-S3-BOX，您手上的是升级版 EP32-S3-BOX-3（看论坛有一样的问题应该也是您提问的https://www.esp32.com/viewtopic.php?f=43&t=40751&p=134522#p1345220，两套板子 bsp 不一样需要注意。

opsnull, 2024-07-07:
可以参考我移植到 EP32-S3-BOX-3 的 slint-printdemo ：
https://github.com/opsnull/rust-slint-printdemo

这个 Repo 里包含一个我用 Rust 写的 [GT911 触摸板驱动](https://github.com/opsnull/rust-slint-printdemo/blob/main/mcu-board-support/gt911.rs)。

jianjiabailu, 2024-07-08:
> 是否有烧录后的 log？我们看一下。 看官方的 Slint 还是用的老版本的 EP32-S3-BOX，您手上的是升级版 EP32-S3-BOX-3（看[论坛](https://www.esp32.com/viewtopic.php?f=43&t=40751&p=134522#p1345220)有一样的问题应该也是您提的），两套板子 [Board Support Package](https://github.com/espressif/esp-bsp/tree/master/bsp/esp-box-3) 不一样需要注意。

[image]
这是烧录之后的端口监听日志，还有屏幕显示
[image]

jianjiabailu, 2024-07-08:
> 可以参考我移植到 EP32-S3-BOX-3 的 slint-printdemo ： https://github.com/opsnull/rust-slint-printdemo
> 
> 这个 Repo 里包含一个我用 Rust 写的 [GT911 触摸板驱动](https://github.com/opsnull/rust-slint-printdemo/blob/main/mcu-board-support/gt911.rs)。

你的这个例子我编译成功了，但是烧录之后无法点击屏幕，是没有连接触摸驱动吗？

jianjiabailu, 2024-07-08:
> > 可以参考我移植到 EP32-S3-BOX-3 的 slint-printdemo ： https://github.com/opsnull/rust-slint-printdemo
> > 这个 Repo 里包含一个我用 Rust 写的 [GT911 触摸板驱动](https://github.com/opsnull/rust-slint-printdemo/blob/main/mcu-board-support/gt911.rs)。
> 
> 你的这个例子我编译成功了，但是烧录之后无法点击屏幕，是没有连接触摸驱动吗？

[2024-07-08T03:27:56Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x40378bdd
0x40378bdd - save_context
    at C:\Users\105352\.cargo\registry\src\index.crates.io-6f17d22bba15001f\xtensa-lx-rt-0.16.0\src\exception\assembly_esp32.rs:104
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3818,len:0x16f8
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
I (24) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (24) boot: compile time Jun  7 2023 08:07:32
I (25) boot: Multicore bootloader
I (29) boot: chip revision: v0.2
I (33) boot.esp32s3: Boot SPI Speed : 40MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (42) boot.esp32s3: SPI Flash Size : 16MB
I (47) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 factory          factory app      00 00 00010000 00fa0000
I (86) boot: End of partition table
I (90) esp_image: segment 0: paddr=00010020 vaddr=3c0f0020 size=42d50h (273744) map
I (167) esp_image: segment 1: paddr=00052d78 vaddr=3fc891c0 size=00064h (   100) load
I (167) esp_image: segment 2: paddr=00052de4 vaddr=40378000 size=011c0h (  4544) load
I (174) esp_image: segment 3: paddr=00053fac vaddr=00000000 size=0c06ch ( 49260)
I (193) esp_image: segment 4: paddr=00060020 vaddr=42000020 size=e2e24h (929316) map
I (425) boot: Loaded app from partition at offset 0x10000

ESP-Mars, 2025-04-01:
This issue has remained inactive for an extended period. Closing it now.

closed 2025-04-01 by ESP-Mars