espressif/esp-box #160: 代码报错

NanaZhang4524, 2024-07-08:
[image]
这个报错什么意思啊？之前这些代码是从实例库里截取的……

NanaZhang4524, 2024-07-08:
【金山文档 | WPS云文档】 新建 DOCX 文档
https://kdocs.cn/l/cpcFKxhqY1OA
这个代码报的错

espressif2022, 2024-07-09:
首先，这些代码应该不是我们官方仓库提供的吧？

如果需要实现此功能，可以对比下[这里](https://github.com/espressif/esp-dev-kits/blob/1f3401e7386adab35617ec0d010c3d4620768b97/esp32-s3-lcd-ev-board/examples/86box_smart_panel/main/app/app_weather.c#L203C26-L203C42)的代码

NanaZhang4524, 2024-07-09:
> 首先，这些代码应该不是我们官方仓库提供的吧？
> 
> 如果需要实现此功能，可以对比下[这里](https://github.com/espressif/esp-dev-kits/blob/1f3401e7386adab35617ec0d010c3d4620768b97/esp32-s3-lcd-ev-board/examples/86box_smart_panel/main/app/app_weather.c#L203C26-L203C42)的代码
esp32s3box3b适合直接用这套代码吗？或者是直接烧录？上面的报错的代码是参考esp_http_client编写的代码，再加上根据视频网站ESP教程写的代码。

espressif2022, 2024-07-09:
> 首先，这些代码应该不是我们官方仓库提供的吧？ [...]

不太一样，BSP 不一样。但是模组是一个，你可以直接移植，且这些是业务代码，和哪个硬件不相关

ESP-Mars, 2025-04-01:
This issue has remained inactive for an extended period. Closing it now.

closed 2025-04-01 by ESP-Mars