espressif/esp-box #161: gpt_demo例程：增加i2s_stream处理后，无法唤醒设备。

welkinchan, 2024-08-04:
我最近在尝试给gpt_demo的tts语音增加i2s_stream的流式语音播放功能，
但是测试下来，发现一个问题，就是i2s_stream的代码执行一次之后，就无法再用唤醒指令唤醒设备了。

具体说明如下：
1. 我当前的代码中，当使用http tts的功能，通过http协议请求到tts的完整语音包后，再播放tts语音，程序可以多次语音唤醒设备，功能没有异常。请见附件的http tts的log。
2. 但是，当我使用i2s_stream，通过i2s的pipeline来进行流式播放tts的语音后，当语音播放完毕后，就无法再语音唤醒设备了，无论我在tts语音播放后是否清除pipeline，都无法唤醒设备。i2s_stream方案的代码以及调试log请见附件。

从我个人分析来看，当使用i2s的pipeline进行tts语音的播放后，audio_feed_task 以及audio_detect_task 都是正常在执行的。

使用http tts思路的代码，tts语音播放后，如下log中的i2s_if以及adev_codec的这几行代码会执行；
而使用i2s_pipeline tts的代码，log中的这几行代码就不会执行了。
所以，请问：
1. 能否帮忙分析一下，是什么原因导致无法唤醒的？
2. 我的i2s_pipeline tts的代码能否怎么修改后可以不影响语音唤醒功能？
3. 以下log中的几行代码是在哪个地方控制运行的？是不是由于缺少了相关函数的调用执行，所以导致无法唤醒？
4. 如果i2s_pipeline tts与语音唤醒冲突，请问是否有别的思路可以实现tts语音播放的流式处理？

I (20373) I2S_IF: Pending out channel for in channel running
E (20389) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (20389) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (20393) I2S_IF: STD Mode 1 bits:16/16 channel:2 sample_rate:16000 mask:3
I (20503) Adev_Codec: Open codec device OK
E (20503) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (20505) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (20511) I2S_IF: STD Mode 0 bits:16/16 channel:2 sample_rate:16000 mask:3
I (20519) ES7210: Bits 16
I (20525) ES7210: Enable ES7210_INPUT_MIC1
I (20528) ES7210: Enable ES7210_INPUT_MIC2
I (20535) ES7210: Unmuted
I (20535) Adev_Codec: Open codec device OK

[http tts可以多次唤醒.txt](https://github.com/user-attachments/files/16487718/http.tts.txt)
[i2s tts第二次无法唤醒.txt](https://github.com/user-attachments/files/16487719/i2s.tts.txt)
[i2s_stream tts代码.txt](https://github.com/user-attachments/files/16487720/i2s_stream.tts.txt)

espressif2022, 2024-08-05:
你这用了 ADF 的 I2S pipline 接口？

welkinchan, 2024-08-05:
> 你这用了 ADF 的 I2S pipeline 接口？

是的。代码可见
[i2s_stream tts代码.txt](https://github.com/user-attachments/files/16487720/i2s_stream.tts.txt)

espressif2022, 2024-08-05:
> > 你这用了 ADF 的 I2S pipeline 接口？
> 
> 是的。代码可见 [i2s_stream tts代码.txt](https://github.com/user-attachments/files/16487720/i2s_stream.tts.txt)

[image]
我看你播放的音源改了  I2S 的配置，上传给 chatgpt 的录音处理有没有同步修改，是不是两边对不上了？

welkinchan, 2024-08-05:
> > > 你这用了 ADF 的 I2S pipeline 接口？
> > 
> > 
> > 是的。代码可见 [i2s_stream tts代码.txt](https://github.com/user-attachments/files/16487720/i2s_stream.tts.txt)
> 
> ![企业微信截图_17228261912453](https://private-user-images.githubusercontent.com/111102666/354955425-5714ef86-6bfd-4f0b-8646-6f867587b2ee.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI4Mjg2NjIsIm5iZiI6MTcyMjgyODM2MiwicGF0aCI6Ii8xMTExMDI2NjYvMzU0OTU1NDI1LTU3MTRlZjg2LTZiZmQtNGYwYi04NjQ2LTZmODY3NTg3YjJlZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgwNVQwMzI2MDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01MTg5ODY4MzQ3YzcyYjU3NWNjNTUwNzA0YzdmZmVkNzc3Yjg0ZmNhMjM0ZDE1ODFlYzAyYzc0NGFmMGM3Y2FhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.bnum_pfP1DOLL-AgYdo8T7qoftCC3SJCZTPwcowLSEg) 我看你播放的音源改了 I2S 的配置，上传给 chatgpt 的录音处理有没有同步修改，是不是两边对不上了？

上传给 chatgpt 的录音处理我没有进行任何的修改，几乎是保留原来的代码。

i2s_stream这部分的代码我是参考google_translate的例程写的，我其实不是很理解这句代码的作用。
1. 我怀疑过gpt_demo里边是不是也用了pipeline，或者i2s_stream。
但是找了很久，没找到gpt_demo代码中有i2s_stream相关的代码。
2. 我印象里边，好像有资料说pipeline还是i2s_stream可以有多个。
如果我在这里新定义一个i2s_stream，根据你的意思，是不是说会同时影响到录音部分的i2s通道？
3. 我的代码是不是应该这么更正？根据目前gpt_demo中的录音的配置更改我的i2s_stream_cfg_t i2s_cfg = I2S_STREAM_CFG_DEFAULT_WITH_PARA的配置项？

welkinchan, 2024-08-05:
> 给 chatgpt 的录音处理有没有同步修改，是不是两边对不

1. 我把i2s_stream改成了立体声，测试下来，还是无法唤醒。
    i2s_cfg.std_cfg.slot_cfg.slot_mode = I2S_SLOT_MODE_STEREO;
    i2s_cfg.std_cfg.slot_cfg.slot_mask = I2S_STD_SLOT_BOTH;
    
 2. 我把slot_mode以及slot_mask这两行代码注释掉，也是无法唤醒的。

3. 在这个阶段进行录音的代码我还没尝试。

4. 我尝试在上述更改的基础上，并且在i2s_stream的这段tts代码的末尾，
    也就是执行baidu_tts_destroy(tts);后，再执行bsp_codec_set_fs(16000, 16, 2);，此时log会多出如下记录，但是此时还是无法唤醒。

W (30824) HTTP_STREAM: No more data,errno:0, total_bytes:66240, rlen = 0
I (30824) AUDIO_ELEMENT: IN-[tts_http] AEL_IO_DONE,0
I (31835) AUDIO_ELEMENT: IN-[tts_mp3] AEL_IO_DONE,-2
I (32049) MP3_DECODER: Closed
I (32203) AUDIO_ELEMENT: IN-[tts_i2s] AEL_IO_DONE,-2
W (33755) app_sr: cwg================audio_feed_task running=============feed_task_while_times=100 again
W (33820) app_sr: cwg================audio_detect_task running=============while_times=100 again
W (36114) app_main:     Description     Internal        SPIRAM
W (36114) app_main: Current Free Memory 89531           2263724
W (36114) app_main: Min. Ever Free Size 86687           2213080
W (37035) app_sr: cwg================audio_detect_task running=============while_times=100 again
W (40220) app_sr: cwg================audio_feed_task running=============feed_task_while_times=100 again
W (40284) app_sr: cwg================audio_detect_task running=============while_times=100 again
W (41126) app_main:     Description     Internal        SPIRAM
W (41126) app_main: Current Free Memory 89531           2263724
W (41126) app_main: Min. Ever Free Size 86687           2213080
I (43397) TTS-Api: [ 7 ] cwg tts stream==========pdMS_TO_TICKS(15000)==============DONE
I (43397) TTS-Api: [ 6 ] cwg tts stream==========pdMS_TO_TICKS(15000)===============Stop audio_pipeline
W (43404) AUDIO_ELEMENT: [tts_http] Element already stopped
W (43411) AUDIO_ELEMENT: [tts_mp3] Element already stopped
W (43417) AUDIO_ELEMENT: [tts_i2s] Element already stopped
W (43425) AUDIO_ELEMENT: [tts_http] Element has not create when AUDIO_ELEMENT_TERMINATE
W (43432) AUDIO_ELEMENT: [tts_mp3] Element has not create when AUDIO_ELEMENT_TERMINATE
W (43440) AUDIO_ELEMENT: [tts_i2s] Element has not create when AUDIO_ELEMENT_TERMINATE
W (43449) AUDIO_PIPELINE: There are no listener registered
I (43455) AUDIO_PIPELINE: audio_pipeline_unlinked
W (43460) AUDIO_ELEMENT: [tts_http] Element has not create when AUDIO_ELEMENT_TERMINATE
W (43469) AUDIO_ELEMENT: [tts_mp3] Element has not create when AUDIO_ELEMENT_TERMINATE
W (43479) AUDIO_ELEMENT: [tts_i2s] Element has not create when AUDIO_ELEMENT_TERMINATE
I (43489) I2S_IF: Pending out channel for in channel running
W (43499) app_sr: cwg================audio_detect_task running=============while_times=100 again
E (43509) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (43511) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (43517) I2S_IF: STD Mode 1 bits:16/16 channel:2 sample_rate:16000 mask:3
I (43627) Adev_Codec: Open codec device OK
E (43627) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
I (43629) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (43636) I2S_IF: STD Mode 0 bits:16/16 channel:2 sample_rate:16000 mask:3
I (43643) ES7210: Bits 16
I (43649) ES7210: Enable ES7210_INPUT_MIC1
I (43652) ES7210: Enable ES7210_INPUT_MIC2
I (43659) ES7210: Unmuted
I (43659) Adev_Codec: Open codec device OK
E (43664) app_main: cwg==========================Success creating TTS request: ESP_OK
W (43696) app_audio: CWG====ESP_MN_STATE_TIMEOUT == result.stateT===============start_openai() done========;
W (46132) app_main:     Description     Internal        SPIRAM
W (46132) app_main: Current Free Memory 107259          2315568
W (46132) app_main: Min. Ever Free Size 86687           2213080
W (46353) app_sr: cwg================audio_feed_task running=============feed_task_while_times=100 again
W (46420) app_sr: cwg================audio_detect_task running=============while_times=100 again
W (49616) app_sr: cwg================audio_detect_task running=============while_times=100 again

espressif2022, 2024-08-06:
录音测试的，建议可以把二次录音buffer i2s_write 播放简单验证下。
无法唤醒，我觉得就两个原因，feed 的音频文件不对；wakenet 的模型没有 enable.

welkinchan, 2024-08-06:
> 录音测试的，建议可以把二次录音buffer i2s_write 播放简单验证下。 无法唤醒，我觉得就两个原因，feed 的音频文件不对；wakenet 的模型没有 enable.

这个项目里边，我写了两个tts语音播放的处理逻辑。
一个是用这个的i2s_stream，另外一个是用audio_player_play，代码如附件所示。
当我使用audio_player_play的函数时候，语音都是可以正常唤醒的。
当我使用i2s_stream的函数的时候，语音就没办法唤醒了。
其他的代码都是一样的。

所以，肯定不是wakenet模型没有enable的问题，
feed 音频文件我晚上试试看。

[tts_api.txt](https://github.com/user-attachments/files/16504419/tts_api.txt)

welkinchan, 2024-08-06:
> 录音测试的，建议可以把二次录音buffer i2s_write 播放简单验证下。 无法唤醒，我觉得就两个原因，feed 的音频文件不对；wakenet 的模型没有 enable.

验证了一下，使用了我的i2s_stream的代码之后，录音文件确实是异常的。
看来结论确实是i2s的参数被更改了。
请问，在执行了我的i2s_stream的代码之后，应该怎么恢复原先的i2s参数，使得录音可以正常工作？

我在后续的代码里面，把bsp_board_init();以及audio_record_init();都运行了之后，录音还是不正常，无法唤醒。

espressif2022, 2024-08-07:
> > 录音测试的，建议可以把二次录音buffer i2s_write 播放简单验证下。 无法唤醒，我觉得就两个原因，feed 的音频文件不对；wakenet 的模型没有 enable.
> 
> 验证了一下，使用了我的i2s_stream的代码之后，录音文件确实是异常的。 看来结论确实是i2s的参数被更改了。 请问，在执行了我的i2s_stream的代码之后，应该怎么恢复原先的i2s参数，使得录音可以正常工作？
> 
> 我在后续的代码里面，把bsp_board_init();以及audio_record_init();都运行了之后，录音还是不正常，无法唤醒。

`bsp_codec_set_fs(16000, 16, 2)`， 这个应该是可以的，我们在本地也有尝试播放过其他设置的音频，这个可以恢复，有测试过吗？

welkinchan, 2024-08-07:
> 我的i2s_stream的代码

我的i2s_stream的代码执行之后，在pipeline unlink以及destroy之后，我让代码再重新执行bsp_codec_set_fs(16000, 16, 2)，录音文件一直是有问题的。
不仅仅是bsp_codec_set_fs(16000, 16, 2)，
bsp_board_init()，audio_record_init()这些都试了，
一直不行。
连vad检测都判断为没有检测到人声。

espressif2022, 2024-08-07:
> > 我的i2s_stream的代码
> 
> 我的i2s_stream的代码执行之后，在pipeline unlink以及destroy之后，我让代码再重新执行bsp_codec_set_fs(16000, 16, 2)，录音文件一直是有问题的。 不仅仅是bsp_codec_set_fs(16000, 16, 2)， bsp_board_init()，audio_record_init()这些都试了， 一直不行。 连vad检测都判断为没有检测到人声。

如果需要，可以将你这部分的代码，递交一个 PR，我们也这看一下。

welkinchan, 2024-08-08:
> 代码可见 [i2s_stream tts代码.txt](https://github.com/user-attachments/files/16487720/i2s_stream.tts.txt)

感谢。
1. i2s_stream的代码可见上述的链接哈。 [i2s_stream tts代码.txt](https://github.com/user-attachments/files/16487720/i2s_stream.tts.txt)
2. 如果要把整个项目整理上传，我这边需要整理一下，目前只有晚上的时间才有空处理。
3. 我的项目其实是在这个项目的基础上改的，
https://github.com/Acoucou/MY_Assistant/blob/main/main/WebAPI/tts_api.c
然后，尝试在上述项目的基础上，加上这个项目的i2s_stream的逻辑
https://github.com/qq2213216806/esp32-ai-chat/blob/main/my_ai_chat_raw/components/volcengine/volcengine_tts.c

welkinchan, 2024-08-11:
> > > 我的i2s_stream的代码
> > 
> > 
> > 我的i2s_stream的代码执行之后，在pipeline unlink以及destroy之后，我让代码再重新执行bsp_codec_set_fs(16000, 16, 2)，录音文件一直是有问题的。 不仅仅是bsp_codec_set_fs(16000, 16, 2)， bsp_board_init()，audio_record_init()这些都试了， 一直不行。 连vad检测都判断为没有检测到人声。
> 
> 如果需要，可以将你这部分的代码，递交一个 PR（或者你仓库的地址），我们也这看一下。

我把我的项目上传到这里了，如果可以，请帮我看一下哈。https://github.com/welkinchan/chatgpt_wgdemo
1. 项目使用的是esp32s3-box3,idf5.1.0
2. 目前此处tts调用的是这个i2s stream的函数，不知道为什么，现在连第一次tts都不能播放了，https://github.com/welkinchan/chatgpt_wgdemo/blob/main/main/main.c#L220
3. 如果使用这个tts函数，则可以正常唤醒后对话。
https://github.com/welkinchan/chatgpt_wgdemo/blob/main/main/main.c#L216

welkinchan, 2024-08-19:
> 我把我的项目上传到这里了，如果可以，请帮我看一下哈。https://github.com/welkinchan/chatgpt_wgdemo
>
> 1. 项目使用的是esp32s3-box3,idf5.1.0
> 2. 目前此处tts调用的是这个i2s [...]

@espressif2022 请问adf的i2s stream是不是跟idf的audio组件不兼容？所以导致idf的audio组件无法正常播音？

espressif2022, 2024-08-20:
> @espressif2022 请问adf的i2s stream是不是跟idf的audio组件不兼容？所以导致idf的audio组件无法正常播音？

audio 组件本身也是 adf 中抽象出来的，并不是 IDF 的功能，最近有点事，还没来得及处理，非常抱歉。

ESP-Mars, 2025-04-01:
The current issue has been resolved, so we are closing this issue. For the ADF-related problem reported by the user, we recommend opening a new issue.

closed 2025-04-01 by ESP-Mars