espressif/esp-box #162: VAD检测时的舵机声音干扰

welkinchan, 2024-08-04:
最近我在开发一个带有舵机的语音交互项目。
发现，唤醒设备后的VAD检测过程中，VAD经常将舵机的转动声音当成是人说话的声音，
导致VAD一直认为是人在说话，
请问针对这个问题，是否有具体的解决方案或可供参考的解决案例？

espressif2022, 2024-08-05:
后续没有命中命令词，没有超时退出吗？
VAD 是检测语音，但是目前使用的vad并不能有效过滤噪声。

welkinchan, 2024-08-05:
> 后续没有命中命令词，没有超时退出吗？ VAD 是检测语音，但是目前使用的vad并不能有效过滤噪声。

我的代码的应用场景是gpt_demo里边，当ASR阶段，人声发言结束后，系统判断VAD无人声活动后，提交语音文件给ASR接口。
但是因为舵机的干扰，导致VAD一直以为有人声活动，因此无法自动终结录音。
如果是进行命令词输入的话，舵机的噪音倒是不会影响。

请问目前有没有什么办法可以简单过滤舵机的噪音吗？

espressif2022, 2024-08-05:
> 无法自动终结录音

这个好像是我们 chatgpt 那边 https://github.com/espressif/esp-box/issues/162 处理有点简单，有声音就会一直触发
这个需要你自己调整下吧，例如没有命令词触发多少时间，强制退出。

welkinchan, 2024-08-05:
> > 无法自动终结录音
> 
> 这个好像是我们 chatgpt 那边
> 
> https://github.com/espressif/esp-box/blob/18c5d83f72221fe0608a65adb278a187879f0615/examples/chatgpt_demo/main/app/app_sr.c#L131
> 
> 处理有点简单，有声音就会一直触发
> 这个需要你自己调整下吧，例如没有命令词触发多少时间，强制退出。

请问乐鑫SDK中的BSS或者NS打开的话，对抑制排除舵机的这种噪音有效果吗？

espressif2022, 2024-08-05:
> VAD 是检测语音，但是目前使用的vad并不能有效过滤噪声

还是如上描述，VAD 阶段是检测语音，但是目前使用的 vad 并不能有效过滤噪声

closed 2024-08-09 by welkinchan