espressif/esp-box #165: 如何在 ESP32-S3-BOX-3 的硬件上进行 gdb 调试

AaronPMM, 2024-09-20:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

1. ESP32-S3-BOX-3 如何与电脑连接？
2. 我调试的命令是否正确？

### Actual behavior (suspected bug)

目的：
我希望在硬件  ESP32-S3-BOX-3 进行 gdb 调试测试，以方便后续在项目开发上的调试

系统：
ubuntu 20.04

步骤：
1. 安装 openocd-esp32
2. 添加 udev 规则文件，重启系统
3. 编译例程
4. 使用命令 idf.py gdb 或 idf.py oenocd

结果：
gdb 提示：3333 连接超时
openocd 提示：Error:esp_usb_jtag: could not find or open device!

### Steps to reproduce the behavior

正常输入调试命令

### Project release version

master

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

ubuntu 20.04

### Shell

Bash

leeebo, 2024-09-23:
ESP32-S3 默认走 USB-Serial-JTAG 接口进行 GDB 调试，请查阅文档 https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/jtag-debugging/index.html#selecting-jtag-adapter

同时要确认，进行调试时， USB 接口未用作其它用途（和 USB-OTG 功能不能同时使用）

AaronPMM, 2024-09-25:
> ESP32-S3 默认走 USB-Serial-JTAG 接口进行 GDB 调试，请查阅文档 https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/jtag-debugging/index.html#selecting-jtag-adapter
> 
> 同时要确认，进行调试时， USB 接口未用作其它用途（和 USB-OTG 功能不能同时使用）

还是没有明白如何连接到硬件，ESP32-S3-BOX-3 带上底座，共有3个 USB 口，我尝试了3个 USB 都出现同样的问题，所以第一步，我该如何将 ESP32-S3-BOX-3 与 电脑连接？

leeebo, 2024-09-25:
@AaronPMM 只连接 esp32-s3-box 主板上的 USB 接口，底座上的口不要同时连接

AaronPMM, 2024-09-25:
> @AaronPMM 只连接 esp32-s3-box 主板上的 USB 接口，底座上的口不要同时连接

如果这样，为什么在调用 idf.py openocd 时，会提示Error:esp_usb_jtag: could not find or open device!？

ESP-Mars, 2025-04-01:
Issue has remained inactive for an extended period. Closing it now.

closed 2025-04-01 by ESP-Mars