espressif/esp-box #166: ESP32S3-BOX-3 examples 中怎样添加蓝牙BLE 连接功能

yuanfangcn, 2024-10-09:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

当前ESP32S3-BOX-3能够WIFI连接和播放mp3（声音通过喇叭），那怎样添加 连接蓝牙耳机，或者蓝牙音响 输出声音的功能？

ESP-Mars, 2024-10-09:
@yuanfangcn  您好，ESP32-S3 不支持经典蓝牙，无法开发连接蓝牙耳机，或者蓝牙音响的功能。 
但是可以支持通过 USB 外设，基于 USB Audio Class，连接标准的 USB 音频设备，例如 USB 耳机，USB 音响。具体可参考此方案 https://mp.weixin.qq.com/s/JJUSqeaMFeglomAG66bq9w

yuanfangcn, 2024-10-09:
[image]
那这 明确写着支持 蓝牙双模？

yuanfangcn, 2024-10-09:
那ESP32所有系列模组中，有没有/有哪些 能支持WIFI + 经典蓝牙(传输音频)的？

ESP-Mars, 2024-10-09:
> 那ESP32所有系列模组中，有没有/有哪些 能支持WIFI + 经典蓝牙(传输音频)的？
双模是指 Wi-Fi & 蓝牙 双模

ESP32 芯片支持 Wi-Fi +经典蓝牙（BR/EDR)+低功耗蓝牙（BLE）

ESP-Mars, 2025-04-01:
This issue has remained inactive for an extended period. Closing it now.

closed 2025-04-01 by ESP-Mars