espressif/esp-box #168: ESP32-s3-box-3 LCD display: Hardware or Software SPI

bpavlov, 2024-11-03:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

I am unable to find how the LCD display is connected to ESP32.
I want to use the Adafruit_ILI9341 library in Arduino IDE. From the code I can see three methods:

```
/**************************************************************************/
/*!
    @brief  Instantiate Adafruit ILI9341 driver with software SPI
    @param    cs    Chip select pin #
    @param    dc    Data/Command pin #
    @param    mosi  SPI MOSI pin #
    @param    sclk  SPI Clock pin #
    @param    rst   Reset pin # (optional, pass -1 if unused)
    @param    miso  SPI MISO pin # (optional, pass -1 if unused)
*/
/**************************************************************************/
Adafruit_ILI9341::Adafruit_ILI9341(int8_t cs, int8_t dc, int8_t mosi,
                                   int8_t sclk, int8_t rst, int8_t miso)
    : Adafruit_SPITFT(ILI9341_TFTWIDTH, ILI9341_TFTHEIGHT, cs, dc, mosi, sclk,
                      rst, miso) {}

/**************************************************************************/
/*!
    @brief  Instantiate Adafruit ILI9341 driver with hardware SPI using the
            default SPI peripheral.
    @param  cs   Chip select pin # (OK to pass -1 if CS tied to GND).
    @param  dc   Data/Command pin # (required).
    @param  rst  Reset pin # (optional, pass -1 if unused).
*/
/**************************************************************************/
Adafruit_ILI9341::Adafruit_ILI9341(int8_t cs, int8_t dc, int8_t rst)
    : Adafruit_SPITFT(ILI9341_TFTWIDTH, ILI9341_TFTHEIGHT, cs, dc, rst) {}

#if !defined(ESP8266)

/**************************************************************************/
/*!
    @brief  Instantiate Adafruit ILI9341 driver with hardware SPI using
            a specific SPI peripheral (not necessarily default).
    @param  spiClass  Pointer to SPI peripheral (e.g. &SPI or &SPI1).
    @param  dc        Data/Command pin # (required).
    @param  cs        Chip select pin # (optional, pass -1 if unused and
                      CS is tied to GND).
    @param  rst       Reset pin # (optional, pass -1 if unused).
*/
/**************************************************************************/
Adafruit_ILI9341::Adafruit_ILI9341(SPIClass *spiClass, int8_t dc, int8_t cs,
                                   int8_t rst)
    : Adafruit_SPITFT(ILI9341_TFTWIDTH, ILI9341_TFTHEIGHT, spiClass, cs, dc,
                      rst) {}
#endif // end !ESP8266
```

The first one uses Software SPI. others - hardware SPI. Since Hardware SPI is faster I want to use hardware SPI. But I don't know if the display is connected via SPI to ESP32. 
I found that the default SPI pins for ESP-S3 are:

SPI    | MOSI     | MISO        | CLK.        | CS
----- | -------- | --------- | --------- | --------
SPI2 | GPIO 35 | GPIO 37  | GPIO 36  | GPIO 39
SPI3 | GPIO 11  | GPIO 13  | GPIO 12   | GPIO 10

But I don't know how to match this with:
[image]

Also in Arduino IDE SPI (HSPI and VSPI) pins depend on the board and cannot be changed. What board should I use?

espressif2022, 2024-11-04:
Hello, this SPI pin is just the default pin. It can be configured to other pins.

bpavlov, 2024-11-04:
> Hello, this SPI pin is just the default pin. It can be configured to other pins.

Thank you for the replay, but it is not very clear. "This SPI pin is just the default pin" - which SPI pin? And how it can be configured in Arduino IDE. And also is the display connected via Hardware SPI to ESP32? And if yes what are the pins?

espressif2022, 2024-11-22:
Sorry, I haven't replied for a long time. Please see if this Arduino [repository](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/master/examples/LCD/SPI/SPI.ino) is helpful to you.

ESP-Mars, 2025-04-01:
This issue has remained inactive for an extended period. Closing it now.

closed 2025-04-01 by ESP-Mars