espressif/esp-box #172: chatGPT example idf build failure

tpang0609, 2024-12-21:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I expect **idf.py build** for chatgpt_demo to finish build successfully.

### Actual behavior (suspected bug)

Instead, the idf.py build results in many compilation errors as shown in the terminal outputs.

### Error logs or terminal output

```shell
esp-box/examples/chatgpt_demo$  idf.py build
Executing action: all (aliases: build)
Running ninja in directory /home/tpang/esp/esp-idf/esp-box/examples/chatgpt_demo/build
Executing "ninja all"...
[0/1] Re-running CMake...
-- Minimal build - OFF
-- Building ESP-IDF components for target esp32s3
Processing 20 dependencies:
[20/20] idf (5.5.0)

.......

In file included from /home/tpang/esp/esp-idf/components/log/include/esp_log.h:19,
                 from /home/tpang/esp/esp-idf/components/esp_common/include/esp_check.h:9,
                 from /home/tpang/esp/esp-idf/esp-box/examples/chatgpt_demo/main/app/app_sr.c:15:
/home/tpang/esp/esp-idf/esp-box/examples/chatgpt_demo/main/app/app_sr.c: In function 'audio_detect_task':
/home/tpang/esp/esp-idf/esp-box/examples/chatgpt_demo/main/app/app_sr.c:106:27: error: expected ')' before 'LOG_BOLD'
  106 |             ESP_LOGI(TAG, LOG_BOLD(LOG_COLOR_GREEN) "wakeword detected");
      |                           ^~~~~~~~
/home/tpang/esp/esp-idf/components/log/include/esp_log_format.h:15:88: note: in definition of macro 'LOG_FORMAT'
   15 | #define LOG_FORMAT(letter, format)  LOG_COLOR_ ## letter #letter " (%" PRIu32 ") %s: " format LOG_RESET_COLOR "\n"
      |                                                                                        ^~~~~~
/home/tpang/esp/esp-idf/components/log/include/esp_log.h:163:38: note: in expansion of macro 'ESP_LOG_LEVEL'
  163 |         if (_ESP_LOG_ENABLED(level)) ESP_LOG_LEVEL(level, tag, format, ##__VA_ARGS__); \
      |                                      ^~~~~~~~~~~~~
/home/tpang/esp/esp-idf/components/log/include/esp_log.h:72:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
   72 | #define ESP_LOGI( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_INFO,    tag, format, ##__VA_ARGS__)
      |                                      ^~~~~~~~~~~~~~~~~~~
/home/tpang/esp/esp-idf/esp-box/examples/chatgpt_demo/main/app/app_sr.c:106:13: note: in expansion of macro 'ESP_LOGI'
  106 |             ESP_LOGI(TAG, LOG_BOLD(LOG_COLOR_GREEN) "wakeword detected");
      |             ^~~~~~~~
/home/tpang/esp/esp-idf/components/log/include/esp_log.h:140:60: note: to match this '('
  140 |         if (level==ESP_LOG_ERROR )          { esp_log_write(ESP_LOG_ERROR,      tag, LOG_FORMAT(E, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
      |                                                            ^
/home/tpang/esp/esp-idf/components/log/include/esp_log.h:163:38: note: in expansion of macro 'ESP_LOG_LEVEL'
  163 |         if (_ESP_LOG_ENABLED(level)) ESP_LOG_LEVEL(level, tag, format, ##__VA_ARGS__); \
      |                                      ^~~~~~~~~~~~~
/home/tpang/esp/esp-idf/components/log/include/esp_log.h:72:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
   72 | #define ESP_LOGI( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_INFO,    tag, format, ##__VA_ARGS__)
      |                                      ^~~~~~~~~~~~~~~~~~~

......

/home/tpang/esp/esp-idf/components/log/include/esp_log_color.h:100:31: error: format '%s' expects a matching 'char *' argument [-Werror=format=]
  100 | #define LOG_COLOR_I           ""
      |                               ^~
/home/tpang/esp/esp-idf/components/log/include/esp_log_format.h:15:37: note: in expansion of macro 'LOG_COLOR_I'
   15 | #define LOG_FORMAT(letter, format)  LOG_COLOR_ ## letter #letter " (%" PRIu32 ") %s: " format LOG_RESET_COLOR "\n"
      |                                     ^~~~~~~~~~
/home/tpang/esp/esp-idf/components/log/include/esp_log.h:144:86: note: in expansion of macro 'LOG_FORMAT'
  144 |         else                                { esp_log_write(ESP_LOG_INFO,       tag, LOG_FORMAT(I, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
      |                                                                                      ^~~~~~~~~~
/home/tpang/esp/esp-idf/components/log/include/esp_log.h:163:38: note: in expansion of macro 'ESP_LOG_LEVEL'
  163 |         if (_ESP_LOG_ENABLED(level)) ESP_LOG_LEVEL(level, tag, format, ##__VA_ARGS__); \
      |                                      ^~~~~~~~~~~~~
/home/tpang/esp/esp-idf/components/log/include/esp_log.h:72:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
   72 | #define ESP_LOGI( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_INFO,    tag, format, ##__VA_ARGS__)
      |                                      ^~~~~~~~~~~~~~~~~~~
/home/tpang/esp/esp-idf/esp-box/examples/chatgpt_demo/main/app/app_sr.c:126:13: note: in expansion of macro 'ESP_LOGI'
  126 |             ESP_LOGI(TAG, LOG_BOLD(LOG_COLOR_GREEN) "AFE_FETCH_CHANNEL_VERIFIED, channel index: %d\n", res->trigger_channel_id);
      |             ^~~~~~~~
cc1: some warnings being treated as errors
[1285/1688] Building C object esp-idf/main/CMa..._idf_main.dir/ui/fonts/ui_font_KaiTiCN20.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/tpang/esp/esp-idf/esp-box/examples/chatgpt_demo/build/log/idf_py_stderr_output_13672 and /home/tpang/esp/esp-idf/esp-box/examples/chatgpt_demo/build/log/idf_py_stdout_output_13672
```

### Steps to reproduce the behavior

Build and Flash

0. $ . $HOME/esp/esp-idf/export.sh
1. Clone the Github repository
	$  git clone https://github.com/espressif/esp-box
2. Change the working directory
	$ cd esp-box/examples/chatgpt_demo/factory_nvs
3. Hardware Selection
To select the appropriate hardware (ESP32-S3-BOX, ESP32-S3-BOX-Lite or ESP32-S3-BOX-3) and access the HMI Board Cofig, use the following command.
	$ idf.py menuconfig
4. Build the factory_nvs
	$ idf.py build
5. Change the working directory
	cd esp-box/examples/chatgpt_demo/
6. Hardware Selection
To select the appropriate hardware (ESP32-S3-BOX, ESP32-S3-BOX-Lite or ESP32-S3-BOX-3) and access the HMI Board Cofig, use the following command.
	$ idf.py menuconfig
7. Build the chatgpt_demo
	$  idf.py build

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Ubuntu 22.04

### Shell

Bash

espressif2022, 2024-12-23:
Hello，I have also encountered the `LOG_BOLD` error. This is related to the IDF version. Can you switch to v5.3 or other versions to test it?

tpang0609, 2024-12-24:
@espressif2022 ,  Thanks. After switching to v5.3, I finished  idf.py build and idf.py flash commands.  Now the esp32s3Box3 screen displayed  the followings: Steps 1. Connect ESP-Box to PC. 2. Open "ESP-Box" folder. 3. Edit "CONFIG.INI" file with Wi-Fi SSID, password, and OpenAI key.  4. Save the file.  5. Restart the device. (Restart icon)   ==> I did the first step and have finished idf.py flash.  For step 2, I do have an 'esp-box" folder which contains the esp-box/examples/chatgpt_demo folder but there is no "CONFIG.INI" file or 'config.ini' file.  Where can I find it ?  Or it is on the ESP-Box device? But the following command get no response from the device: idf.py -p /dev/ttyACM0 monitor
Executing action: monitor
Running idf_monitor in directory /home/tparng/esp/esp-box/examples/chatgpt_demo
Executing "/home/tparng/.espressif/python_env/idf5.3_py3.10_env/bin/python /home/tparng/esp/esp-idf/tools/idf_monitor.py -p /dev/ttyACM0 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 --revision 0 /home/tparng/esp/esp-box/examples/chatgpt_demo/build/chatgpt_demo.elf -m '/home/tparng/.espressif/python_env/idf5.3_py3.10_env/bin/python' '/home/tparng/esp/esp-idf/tools/idf.py' '-p' '/dev/ttyACM0'"...
--- esp-idf-monitor 1.5.0 on /dev/ttyACM0 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H**_**
What should I do now?  Thanks again.

espressif2022, 2024-12-30:
Maybe this [component ](https://github.com/espressif/esp-box/blob/18c5d83f72221fe0608a65adb278a187879f0615/examples/chatgpt_demo/factory_nvs/main/idf_component.yml#L3) is broken，I'll check it later.

miaoshouai, 2025-01-07:
same problem

Horion0415, 2025-01-07:
You can try erasing the flash first and then flashing again. Use the command:  
`idf.py -p /dev/ttyACM0 erase-flash`

muskning, 2025-01-25:
change the log setting to support color

ESP-Mars, 2025-04-01:
The current issue has been resolved, so we are closing this issue.

closed 2025-04-01 by ESP-Mars