espressif/esp-box #173: 关于出厂固件相关问题。

PoohWoah, 2024-12-26:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

我拿到固件后一切正常，但是我已经编写了其他代码，发现我的喇叭声音达不到出厂固件那样，所有的操作我都试过了，请问我是错在哪里了呢？代码如下：
`/* 唤醒成功提示词 */
static esp_audio_handle_t wekeup_player_init()
{
    duer_evt = xEventGroupCreate();

    esp_audio_cfg_t cfg = DEFAULT_ESP_AUDIO_CONFIG();
    audio_board_handle_t board_handle = audio_board_init();

    cfg.vol_handle = board_handle->audio_hal;
    cfg.vol_set = (audio_volume_set)audio_hal_set_volume;
    cfg.vol_get = (audio_volume_get)audio_hal_get_volume;
    cfg.resample_rate = 48000;
    cfg.prefer_type = ESP_AUDIO_PREFER_MEM;
    cfg.vol_set(cfg.vol_handle, 100);
    audio_hal_set_volume(cfg.vol_handle,100);
    pipeline_play = esp_audio_create(&cfg);
    audio_hal_ctrl_codec(board_handle->audio_hal, AUDIO_HAL_CODEC_MODE_BOTH, AUDIO_HAL_CTRL_START);

    // Create readers and add to esp_audio
    spiffs_stream_cfg_t flash_cfg = SPIFFS_STREAM_CFG_DEFAULT();
    flash_cfg.type = AUDIO_STREAM_READER;
    esp_audio_input_stream_add(pipeline_play, spiffs_stream_init(&flash_cfg));

    // Add decoders and encoders to esp_audio
    mp3_decoder_cfg_t mp3_dec_cfg = DEFAULT_MP3_DECODER_CONFIG();
    mp3_dec_cfg.task_core = 1;
    esp_audio_codec_lib_add(pipeline_play, AUDIO_CODEC_TYPE_DECODER, mp3_decoder_init(&mp3_dec_cfg));

    // Create writers and add to esp_audio
#if (CONFIG_ESP32_S3_KORVO2_V3_BOARD == 1) && (CONFIG_AFE_MIC_NUM == 1)
    i2s_stream_cfg_t i2s_writer = I2S_STREAM_CFG_DEFAULT_WITH_PARA(I2S_NUM_0, 48000, I2S_DATA_BIT_WIDTH_16BIT, AUDIO_STREAM_WRITER);
#else
    i2s_stream_cfg_t i2s_writer = I2S_STREAM_CFG_DEFAULT_WITH_PARA(I2S_NUM_0, 48000, CODEC_ADC_BITS_PER_SAMPLE, AUDIO_STREAM_WRITER);
    i2s_writer.need_expand = (CODEC_ADC_BITS_PER_SAMPLE != 16);
#endif

    audio_element_handle_t i2s_stream_writer = i2s_stream_init(&i2s_writer);
    esp_audio_output_stream_add(pipeline_play, i2s_stream_writer);

    esp_audio_vol_set(pipeline_play, 100);

    return pipeline_play;
}`

espressif2022, 2024-12-26:
> 喇叭声音达不到出厂固件那样

是指音量吗，用的也是 factory_demo 本地的音频？

PoohWoah, 2024-12-26:
> > 喇叭声音达不到出厂固件那样
> 
> 是指音量吗，用的也是 factory_demo 本地的音频？

不是是我从其他例子里面复制过来的 haode.mp3

另外我也在尝试调试录音播放方面的代码，用的是例子spiffs_amr_resample_example，在这里面 我也发现了同样的问题，我录音后播放，除非我说话声音特别大，不然他播放的声音很小甚至没有，（PS:我说话的声音很大他出来的声音也不大，只是可以贴近听到），用的也是box3 没有改动过

PoohWoah, 2024-12-26:
> > > 喇叭声音达不到出厂固件那样
> > 
> > 
> > 是指音量吗，用的也是 factory_demo 本地的音频？
> 
> 不是是我从其他例子里面复制过来的 haode.mp3
> 
> 另外我也在尝试调试录音播放方面的代码，用的是例子spiffs_amr_resample_example，在这里面 我也发现了同样的问题，我录音后播放，除非我说话声音特别大，不然他播放的声音很小甚至没有，（PS:我说话的声音很大他出来的声音也不大，只是可以贴近听到），用的也是box3 没有改动过

我将功放喇叭换成了MAX98357A进行了尝试，很遗憾，结果和上述相同

espressif2022, 2024-12-30:
audio_hal_set_volume(cfg.vol_handle,100); 音量设置接口是正常的么？

ESP-Mars, 2025-04-01:
This issue has remained inactive for an extended period. Closing it now.