espressif/esp-box #174: 更新LVGL版本

catmeowjiao, 2025-01-01:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Described the feature in detail and justified the reason for the request.
- [X] Provided specific use cases and examples.

### Feature description

我的ESP32-S3-BOX上的LVGL版本较旧，我希望更新LVGL以支持新特性。

我试过在`idf_component.yml`中将`esp-box`改为最新版本来支持LVGL v9，但是`idf.py build`时报错：
```
CMake Error at /home/meow/esp/esp-idf/tools/cmake/build.cmake:552 (message):
  ERROR: Because project depends on both espressif/esp-box (>=3.1.0~1) and
  espressif/esp-box (3.0.*), version solving failed.

Call Stack (most recent call first):
  /home/meow/esp/esp-idf/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:13 (project)
```

请问`espressif/esp-box (3.0.*)`这个依赖是在哪里出现的？有什么办法更新LVGL到v9吗？

### Use cases

LVGL v9提升了性能，进行了许多内部重构。

catmeowjiao, 2025-01-01:
@espressif2022

espressif2022, 2025-01-02:
[espressif/esp-box (3.0.*) 的依赖 ](https://github.com/espressif/esp-box/blob/18c5d83f72221fe0608a65adb278a187879f0615/components/bsp/idf_component.yml#L9)
这边 BSP 使用的老版本，强制依赖 LVGL 8.4 版本，V8 和 V9 的工程不能兼容，并不能通过切换 LVGL 版本来适配工程

catmeowjiao, 2025-01-04:
谢谢

closed 2025-01-04 by catmeowjiao